realsenseai/librealsense #3759: Any suggestion for jetson nano installation?

dongxuanlb, 2019-04-15:
I notice that jetson nano with kernel-4.9 and jetpack3.3. And the cpu is a57 and no onboard emmc.

Has any one had installed the sdk successfully?

christopherbate, 2019-04-16:
NVIDIA is using the camera (435 I believe) as a demonstrated supported part in their[ Isaac robotics ](https://docs.nvidia.com/isaac/isaac/doc/setup.html) program. In some youtube videos you can see their demonstrated robots with the camera and theJetson Nano.

With the Isaac build system, everything is cross compiled on an x86 host and shipped over the network to the Nano, and they are using Jetpack 4.2

I've been trying to get my camera (435i) to work with the nano, but so far no luck. It seems to show up as a USB2.0 device. Might be a power issue as the same thing happens with my laptop sometimes. 

I've cross compiled their kernel with the suggested patches from Intel, should be able to test that later tonight.

constructive-realities, 2019-04-19:
Hmm...  yes it would be nice to use 400 series with jetpack 4.2 without a ton of kernel module hijinx.

bys1123, 2019-04-19:
@christopherbate Is that works?

ChaseLewis, 2019-04-20:
trying to compile the real sense library on the nano but when I run cmake I get 'xinerama headers were not found'. Any ideas?

christopherbate, 2019-04-20:
Hi, I would suggest following the directions on the Isaac robotics link I posted above. It avoid having to compile the library on the device. They also give instructions on some dependencies which need to be installed through APT. I believe they use 430/415, and my issues with 435i may be particular to the 435i model. I will try some more this weekend.

christopherbate, 2019-04-21:
@ChaseLewis You need the graphical libraries installed (glfw). Remove your cmake "build" folder and re-run CMake with the `-DBUILD_GRAPHICAL_EXAMPLES=false` flag. This will eliminate building the viewer and other tools that rely on GUI elements.

Here are the steps I followed for running RS435i on the Nano. We rebuilt the kernel on the device to make things simpler. I recommend starting with a fresh SD card just flashed with the latest JetPack release.

1. Go to [the L4T download page](https://developer.nvidia.com/embedded/linux-tegra) and download the "BSP Sources" for the Nano. 
2. Untar the sources with `untar -xf [name of downloaded file`
3. `cd public_sources` then `untar -xf kernel_src.tbz2`
4. `cd kernel/kernel4.9` and `zcat /proc/config.gz > .config`
5. `make menuconfig` and enable `HID_SENSOR_ACCEL_3D` and `HID_SENSOR_GYRO_3D` under `device drivers -> Industrial I/O (IIO) -> digital gyroscopes / accelerometers` These are needed by 435i
6. Apply HID patch `patch -p1 < [librealsense cloned repo directory]/scripts/realsense-hid-ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1.patch`
6. `make -j(use a number 3-5)`. Recommend doing this from terminal over SSH as it will lock up GUI.
7. `sudo make_modules_install` 
8. `sudo mv /boot/Image /boot/Image-original` and `cp ./arch/arm64/boot/Image /boot/Image`
9. Do a reboot and verify system loads. 
10. Build realsense library and install. Verify device detected with `rs-enumerate-devices` then use `rs-data-collect` to collect some data. Library was built with options `-DHWM_OVER_XU=true -DBUILD_GRAPHICAL_EXAMPLES=false`

Hope this helps. Again, this is for the 435i, the only camera I have.

bys1123, 2019-04-21:
Thanks, I'm also only have D435i, I want to install it to my TX2 runs Jetpack4.2(ubuntu18.04), This information really helpful.

eric-schleicher, 2019-04-22:
@christopherbate, can you elaborate on step 5.   `make menuconfig`  

following through the instructions successfully up until this point.

I get the following error
```
rosuser@jet-nano:~/Downloads/public_sources/kernel$ make menuconfig
make: *** No rule to make target 'menuconfig'.  Stop.
rosuser@jet-nano:~/Downloads/public_sources/kernel$ cd kernel-4.9/
rosuser@jet-nano:~/Downloads/public_sources/kernel/kernel-4.9$ make menuconfig
  HOSTCC  scripts/kconfig/mconf.o
<command-line>:0:12: fatal error: curses.h: No such file or directory
compilation terminated.
scripts/Makefile.host:118: recipe for target 'scripts/kconfig/mconf.o' failed
make[1]: *** [scripts/kconfig/mconf.o] Error 1
Makefile:565: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 2
```
I'm inferring that the steps implies we should expect some UI in which we change the `HID_SENSOR_ACCEL_3D` and `HID_SENSOR_GYRO_3D` values?   or did i read that wrong?

I opened the .config file and found the HID sensor_accell_3d entries, but the file state very clearly not to edit (at the top)

Any help appreciated.

christopherbate, 2019-04-23:
You need to install `libncurses5` i.e. `sudo apt-get install libncurses5".

But yes, you will get a terminal UI to make changes. But yes, typically in
doing something like this you can edit that .config file directly. If
you've never done this before, it might be helpful to read a general
tutorial about how to compile the Linux kernel and about LKMs. Then use a
virtual machine running Linux linux to practice modifying the kernel. Then
this will become familiar to you.

Now I have to mention some unfortunate news... these steps might not work
for you. My camera was recognized but didn't fully work. I had to apply
another patch and now have it working. I'll post a more in-depth follow up
later today.

eric-schleicher, 2019-04-24:
ok...  eagerly awaiting.

ChaseLewis, 2019-04-24:
So I got it to build, but now it says 'no module named pyrealsense2'. If  I use rs-enumerate devices the camera is found so it has something to do with the python installation after building.

I set the executable to python3 and used sudo make install so I'm a bit confused what I'm doing wrong. I'm trying to run the script with python 3.6.7
```
chaserlewis@chaserlewis-desktop:/usr/local/lib$ ls
cmake                    pybackend2.cpython-36m-aarch64-linux-gnu.so
libfw.a                  pybackend2.cpython-36m-aarch64-linux-gnu.so.2
librealsense2.so         pybackend2.cpython-36m-aarch64-linux-gnu.so.2.20.0
librealsense2.so.2.20    pybackend2.cpython-36m-aarch64-linux-gnu.so.2.21.0
librealsense2.so.2.20.0  pyrealsense2.cpython-36m-aarch64-linux-gnu.so
librealsense2.so.2.21    pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.21
librealsense2.so.2.21.0  pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.21.0
librealsense-file.a      python2.7
libtm.a                  python3.6
pkgconfig
```
 PythonPath
```
chaserlewis@chaserlewis-desktop:/usr/local/lib$ echo $PYTHONPATH
:/usr/local/lib

```

How I called CMAKE
`chaserlewis@chaserlewis-desktop:/usr/local/lib$ sudo cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python3`

Any ideas?

christopherbate, 2019-04-24:
@ChaseLewis Not sure as I don't use the python module. Generally for cmake, you call "cmake ../ (options)" then "make" then "sudo make install". 

I have some more complete instructions for those interested [on my github](https://github.com/christopherbate/Nano-Realsense-Instructions) that includes an example `.config` file and a single `.patch` file for the R32.1 kernel. A few things had to be manually tweaked in the source for the patches to take, so I just wrapped it into one patch. Let me know if any issues and I can try to resolve. The instructions are for cross-compiling the kernel from a seperate computer, so it should be faster.

kevkruemp, 2019-05-04:
@christopherbate I'm trying to follow your guide but I'm encountering an issue with step 4. I'm in a terminal in the ~/Downloads/public_sources/kernel/kernel-4.9 folder but when I try to do the command `sudo zcat /proc/config.gz > .config` I get the error: "bash: .config: Permission denied" How do I get permission? I thought sudo giving me admin access would be enough, but apparently it is not. Any help you can provide would be greatly appreciated. (I have no experience doing kernel patches) :(

christopherbate, 2019-05-05:
Ok, that command is for grabbing the configuration for the kernel for the computer you run that command on. So you need to be in your Nano's terminal for that command. Then maybe try this (on your Nano):
`sudo zcat /proc/config.gz > ~/my-config`

Then copy that file `~/my-config` to your x86 computer and place it in the `/mnt/tegra` or whichever directory you specified as the output directory for the kernel build.

evovolker, 2019-05-17:
@christopherbate I am working through your kernel patch instructions. When I run the patches I get a message "patching file drivers/iio/accel/hid-sensor-accel-3d.c
Reversed (or previously applied) patch detected!  Assume -R? [n]" 
Any ideas?

RealSenseCustomerSupport, 2019-06-06:
----------------------------------------------

@dongxuanlb @evovolker Did you still have issue? Looking forward to your update. Thanks!

evovolker, 2019-06-06:
@RealSenseCustomerSupport I got it to work by merging the patches provided by @jetsonhacksnano with an additional patch from @christopherbate,
Thanks

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@evovolker Great news!
@dongxuanlb Any good news from your side? Thanks!

RealSenseCustomerSupport, 2019-06-19:
----------------------------------------------

@dongxuanlb If no other questions for this ticket, will close it now. Please feel free to create new ticket if you still have questions/issues. Thanks!

closed 2019-06-19 by RealSenseCustomerSupport

bigbellmercy, 2020-10-05:
If a RealSense camera does not work after installation on Jetson Nano, see a solution here (Installation method only for Jetson Nano):
https://github.com/IntelRealSense/librealsense/issues/7333#issuecomment-691970195

In my case, 'No RealSense devices were found!' happened when I installed `ros-melodic-realsense2-camera`. And, `librealsense2-dkms` cannot be obtained when I use `apt-get` using the standard 'Linux Distribution' installation method. But, after I used the method of the link above, it worked well.