realsenseai/librealsense #3762: When I use the function new pcl::io::OpenNI2Grabber(); An exception appears

coldcoda, 2019-04-15:
Hello @dorodnic ,
Here is my code , When I run it, an exception appears:
Exception:
[image]

Code:
#include <pcl/io/openni2_grabber.h>
#include <pcl/visualization/cloud_viewer.h>

#ifdef x64
# define sleep(x) Sleep((x)*1000)
#endif

class SimpleOpenNIViewer
{
public:
	SimpleOpenNIViewer() : viewer("PCL OpenNI Viewer") {}

	void cloud_cb_(const pcl::PointCloud<pcl::PointXYZ>::ConstPtr &cloud)
	{
		if (!viewer.wasStopped())
			viewer.showCloud(cloud);
	}

	void run()
	{
		pcl::Grabber* interface = new pcl::io::OpenNI2Grabber();

		boost::function<void(const pcl::PointCloud<pcl::PointXYZ>::ConstPtr&)> f =
			boost::bind(&SimpleOpenNIViewer::cloud_cb_, this, _1);

		interface->registerCallback(f);

		interface->start();

		while (!viewer.wasStopped())
		{
			boost::this_thread::sleep(boost::posix_time::seconds(1) * 1000);
		}

		interface->stop();
	}

	pcl::visualization::CloudViewer viewer;
};

int main()
{
	SimpleOpenNIViewer v;
	v.run();
	return 0;
}
But I can run  the examples in OpenNI2/Samples, I don't know how to solve it. Can you give me some advices?

Thanks a lot!!!

coldcoda, 2019-04-18:
I have soled it!

closed 2019-04-29 by RealSenseCustomerSupport

ifq, 2019-08-23:
Hi, @coldcoda how did you solve it? I have the same issue.