realsenseai/librealsense #3765: The error(SimpleViewer: Device open failed: DeviceOpen using default: no devices found)

coldcoda, 2019-04-15:
Hello @dorodnic , @martinus , @seebq 
When I recompile the SimpleViewer project (the example of OpenNI2), and run it that prompting no device.
[image]

But I can launch any OpenNI2 example (SimpleRead SimpleViewer NiViewer) located at OPENNI2_DIR/Samples/Bin/ .

So what's wrong with my project? And how can I solve it?
Can you give me some advices?
Thanks a lot!!!

coldcoda, 2019-04-18:
I have solved it!

AnxiousBuddha, 2019-04-18:
I'm having the same problem but with processing what did you do to fix it?

coldcoda, 2019-04-18:
Have you got the rs2driver.dll and realsense2.dll? You can copy them to your OPENNI2's directory (like me:    D:\Program Files\OpenNI2\Redist\OpenNI2\Drivers)

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

Hi coldcoda,
Thanks for the updates.

closed 2019-05-07 by RealSenseCustomerSupport

eliotts, 2020-03-09:
Hi, maybe I am missing something, but where can I find rs2driver.dll and realsense2.dll?

thanks

jiapei100, 2024-07-10:
> I have solved it!

How? 
Are you sure you can stream a **Real Sense Sensor** using **OpenNI2**?

MartyG-RealSense, 2024-07-11:
Hi @jiapei100  Yes, RealSense cameras have a compatibility 'wrapper' for OpenNI2.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2