realsenseai/librealsense #3768: Optical filters and lenses for RealSense cameras

TDVL, 2019-04-15:
generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

TDVL, 2019-04-15:
I am using the RealSense D415 camera and it gives me good results, I only have a big problem with the sunlight.
I saw one post about optical filters.
I have a question.
Are there any UV, optical, ND filters and lenses designed for the RealSense cameras (can I buy them somewhere)?
I mean, filters designed to fit in front the cameras well. I can get an UV filter, but it is a circle-shaped and there is a possibility that the sunlight will penetrate between the filter and sensor. That can be a serious problem.
Thank you in advance.
Best regards,

MartyG-RealSense, 2019-04-15:
I have posted a reply on the Intel Support site's copy of this question.  Thank you.

https://forums.intel.com/s/question/0D70P000006GoP6SAK

closed 2019-05-24 by RealSenseCustomerSupport