realsenseai/librealsense #3769: AddVideoStream generates a NullReferenceException in unity while attempting to create SoftwareVideoStream based RsFrameProvider

stephen-systemfriend, 2019-04-16:
| Platform                                 | PC |
| SDK Version                            |  2.20 }                          | 
| Language                            |  unity                           | 

When I attempt to call add videostream following the same pattern as in https://github.com/IntelRealSense/librealsense/tree/development/examples/software-device it fails with a NullReferenceException when I call AddVideoStream.

As a bonus question, I also can't find a way to populate ActiveProfile in C# without creating a Pipeline, which seems wrong given that here I'm not using it.

heres an MWE of the issue - if you put this object into unity it will fail when it calls add video stream.

`using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using Intel.RealSense;
using System;

public class RSSoftwareDevTest : RsFrameProvider
{

    public float[] intrinsicsCoeffics = new float[] { 0, 0, 0, 0, 0 };

    //the device by which we generate our frames
    private SoftwareDevice framegenerator;
    private SoftwareSensor colorGenerator;
    private VideoStreamProfile colorStream;
    private Syncer sync;
    private int imgWidth, imgHeight;

    void Awake()
    {
        framegenerator = new SoftwareDevice();
        colorGenerator = framegenerator.AddSensor("Color");
    }
    
    //this is a frame provider
    /// <summary>
    /// Notifies upon streaming start
    /// </summary>
    public override event Action<PipelineProfile> OnStart;

    /// <summary>
    /// Notifies when streaming has stopped
    /// </summary>
    public override event Action OnStop;

    /// <summary>
    /// Fired when a new frame is available
    /// </summary>
    public override event Action<Frame> OnNewSample;

    private byte[] bufdata;

    // Update is called once per frame
    void Update()
    {
        int newWidth = 100;
        int newHeight = 80;
        if (newWidth != imgWidth || newHeight != imgHeight)
        {
            //different frame shape therefore we have to setup the streams
            imgHeight = newHeight;
            imgWidth = newWidth;

            var colorProfile = new SoftwareVideoStream
            {
                type = Stream.Color,
                index = 0,
                uid = 0,
                width = imgWidth,
                height = imgHeight,
                fps = 10,
                bpp = 3,
                format = Format.Rgb8
            };

            var intrinsics = new Intrinsics
            {
                width = imgWidth,
                height = imgHeight,
                ppx = imgWidth / 2,
                ppy = imgHeight / 2,
                fx = imgWidth,
                fy = imgHeight,
                model = Distortion.BrownConrady,
                coeffs = intrinsicsCoeffics
            };
            colorProfile.intrinsics = intrinsics;

            colorStream = colorGenerator.AddVideoStream(colorProfile);
            
            colorGenerator.Open(colorStream);
            sync = new Syncer();

            colorGenerator.Start(sync.SubmitFrame);
        }

        //this line wont work in the MWE, but we dont get here anyway so...
        //colorGenerator.AddVideoFrame<byte>(buf.Color.ImageData.ToByteArray(), buf.Color.Width * buf.Color.BytesPerPixel,
        //    buf.Color.BytesPerPixel, buf.Timestamp, TimestampDomain.HardwareClock, buf.Frame, colorStream);
        var frameset = sync.WaitForFrames();

        //send it using the RS pipeline
        OnNewSample(frameset.AsFrame());

    }
}
`

ogoshen, 2019-04-16:
I tried your script, and I'm not getting a NRE.
I am getting an exception later in `sync.WaitForFrames` (`ExternalException` *Frame did not arrive in time!*) which is correct since you're not adding any frames (commented out code).

You're welcome to post the full stack trace and any other bits of code or scene setup that you can share.

Regrading the `ActiveProfile`, that's right, you cannot create a `PipelineProfile` object yourself.
There might be a way to use the (unexposed so far in C#) `rs2_context_add_software_device` function to add the the device first and then use the regular pipeline with it.

Please try and explain the use-case here, maybe there's a way to support it, or a different way to achieve the same.

stephen-systemfriend, 2019-04-17:
So I have a remote device with a RealSense camera attached to it, and it is streaming some of the frames across to me over the network. I want to display the stream from the RealSense Device in the unity app I am building, and possibly use some of the RealSense algorithms at this end.

The scene is literally just an empty scene with an object with that script on it (obviously not in the actual thing, but the error is the same either way).

If it helps, heres the stack trace

NullReferenceException: Object reference not set to an instance of an object
Intel.RealSense.SoftwareSensor.AddVideoStream (Intel.RealSense.SoftwareVideoStream profile) (at <6a9ff6a231d443aba045662b8ae53674>:0)
RSSoftwareDevTest.Update () (at Assets/Scripts/RSSoftwareDevTest.cs:80)

ogoshen, 2019-04-17:
Got it, and `SoftwareSensor` is definitely the right way to go if you want to use the native processing blocks.

Like I've mentioned, the code you posted worked for me without throwing exceptions...
Clearly the problem is the `Intel.RealSense.SoftwareSensor` instance itself, can't see why it would be GC'ed\nullified in the snippet above though...

stephen-systemfriend, 2019-04-18:
So according to VS, the only members of the SotfwareSensor that are null are:
AutoExposureSettings
m_callback
m_queue

are any of these necessary?

stephen-systemfriend, 2019-04-18:
Swapped to Unity 2018.2.21 (from 2017) and now it works.
Do you specify what the minimum unity version we should use is anywhere, or is this actually due to 2018 using .net 4.x rather than 3.5 by default?

closed 2019-04-18 by stephen-systemfriend