realsenseai/librealsense #3770: USB, hwmon, fatal and inconsistent errors using D415, D435i and T265 configurations

hperik1, 2019-04-16:
Looks like this _librealsense_ forum is getting more traction wrt review and comments than at the _ROS wrapper_ side. Therefor I like to provide the link to the issue #727 posted at the _wrapper_ side and maybe somebody can help me out ...
[https://github.com/intel-ros/realsense/issues/727](url)

### Short intro
**Multiple issues** are keep popping up using single and mixed configurations of T265, D415 and D435i. Even not stable starting up using same launch files in a row experienced.
Below a summary of received errors seen on multiple PCs and different librealsense, wrapper and even D415 and D435i firmware versions (5.11.4, 5.11.6.200).
I throw some errors I encountered, but more specific data (screen logs) are available for different configurations ... on request.
Hopefully somebody can already indicate if there is an obvious solution or give me a pointer on getting this resolved as for now a reliable system can not be configured ....

ev-mp, 2019-04-16:
@hperik1 hello,
Please see the answers in the original thread.

hperik1, 2019-04-24:
As indicated by @ev-mp this issue is tracked at the ROS wrapper, thus closed here.

closed 2019-04-24 by hperik1