realsenseai/librealsense #3772: ubuntu18.04 libgtk-3-dev unmet dependency.

trustfarm-dev, 2019-04-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  4.15.0-36-generic             | 
| Platform                                 |  |
| SDK Version                            |  v2                           | 
| Language                            |                         | 
| Segment			|                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>
```
The following packages have unmet dependencies:
 librealsense2-utils : Depends: libgtk-3-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

Is there a good way to solve?

ev-mp, 2019-04-16:
@trustfarm-dev , you should run
>sudo apt-get install -f

to rectify the packages discrepancies on the host PC, as suggested, before installing Librealsense.

trustfarm-dev, 2019-04-16:
@ev-mp thanks for reply.

it is not a solution to me.
```
(base) fpga@fpgawork:~/visionprj$ sudo apt-get install -f libgtk-3-dev 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libglib2.0-dev (>= 2.49.4) but it is not going to be installed
                Depends: libgdk-pixbuf2.0-dev (>= 2.30.0) but it is not going to be installed
                Depends: libpango1.0-dev (>= 1.40.5) but it is not going to be installed
                Depends: libatk1.0-dev (>= 2.15.1) but it is not going to be installed
                Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libcairo2-dev (>= 1.14.0) but it is not going to be installed
                Depends: libfontconfig1-dev but it is not going to be installed
                Depends: libx11-dev but it is not going to be installed
                Depends: libxext-dev but it is not going to be installed
                Depends: libxinerama-dev but it is not going to be installed
                Depends: libxi-dev but it is not going to be installed
                Depends: libxrandr-dev but it is not going to be installed
                Depends: libxcursor-dev but it is not going to be installed
                Depends: libxfixes-dev but it is not going to be installed
                Depends: libxcomposite-dev but it is not going to be installed
                Depends: libxdamage-dev but it is not going to be installed
                Depends: libegl1-mesa-dev but it is not going to be installed
                Depends: libxkbcommon-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
(base) fpga@fpgawork:~/visionprj$ 

```

ev-mp, 2019-04-16:
The packages are designed for deployment on official Ubuntu distros with X11 server .Are you running some non-standard/modified image ?  

you can list and inspect the required dependencies with
>apt-rdepends libgtk-3-0

There were some reports on AskUbuntu forum -   like
https://askubuntu.com/questions/741988/unmet-dependencies-while-installing-gtk-3  that suggest that the actual issue could be not the broken packages but that Ubuntu couldn't find these packages as the source list is corrupted/not up to date.

ev-mp, 2019-04-23:
@trustfarm-dev , any luck with the `sources.list` fix?

ev-mp, 2019-05-01:
@trustfarm-dev hi, do you still need assistance with that matter?

closed 2019-05-05 by dorodnic

kaln27, 2024-09-14:
use `sudo aptitude install libgtk-3-dev`