realsenseai/librealsense #3774: T265 gets 'no device connected' in Matlab

jattiogbe, 2019-04-16:
| Camera Model                       | T265 | 
| Firmware Version                   | SDK 2.20| 
| Operating System & Version |   Windows 10|  
| Platform                                 | PC | 
| Language                            |  Matlab                        |  

Hi,

When I use T265 in Matlab 2019a, I can only my program once before I get this error : 

```
>> realsense.test_T265
Error using librealsense_mex
No device connected

Error in realsense.pipeline/start (line 27)
                out = realsense.librealsense_mex('rs2::pipeline', 'start', this.objectHandle);

Error in realsense.test_T265 (line 13)
profile = pipe.start() ;
```

 I need to restart Matlab so that it gets back to normal and works fine. Only unplugging the camera isn't enough although it is enough for detecting it in RealSense Viewer 2.20.

I have no idea why I get that.
Here is my code:

```
pipe = realsense.pipeline(); 
config = realsense.config();

config.enable_stream(realsense.stream.gyro);
config.enable_stream(realsense.stream.accel);
config.enable_stream(realsense.stream.pose);
  
%Start streaming on an arbitrary camera with default settings
profile = pipe.start() ;
 
tic
while toc < 3
      
fs = pipe.wait_for_frames();

frame = fs.first_or_default(realsense.stream.pose);
data = frame.as('pose_frame').get_pose_data();
 
trans= data.translation;
fprintf('Translation : %f \n', trans); 

rota= data.rotation;
fprintf('Rotation : %f \n', rota);
 
 end  % fin while 
```
  
I tried adding pipe.stop() at the end of the script, but it's not better.

jattiogbe, 2019-04-16:
Hi again,

It seems to be working by adding  `clear
`
at the beginning of the script... If anyone knows why Matlab's workspace needs to be cleared. Inever had this problem with D415.

lramati, 2019-04-18:
This might be related to context objects taking ownership of t265s. When
you create a default pipeline, it creates a context, which takes ownership
of the t265. When you run the script a second time, a new pipeline is
created with a new context before the first one is deleted, and as such the
new context can't "see" the t265. Clearing the workspace deletes the old
pipeline and thus it's context, releasing the t265 to be seen by the new
context in the next run

lramati, 2019-04-18:
If my hunch is correct, modifying test_T265 so that it takes a context and passes it to the pipeline constructor could fix your issue, provided you pass the same context into the function each time. Explicitly deleting only the pipeline should also work

jattiogbe, 2019-04-18:
Thanks for your contribution, you must be right.
clearing/ deleting as I wrote before solved my problem.

closed 2019-04-18 by jattiogbe