realsenseai/librealsense #3776: d435i timestamp is error

hitxuhao, 2019-04-17:
src:(imu_callback)
 double frame_time = frame.get_timestamp();
printf("imu_frame_time:%lf,_camera_time_base:%lf,elapsed_camera_ms1:%lf\n",frame_time,_camera_time_base,elapsed_camera_ms);

printf result：
imu_frame_time:688.581144,_camera_time_base:688.581144,elapsed_camera_ms1:0.000000
......
imu_frame_time:4293.008144,_camera_time_base:688.581144,elapsed_camera_ms1:3.604427
imu_frame_time:4294.751144,_camera_time_base:688.581144,elapsed_camera_ms1:3.606170
imu_frame_time:0.544848,_camera_time_base:688.581144,elapsed_camera_ms1:-0.688036

question: why frame_time become small?  note:my device no stop

ev-mp, 2019-04-17:
@hitxuhao hi, this is a known issue #3675.
Scheduled for the upcoming release.

ev-mp, 2019-04-23:
@hitxuhao ,
The fix is available in [v2.21.0](https://github.com/IntelRealSense/librealsense/tree/v2.21.0) release.

hitxuhao, 2019-04-23:
Thanks， I will have a try

| |
徐浩
|
|
邮箱：hitxuhao@163.com
|

签名由 网易邮箱大师 定制

On 04/23/2019 19:03, ev-mp wrote:

@hitxuhao ,
The fix is available in v2.21.0 release.

ev-mp, 2019-05-01:
@hitxuhao , closing this now as the fix has been confirmed in the main thread #3675.

closed 2019-05-01 by ev-mp