realsenseai/librealsense #3779: Cannot start record cause the windows username contains a whitespace

bliet, 2019-04-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win 10   | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.0.570                          | 

### Issue Description
I cannot start the recording if the filename is automaticlly selected. The problem is that my windows user contains a whitespace. The Viewer tried to save the file on a none existing place.

Example: Username is "Max Power". 
Then the selected path is "C:/Users/Max/....bag" instead of "C:/Users/Max Power/....bag"

If I selecet the setting "Ask me every time" I can start the recording.

ev-mp, 2019-04-18:
@bliet hello, thanks for reporting this.
It seem like a UI bug on our side - we'll look into it.

dorodnic, 2019-11-19:
Should be resolved now

closed 2019-11-19 by dorodnic