realsenseai/librealsense #3780: Object tracking C/M/shift algorithm in RS camera

lucianoclugli, 2019-04-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (05.10.13.00) | 
| Operating System & Version | Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.15.0-47-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.20.0                          | 
| Language                            |  Python3.5                          | 
| Segment			| Computer Vision                          | 

### Issue Description
I am trying to get the first frame from a D435 into a object tracking agorithm (C/M/shift), but I'm still not understanding how to extract this single frame.
    * Something similar to [ret, frame = cap.read()], where frame is intended to be the first frame from RS camera; 
    * All parameters for and from pyrealsense2 library work fine for an ordinary streaming;
    * I'm using python3.5;
    * The 'ret' variable is also important, because it implies a parameter for the meanshift location process algorithm (cv2.CamShift);

####################################################

#### The original code is:

import numpy as np
import cv2

cap = cv2.VideoCapture('filename')

##### Take first frame of the video
ret,frame = cap.read()

##### Setup initial location of window
r,h,c,w = 250,90,400,125  # simply hardcoded the values
track_window = (c,r,w,h)

##### Set up the ROI for tracking
roi = frame[r:r+h, c:c+w]
hsv_roi =  cv2.cvtColor(frame, cv2.COLOR_BGR2HSV)
mask = cv2.inRange(hsv_roi, np.array((0., 60.,32.)), np.array((180.,255.,255.)))
roi_hist = cv2.calcHist([hsv_roi],[0],mask,[180],[0,180])
cv2.normalize(roi_hist,roi_hist,0,255,cv2.NORM_MINMAX)

##### Setup the termination criteria, either 10 iteration or move by atleast 1 pt
term_crit = ( cv2.TERM_CRITERIA_EPS | cv2.TERM_CRITERIA_COUNT, 10, 1 )

while(1):
    ret ,frame = cap.read()

    if ret == True:
        hsv = cv2.cvtColor(frame, cv2.COLOR_BGR2HSV)
        dst = cv2.calcBackProject([hsv],[0],roi_hist,[0,180],1)

        ##### Apply meanshift to get the new location
        ret, track_window = cv2.CamShift(dst, track_window, term_crit)

        ##### Draw it on image
        pts = cv2.boxPoints(ret)
        pts = np.int0(pts)
        img2 = cv2.polylines(frame,[pts],True, 255,2)
        cv2.imshow('img2',img2)

        k = cv2.waitKey(60) & 0xff
        if k == 27:
            break
        else:
            cv2.imwrite(chr(k)+".jpg",img2)

    else:
        break

cv2.destroyAllWindows()
cap.release()

####################################################

#### The customized code:

import pyrealsense2 as rs
import numpy as np
import cv2

##### Real Sense - Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)  # 640, 480
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)  # 640, 480

##### Real Sense - Start streaming
pipeline.start(config)

framesC = pipeline.wait_for_frames()
colorframeC = framesC.get_color_frame()
colorimageC = np.asanyarray(colorframeC.get_data())

##### Resize images dimensions (just for compacting)
percentC = 40  # percent of original size
wC = int(colorimageC.shape[1] * percentC / 100)
hC = int(colorimageC.shape[0] * percentC / 100)
dimensionC = (wC, hC)
colorimageC = cv2.resize(colorimageC, dimensionC, interpolation=cv2.INTER_AREA)

framestatic = colorimageC

##### Setup initial location of window
r,h,c,w = 250,90,400,125  # simply hardcoded the values
track_window = (c,r,w,h)

##### Set up the ROI for tracking
roi = framestatic[r:r+h, c:c+w]
hsv_roi =  cv2.cvtColor(framestatic, cv2.COLOR_BGR2HSV)
mask = cv2.inRange(hsv_roi, np.array((0., 60.,32.)), np.array((180.,255.,255.)))
roi_hist = cv2.calcHist([hsv_roi],[0],mask,[180],[0,180])
cv2.normalize(roi_hist,roi_hist,0,255,cv2.NORM_MINMAX)

##### Setup the termination criteria, either 10 iteration or move by atleast 1 pt
term_crit = ( cv2.TERM_CRITERIA_EPS | cv2.TERM_CRITERIA_COUNT, 10, 1 )

while(1):
    framedyn = colorimageC

    hsv = cv2.cvtColor(framedyn, cv2.COLOR_BGR2HSV)
    dst = cv2.calcBackProject([hsv],[0],roi_hist,[0,180],1)

    ##### Apply meanshift to get the new location
    framstatic, track_window = cv2.CamShift(dst, track_window, term_crit)

    ##### Draw it on image
    pts = cv2.boxPoints(framestatic)
    pts = np.int0(pts)
    final = cv2.polylines(framedyn,[pts],True, 255,2)
    cv2.imshow('Object tracking...', final)
    cv2.waitKey(0)

cv2.destroyAllWindows()
cap.release()

####################################################
I appreciate any help, guys...

ev-mp, 2019-04-18:
@lucianoclugli hi,
it is not clear whether this is a question or you the code snippet is not functional, can you be more specific?

One thing I noticed though, is that you seem to skip the check of return value when querying frames (refer to [example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py#L19-L21)):
```python
framesC = pipeline.wait_for_frames()
colorframeC = framesC.get_color_frame()
colorimageC = np.asanyarray(colorframeC.get_data())

# Check that both frames are valid
if not colorframeC or not colorimageC :
    continue
```

lucianoclugli, 2019-04-18:
Thanks, @ev-mp, actually I need some help in how to get/extract the first frame from D435 and apply a bounding box to set a ROI for tracking in camshift algorithm. The first code I uploaded is the original one, with a reading video file (also works for a normal USB cam in real-time), and the last code is my attempt to customize the code with the realsense camera.
Note: even with the value return checking when querying frames, it doesn't work.

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

Hi lucianoclugli,

Wonder if there is any update on this one?

To my understanding, some comments here:
a) there should put a while() loop after pipeline.start() to keep feeding images in.
b) at the end, there should be a pipeline.stop().
c) the while(1) before "framedyn = colorimageC" should be removed. And actually it should be the while() I mentioned above in a).
d) And the last line of "cap.release()" should be removed, as it is not for Realsense case.

Hope this can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Hi lucianoclugli,

Any update on this one?

Thanks!

lucianoclugli, 2019-05-18:
Dear @RealSenseCustomerSupport, I made the suggestions above (and some adjustments) and now the Mean/Cam-shift algorithm works fine with D435... latter I'll post the code.

I appreciate all the help, thanks...

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi lucianoclugli,

Thanks for the updates, and glad to hear that it helped.
Looking forward to your more sharing.

Thanks!

RealSenseCustomerSupport, 2019-05-28:
----------------------------------------------

Hi lucianoclugli,

Wonder if you are ready to share something here?

Thanks!

lucianoclugli, 2019-05-29:
Hi @RealSenseCustomerSupport ... sorry about the delay...

Here is the Cam-shift code implemented with RS D435... one thing is that I cannot manually define a tracking area/region... it starts with a random region and try to track since then... but, at least, the frames' pipeline work fine while streaming.

```python
#Version 2019/May

import pyrealsense2 as rs
import numpy as np
import cv2

# Real Sense - Configure depth and color streams

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30) # 640, 480
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30) # 640, 480

# Real Sense - Start streaming

pipeline.start(config)

framesC = pipeline.wait_for_frames()
colorframeC = framesC.get_color_frame()
colorimageC = np.asanyarray(colorframeC.get_data())

# Resize images dimensions (just for compacting)

percentC = 40 # percent of original size
wC = int(colorimageC.shape[1] * percentC / 100)
hC = int(colorimageC.shape[0] * percentC / 100)
dimensionC = (wC, hC)
colorimageC = cv2.resize(colorimageC, dimensionC, interpolation=cv2.INTER_AREA)

framestatic = colorimageC

# Setup initial location of window

# The numbers below mean the coordinates of a bounding box denoting the EPP
# it is related to how the descriptors present the necessary information for tracking.
# r,h,c,w = 250,90,400,125 # simply hardcoded the values (These values can change)
r,h,c,w = 20,10,10,20
track_window = (c,r,w,h)

# Set up the ROI for tracking

roi = framestatic[r:r+h, c:c+w]
hsv_roi = cv2.cvtColor(framestatic, cv2.COLOR_BGR2HSV)
mask = cv2.inRange(hsv_roi, np.array((0., 60.,32.)), np.array((180.,255.,255.)))
roi_hist = cv2.calcHist([hsv_roi],[0],mask,[180],[0,180])
cv2.normalize(roi_hist,roi_hist,0,255,cv2.NORM_MINMAX)

# Setup the termination criteria, either 10 iteration or move by atleast 1 pt

term_crit = ( cv2.TERM_CRITERIA_EPS | cv2.TERM_CRITERIA_COUNT, 10, 1 )

while (1):
    # Part dedicated to dynamic frames in tracking points:

    framesD = pipeline.wait_for_frames()
    colorframeD = framesD.get_color_frame()
    colorimageD = np.asanyarray(colorframeD.get_data())

    # Resize images dimensions (just for compacting)

    percentD = 40  # percent of original size
    wD = int(colorimageD.shape[1] * percentD / 100)
    hD = int(colorimageD.shape[0] * percentD / 100)
    dimensionD = (wD, hD)
    colorimageD = cv2.resize(colorimageD, dimensionD, interpolation=cv2.INTER_AREA)

    framedyn = colorimageD

    hsv = cv2.cvtColor(framedyn, cv2.COLOR_BGR2HSV)
    dst = cv2.calcBackProject([hsv],[0],roi_hist,[0,180],1)

    ##### Apply meanshift to get the new location
    framestatic, track_window = cv2.CamShift(dst, track_window, term_crit)

    ##### Draw it on image
#    rect = cv2.minAreaRect(framestatic)
    pts = cv2.boxPoints(framestatic)
    pts = np.int0(pts)
    final = cv2.polylines(framedyn,[pts],True, 255,2)
    cv2.imshow('Object tracking...', final)

    key = cv2.waitKey(1) & 0xFF
    if key == ord('q'):
        pipeline.stop()
        break

cv2.destroyAllWindows()
```

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@lucianoclugli Thanks for the sharing! Really great!

closed 2019-06-19 by RealSenseCustomerSupport