realsenseai/librealsense #3781: realsense-viewer does not allow export of localized map from T265

vincentchu, 2019-04-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5448 | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |  n/a                                         | 
| Platform                                 | Macbook  |
| SDK Version                            |  24ce2c                          | 
| Language                            |  n/a                         | 
| Segment			|  n/a                          | 

### Issue Description

I compiled a the `development` branch of the librealsense SDK (at `24ce2c`). Compiling on this branch allowed me to use the T265 on MacOS. I'd like to export the localized map from the T265 using the realsense viewer's "Export Localization map" feature. 

However, clicking on the dialog does nothing. I did a little debugging by trying to see where in the code this failed on; I traced it to this [line](https://github.com/IntelRealSense/librealsense/blob/24ce2c4dea27f292de48097cdb9ee8ec3340644a/common/model-views.cpp#L6495). Looks like `file_dialog_open` never returns true because [this](https://github.com/IntelRealSense/librealsense/blob/24ce2c4dea27f292de48097cdb9ee8ec3340644a/third-party/noc_file_dialog.h#L268) returns a `nullptr`. Will there be any change to this soon?

ev-mp, 2019-04-18:
@vincentchu , unfortunately this is a known issue.
See the [release notes](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#known-issues)

vincentchu, 2019-05-10:
@ev-mp Thanks for that. Is there a working example for retrieving this data from the python bindings? I did try to access this part of the API but get keep getting this error message:

```
11:48:13.603 [0] [E] TrackingManager: Stopping Manager - Cleaned 1 non complete callbacks (onVideoFrame, onPoseFrame, onGyroFrame, onAccelerometerFrame, onControllerframe) - latency is too high!
11:48:13.603 [0] [E] Device-BC00: State [ACTIVE_STATE] got event [ON_ASYNC_START] ==> [ERROR_STATE]
11:48:13.603 [0] [E] Device-BC00: Entered state [ERROR_STATE]
^C^CTraceback (most recent call last):
  File "record.py", line 112, in <module>
    _map = pose_sensor.export_localization_map()
RuntimeError: null pointer passed for argument "buffer"
```

Snippet of code:

```
ctx = rs.context()
pipeline = rs.pipeline(ctx)
pipeline.start()

devices = ctx.query_devices()
pose_sensor = devices[0].first_pose_sensor()

# Pause for a few seconds to allow acquisition of map
_map = pose_sensor.export_localization_map()
print(">>>>", _map)
```

As far as I know, this code should be very similar to your C# example here: https://gist.github.com/ev-mp/8d33fa6054f6e62639eab28da932f5e7

closed 2019-05-13 by RealSenseCustomerSupport