realsenseai/librealsense #3782: How to use librealsense in CLion?

off99555, 2019-04-17:
I am new to C++ ecosystem so maybe this is a question that might be quite general (but it might be specific to librealsense).
I have installed CLion, Cygwin, g++, gcc, make, CMake and it's working and I can do hello world C++ project.
But when I tried to `#include <librealsense2/rs.hpp>`, it doesn't work because it could not find that library.
So my question is how do I include librealsense2? What do I need to do to use use librealsense on my Windows 10?

off99555, 2019-04-18:
I've used `include_directories()` command in the CLion and it seems the project now sees those header files.
The problem is that now I don't know where are the definitions of the librealsense2 binary.

Let me simplify the problem. Suppose I use gcc to compile this, I can use `-I` option to specify the `include` directory of realsense. But what do I do with `-L` option? Where should I point it to?

off99555, 2019-04-18:
I think this should be mentioned somewhere in the document about how to include the  SDK in the code: @dorodnic Do you know the solution? I'm not building from source.

RealSenseCustomerSupport, 2019-04-24:
----------------------------------------------

Hi @of99555,
Which toolchain do you use in CLion?

I use VS2017 toolchain in CLion. And CMakeLists.txt like this,

> include_directories("C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\include")
> link_directories("C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\lib\\x64")
> add_executable(rs-example rs-example.cpp)
> target_link_libraries(rs-example realsense2.lib)

off99555, 2019-04-24:
I have Cygwin toolchain and VS2017 toolchain. I used Cygwin before.
So now I'm using VS2017 toolchain like you.
I still get an error (the program interrupted without returning 0).
@RealSenseCustomerSupport 

Here is the toolchain setting:
[image]
Is the architecture correct? I tried with x86 but it did not work also.

This is CMakeLists.txt file:
[image]

The project name is TestC__ (new project in CLion). Actually, it was TestC++ but the ++ symbol is not allowed in the name.

Here is the main file:
[image]

Here's the output after Run (Shift+F10):
[image]
You will see that the hello world is not even printing.

How do I fix this?

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

There is no this issue in my test project.
I guess, there are different version realsense2.dll in your PC.
You can try copy C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64\realsense2.dll to your "cmake-build-debug" folder.

off99555, 2019-04-26:
Thank you for the suggestion.

At first, there is no `realsense2.dll` file in this folder. Then I copied as you requested:
[image]

The build passes successfully (the previous version was successful as well):
[image]

It fails when I run (with a different message):
[image]

You will see that the hello world message is now printed. But the program doesn't continue. When I press Retry, the program quits without returning zero:
[image]

What could be the next issue?

RealSenseCustomerSupport, 2019-04-27:
----------------------------------------------

Throw an except in your code (Line 9), I think.

There is an article about Windows how to search the .dll files.
https://docs.microsoft.com/en-us/windows/desktop/dlls/dynamic-link-library-search-order

off99555, 2019-04-27:
Hey, it's working now!
[image]

So It seems like whenever there is a runtime error, the program quits without informing me anything useful. (I have also changed `runtime_error` to `exception` as you suggested. I'm not sure if that's what you meant though.)
[image]

And I also cannot use the debug functionality of CLion when I use VS2017 toolchain.
[image]

How do you usually debug the software when there is a dialog without an informative message? How do you show the exception message?

off99555, 2019-04-27:
This is how I'm dealing with the error for now:
[image]

Suggest me if there is a better way.

RealSenseCustomerSupport, 2019-04-27:
----------------------------------------------

Why don't you use Visual Studio to replace CLion on Windows? if you want to use the debugger on Windows.

The exception is not an issue, according to your code.
more information about exception handling.
https://docs.microsoft.com/en-us/windows/desktop/debug/exception-dispatching

off99555, 2019-04-27:
Sure. I will use Visual Studio when I want to debug.
I read what you sent and from what I understand it simply means that if I throw something in the main() function, the program will quit without telling me any exception message?
How do I show the message then? Is the way I did using `cerr` already good enough?

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

How to process an exception, that's your decision.

off99555, 2019-04-28:
Yep. It is my decision. But I need to see a practical example. That's why I asked in your case what you do.
Then I can decide what to do with it.

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

@off99555 For exception handling, https://docs.microsoft.com/en-us/windows/desktop/debug/exception-dispatching should have provided the guidance about this.  Any other questions?

off99555, 2019-05-07:
OK. No other questions. Thank you!

closed 2019-05-07 by off99555