realsenseai/librealsense #3785: D415 showing as D410; showing error ISP update failed

hueri, 2019-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.11.01.100 | 
| Operating System & Version |  Win 10  |                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                        | 
| Language                            |  python / realsense viewer                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hi there,

i have a D415 that has started showing up as a D410.  i've tried on multiple machines, with different cables and ports - and have tried to downgrade the firmware (5.11.6.200, 5.11.1.100 (Latest), 5.10.3, 5.10.6) but have had no success.  

Any ideas would be greatly appreciated - i'm planning to use them next week!

ev-mp, 2019-04-18:
@hueri hello,
- What is the output of `rs-enumerate-devices -m` ?
- How is the device reported in the Device Manager?
[image]
:

hueri, 2019-04-18:
Hi @ev-mp - thanks for responding!

rs-enumerate-devices -m gives a larger output, attached here (
[rs-enumerate-devices_OUTPUT.txt](https://github.com/IntelRealSense/librealsense/files/3094596/rs-enumerate-devices_OUTPUT.txt) but the top-level is here :
 Device info:
    Name                          :     Intel RealSense D410
    Serial Number                 :     840412060862
    Firmware Version              :     05.10.06.00
    Recommended Firmware Version  :     05.10.03.00
    Physical Port                 :     \\?\usb#vid_8086&pid_0ad2&mi_00#6&11c170aa&0&0000#{24e552d7-6523-47f7-a647-d3465bf1f5ca}\global
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0AD2
    Usb Type Descriptor           :     3.2

Device Manager output is here : 

USB\VID_8086&PID_0AD2&MI_00\6&11C170AA&0&0000
[image]

Firmware is currently 5.10.06 after a few downgrade / upgrade attempts.

Thanks for checking in, and hope this helps!!

ev-mp, 2019-04-18:
@hueri , please collect the firmware logs:
- replug the camera
- run `rs-fw-logger > fw_log.txt`
- run `rs-enumerate-devices`
- post the resulted log file

hueri, 2019-04-18:
the fw-logger didn't seem to end, but the txt file looked to be repeating.  attached the outputs here :

[fw_log.zip](https://github.com/IntelRealSense/librealsense/files/3095137/fw_log.zip)
[rs_enumerate_devices_OUTPUT.2.txt](https://github.com/IntelRealSense/librealsense/files/3095139/rs_enumerate_devices_OUTPUT.2.txt)

Thank you!

hueri, 2019-04-19:
@ev-mp - hiya, wanted to check in if any of that was helpful.  Greatly appreciate your support!

ev-mp, 2019-04-22:
@hueri, unfortunately not.
We'll keep you updated

RealSenseCustomerSupport, 2019-04-24:
----------------------------------------------

Hi

The PID shows "0AD2" which means the D410. It could be an issue in USB enumeration or hardware failure in RGB camera connection.

Let's try this first. Please get another computer (Win10) which never connected with D415 camera before, and try roll back firmware to FW5.8.15 from here: https://downloadcenter.intel.com/download/28724/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t

Good luck.

hueri, 2019-04-25:
hi there,

i tried those steps, but it still registers as a D410.

To make matters worse, another camera has started reporting as a D410.

I've tried different computers, different USB cables (length / make), rollbacks and upgrades, and am somewhat at a loss.

I was planning on using them tomorrow, but don't see much benefit at the moment.  bummer!

thank you for your help so far!

.m

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

Hi
Sorry to hear about that. Where are you located? Let me see if there is any FAE over there can help you out locally.

I am trying to figure out how to reproduce this issue here. Can you let me know how this happened?
- Do you remember what was the firmware version when you opened the box?
- Did this issue happen right after you made firmware update? if so, do you remember which firmware version might cause it?
- Was this happening on Windows platform or others?

Last try, FW5.8.15. (ignore it if you had exactly tried)

Thanks

KaistInstitute, 2019-04-26:
Hi, exact same issue of sudden change to D410 camera after upgrading to the recommended most recent firmware for the D415 camera.

1) I first run realsense-viewer with the D415 camera. It notifies that my version is 05.09.02.00 and requires minimum of 05.10.03.00. 
[image]

2) I follow the Download Update link, use the "Intel Realsense D400 Series DFU Tool for Windows" to update the downloaded firmware of "Signed_Image_UVC_5_11_1.0.bin"
[image]

3) I run real-sense viewer again to notice that D415 camera is now registering as D410.
[image]

My Device manager also displays as D410 afterwards. It seems like the UVC firmware v5_11 is incompatible with the D415 camera, (unable to read its RGB color sensor properly?), and hence drops to D410 camera. 

In dire need of clarification and solution, please help. Many of our lab's purchased D415 cameras are now useless D410 cameras.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

Thanks a lot for providing the detailed info.
I think I am able to get this issue reproduced out here. At least I made it happen twice.

The Good News is that, the disappeared RGB camera can be recovered if you do it by following this way:
- Go and use DFU tool to roll back firmware to 5.9.2.0 (or maybe just a different one from your current version); [NOTE: During firmware update, you will see "SR400 Device" being shown under Device Manager (see picture below "1.PNG")];
- They KEY is that, DO NOT disconnect USB3 cable right after firmware update finishes - just leave it there for ~20 seconds .... (You will see "SR400" turns to "D450" (see "2.PNG"), and then turns to D415 (see "3.PNG").
- Now it is safe to unplug the camera if you want.

So every time when you do firmware update, please be patient and leave the camera connected until it is completely enumerated on USB port.

Thanks and Good luck !

Attachment(s):
1.PNG - https://realsensesupport.intel.com/attachments/token/8xK12rFArzuvtJNQiRVF82hiP/?name=1.PNG
2.PNG - https://realsensesupport.intel.com/attachments/token/xAeAn4Tmr4qfVdEpLQXING3fH/?name=2.PNG
3.PNG - https://realsensesupport.intel.com/attachments/token/y21Qq3PsPcgQ2xyIOHrWBV3rW/?name=3.PNG

hueri, 2019-04-28:
Hi there - thanks for following up!  No luck for me so far.  

i'm not able to see those attachments for some reason (tells me i'm not authorized to see the file).  The device had been unplugged previously, though not during a firmware update.  I had been changing ports to find the optimal setup.  

That said, i believe i'm following the above recovery steps properly.  In Device Manager, the device displays as an SR400 under Universal Serial Bus devices during installation, then updates to a D450... and then resolves to a D410 under Cameras.  

Thanks much!

.m

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

Have you successfully rolled back firmware to FW5.9.2?
if YES, then do same thing update it to FW5.11.1. or newer one.

Thanks

hueri, 2019-04-28:
Hi - yes, i'd rolled back to FW5.9.2.0, then updated 5.11.1 as well as rolling back to 5.9.2 and updating to 5.11.6.200.  

Same results - cycling through from SR400 to D450 then finally resolving to a D410.  
i'm using DFU tool Version 2.5.2.0

Thanks back!

hueri, 2019-04-28:
To clarify my steps, because i may be doing something wrong : 
 - realsense viewer is not open
 - connect D415 via USB-C to USB-A cable to USB3 SS port
 - run intel-realsense-dfu.exe
 - option 1 to update camera device
   - shows the D410 there with current firmware
 - pass path to signed_image.bin for 5.9.2 to roll back, then 5.11.1 to update
 - confirm in Device Manager and Realsense Viewer that the device is still reporting as a D410

Thanks, and hope we can get to the bottom of it!

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

Thanks for the details.

- after rolling back to FW5.9.2, wait 20 sec......then Unplug the camera.
- Replug camera, then DFU it to FW5.11.1. and wait 20 sec ......
- still D410?

KaistInstitute, 2019-04-29:
Thanks for providing a clear and easy step to resolve this problem @RealSense-Customer-Engineering .
As you suggested, I rolled back to FW5.9.2.0 and waited until I observed the Device Manager display from D410 to D450 to D415 (within 20 seconds).
I then updated to the most recent FW5.11.6.200 and then verified with realsense-viewer that all my D415 are being recognized correctly.

Thank you for your prompt help!

hueri, 2019-04-29:
Worked for one!  The other -- the first one that i'd had issues with -- hasn't worked after a number of attempts.  

Now, when trying to downgrade to FW5_9_2, i get one of these two errors:
1)
...
Downloaded 1024, bytes remaining 1444636
Downloaded 1024, bytes remaining 1443612
Downloaded 1024, bytes remaining 1442588
WaitForState(DFU_STATE::dfuDNLOAD_IDLE, 5000) Failed with status-2147467259
DownloadImage(reinterpret_cast<const BYTE *>(firmwareImage), imageLength) Failed with status-2147467259
Press any key to continue . . .

2)
Updating firmware to c:\users\hueri\downloads\signed_image_uvc_5_9_2_0.bin
No DFU device was connected
Failed to burn firmware
Press any key to continue . . .

Not sure how to continue on that one...
Thanks!!

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

Is this device still able to be seen in Device Manager?
If it is stuck during mid of FW update, probably you can recover it by choosing option 3 -"Recover device stuck in Upgrade Mode".

Good luck.

hueri, 2019-04-29:
Hi - yes, it still registers as a D410 in Device Manager.  It usually fails, then restarts.  Eventually it shows back up as a D410 again.
i’ve tried option 3 - but it says nothing is stuck in update mode.

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

1) Not sure if this works - Can you try same way by using a self-powered USB3 Hub between your computer and the D415?
2) If still not work, can you run Dynamic Calibration Tool? (here is User Guide if this is 1st time for you.)

Thanks

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

Since Hyperlink doesn't work. Let me copy the address here:
https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API
https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Thanks

icarpis, 2019-05-02:
Hello @hueri, please try the following on the unit that still recognize as D410.
1. Connect the camera and open the [rs-terminal](https://github.com/IntelRealSense/librealsense/tree/master/tools/terminal) tool on Windows or Linux OS by this line ```rs-terminal -d 0```
2. Paste the following byte array and click on the return key
```14 00 ab cd 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00```
3. Wait 20 sec' and then unplug & replug the unit

Please update..

hueri, 2019-05-03:
wow - that looks crazy!  i need to compile rs-terminal, but will try asap (and report back)!

the previous suggestion of trying through a powered hub didn't yield anything different.

thanks!

hueri, 2019-05-04:
Pasted that byte array, and it returned : 
Device ID 0 has loaded.
#>14 00 ab cd 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
f6 ff ff ff

Waited 22 seconds, unplugged, waited 22 seconds, plugged it back in... same as before.  D410, and doesn't accept firmware updates / rollbacks.  

i'm looking into the dynamic calibration tool now.  

Thanks still and again!

hueri, 2019-05-04:
tried the dynamic calibration tool, which uploaded results to the device but didn't seem to change anything else.  Is this something you'd want me to send to you, or is there someone in the bay area that i could meet more locally?  
It's not critical to be solved quickly anymore, but i would like to get it working again.  

Thanks!

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi
If you are based in Bay Area, that's perfect for me.
Would you please drop off the camera to Santa Clara headquarter ? If yes, let me PM you.

Thanks

hueri, 2019-05-06:
i can't hand deliver it to Santa Clara, but happy to ship it to you :)

PM me the address and I'll send it on.

thanks!

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi Hueri

Not sure how people do PM through GitHub, I don't see any contact info from your profile.

Thanks

hueri, 2019-05-06:
Hi - sorry.  i've updated my profile :)  
It's hueristic@hotmail.com...

Thanks!

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Email has been sent out to your personal email address. Please check.

arnoldktph, 2019-05-17:
hi, please help advise the  pn of the cable that conncet to d415 to pc..

usb 2.0 / usb 3.1 gen 1

arnoldktph, 2019-05-17:
do you have picture too? appreciate your help..

closed 2019-05-17 by RealSenseCustomerSupport

roelofvandijk, 2022-10-03:
We encountered this issue on a D415 OEM module today. After a firmware update to 5.13.0.50 the module showed up as D410. Through repeated firmware downgrades and upgrades, we were able to recover it. Still concerning...

Serial is 039422060875.
D4 Board serial is 047223021509.