realsenseai/librealsense #3786: D435 depth breaking up at 60fps #3724

Repulser, 2019-04-18:
### Issue Description
Hey, Ive recently opened issue #3724 and resolved it by updating firmware, but it seems to be happening again on latest (5.11.1.100), Is there anything I can do to resolve this?

ev-mp, 2019-04-22:
@Repulser hello, 
There seem to be an inconsistency on the Realsense [Firmware download page](https://downloadcenter.intel.com/download/28568/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t) - v5.11.1.100 is marked as the newest even though 5.11.6.200 is listed on the same page.

Please upgrade to v5.11.6.200 and update.

RealSenseCustomerSupport, 2019-04-24:
----------------------------------------------

Hi Repulser,

Any update on this one, given what ev-mp commented?
You can find the latest FW 5.11.6.200 here: https://downloadcenter.intel.com/download/28724/Latest-Firmware-for-Intel-RealSense-D400-Product-Family

Thanks!

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi Repulser,

If nothing else is needed, this one will be closed.

Thanks!

closed 2019-05-16 by RealSenseCustomerSupport