realsenseai/librealsense #3787: Improving pointclouds quality with additionnal hardware

AntoineWefit, 2019-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | Latest | 
| Operating System & Version |   Win (10) / Linux  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC |
| SDK Version                            |  Latest                         | 
| Language                            |                            | 
| Segment			|                         | 

### Looking for suggestions about equipment to improve quality of capture

I can't give too much info about my setup but I want to improve the quality of the .ply out of the D415. I'm working at really close range (~300 mm), and yes I've tuned the parameters of the cameras.
So far, I've read this [doc](https://dev.intelrealsense.com/docs) (which is really good btw) and some solution stood out : 

- Adding an additional projector to increase the pattern density/dot number
- Adding an IR filter (bandpass/longpass) on the stereo sensors

As I currently have 2 D415, I tried the 1st solution simply by enabling both IR emitter in the field of 1 camera. The quality improvement was not significant, so I wonder if this solution is only useful to increase range ([see chapter 4](https://www.intelrealsense.com/wp-content/uploads/2019/03/WhitePaper_on_Projectors_for_RealSense_D4xx_1.0.pdf)) ?

I would also like to know if anyone tried to add an IR filter to the camera, in order to get rid of some annoying ambient lighting (basic office lights for example) ? What should I pick then, a bandpass or a longpass filter ? I'm worried about either filtering too much or not enough wavelength, and thus ruining the data.

Suggestions are welcomed !

AntoineWefit, 2019-04-25:
Any update on this ?

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

Hi AntoineWefit,
what is the key depth quality you are going to improved, ex :depth accuracy  or fill rate...
(1) use the IR emitter can help to increase the features on object especially for plain surface (such as flat white wall has no any texture) by projecting those invisible dots onto those object's surface
     -so you can see significant difference  (especially for the fill rate) when the surface of object you are going to measure having  more features (dots)  by adding those IR dots

(2) the IR emitter (projector) has nominal wavelength at 850nm
 if you want to have IR cut filter for it, you can use 830-880nm bandpass IR filter

(3) the thinner the lens cover, the small change on the optical characteristic.
if thicker lens is required and has significant change on optical characteristic..
re-calibration may required..

(4) Librealsense has some SW filter (post processing) that can help improve the depth quality as well

AntoineWefit, 2019-04-29:
Thank you for your suggestions !

I'm basically trying to get the maximum overall quality of acquisition (so both depth accuracy and fill rate are important, as well as sharp and clean object borders).

(1) OK I got it !

(2) That's what I thought. **What I want to know is if some tests have been made with such a setup, and if it is indeed worth the try ?**

(3) Thanks for the clarification.

(4) Yes, I'm already using those. From a software side, I believe that I'm pretty much out of possibilities to improve mesh quality ; hence looking for hardware-related improvements :)

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

Hi AntoineWefit,
for (2),  you can test the depth in a complete dark room and turn project on/adjust its power
to see if it can help in your case.
best

AntoineWefit, 2019-04-29:
Will try, thank you for the suggestion !

I'm going to let this issue opened a few days in case someone else has other advice.

AntoineWefit, 2019-04-30:
I've just tested your latest suggestion @RealSenseCustomerSupport (isolating the system from ambient lighting to "simulate" a IR/pass filter) and their definitely was a slight improvement of the overall quality of the mesh. As it's possible to find relatively cheap filters, it might worth the try and buy a couple...

Talking about filters, even if I doubt it, I was wondering if their was a way to implement such a filter from a software side ? My guess is that I would have to apply some kind of wizardry (will do more investigation on that later) right before post-processing & saving the frames in my code. But I have no idea :

1. If it is indeed possible with the current version of the SDK ?
2. If I need to use/call an external software/code to process the frames within the main control code which is (obviously) based on librealsense, or can I do everything "internally" ?
3. Last but not least, how to do it ?

RealSenseCustomerSupport, 2019-05-02:
----------------------------------------------

Hi AntoineWefit,
(1) D415 has a VPU, The primary function of Vision Procesor unit D4 is to perform depth stereo vision
processing. The D4 on D4 Card communicates to
the host processor (ex. PC)  through USB3.0 and receives camera sensor raw data from the stereo
depth module (camera head)
that means, the filter you plan to implement will need to be perform by this VPU before it calculate /output the depth to host (PC), i.e, this will require firmware change. the existing SDK should not able to implement it.

AntoineWefit, 2019-05-03:
Thank you for the explanation ! 
I'll close this issue in a few days in case someone has an other advice.

closed 2019-05-13 by AntoineWefit