realsenseai/librealsense #3788: RGB 3D (Point Cloud) is Black

Firifire, 2019-04-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | Tested with 5.11.1.100 & 5.11.6.200 | 
| Operating System & Version |  Win 10 (Build 17763)  |               | 
| Platform                                 | Tested on two different PC |
| Intel Real Sense Viewer  & Unity Package |  2.20.0 to 2.16.0 | 
| Language                            |  unity   | 

### Issue Description
When I run the 3D option in the intel real sense viewer with the Texture input being RGB or I try the RBG Point Cloud in the Unity Sample and I get all the pixels in black. The issue to same as #3213 but resetting calibration by "Intel.Realsense.CustomRW.exe -g" did not work

Only Intel Real Sense Viewer 2.19 gave SCP overflow error when changing the texture input to RBG.

On a PC i tried  2 months ago (I dont have access to it anymore) it was working fine with intel real sense viewer 2.19 & 2.17.

ev-mp, 2019-04-18:
@Firifire , can you plot the output for `rs-enumerate-devices -c` ?

Firifire, 2019-04-18:
[Output.txt](https://github.com/IntelRealSense/librealsense/files/3093613/Output.txt)

The above is the output from "rs-enumerate-devices -c"

ev-mp, 2019-04-18:
@Firifire , the Depth-Color extrinsic appears broken, similar to #3474.
>Extrinsic from "Depth"    To      "Color" :
 Rotation Matrix:
   0   0   0
   0   0   0
   0   0   0

 Translation Vector: 0  0  0

Can you recall which FW were you using prior to this? Also what tool/version did you use to upgrade the FW.
Thanks

Firifire, 2019-04-18:
I believe I was originally on FW 5.11.4 and then updated it to 5.11.6.200 and then downgraded to 5.11.1.100. 
I used the "Windows DFU Tool for Intel® RealSense™ D400 Product Family" Ver 2.5.2.0 to update the FW.

[image]
Well it seems like i am unable to run the Dynamic Calibration.

ev-mp, 2019-04-18:
@Firifire , thanks for the info. We'll pass it for review.

Firifire, 2019-04-20:
Is there a mean to fixing this issue?

ev-mp, 2019-04-22:
@Firifire  hello, the update from Engineering Team is that this issue was reproduced on devices that underwent FW downgrade operation.
The good news are that downgrade the calibration tables are actually remain intact, and hence the situation can be reversed by upgrading the back FW to the new version.

Note that this is still after review and there is no full documentation on recovery, nevertheless - 
please perform FW upgrade from the currently installed v5.11.1.100 (according to attached report) to the latest available v5.11.6.200,  rerun `rs-enumerate-devices -c` and update.

Firifire, 2019-04-22:
Do note that I downgraded from v5.11.6.200 only after facing this issue.

I have upgraded back to v5.11.6.200 and issue persists with the `Translation Vector: 0 0 0 For the Extrinsic from "Depth" To "Color"`

I am still unable to run Dynamic Calibration as it has the same issue

ev-mp, 2019-04-24:
@Firifire he,
There is an ongoing effort to find and resolve the issue. Therefore I will ask you to collect the following:
1. Disconnect any additional Realsense devices (if any) except for the faulty unit.
2. Run `rs-terminal` tool:
> rs-terminal -d 0

3. Copy&paste and run these strings in the terminal one by one:
>14 00 ab cd 15 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
14 00 ab cd 1d 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
14 00 ab cd 1d 00 00 00 20 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
14 00 ab cd 09 00 00 00 80 bc 17 00 00 01 00 00 00 00 00 00 00 00 00 00
14 00 ab cd 09 00 00 00 80 cc 17 00 00 01 00 00 00 00 00 00 00 00 00 00

The typical output for each command is
>14 00 ab cd 15 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Offset: 0x00 => 01 00 20 00 F0 00 00 00 FF FF FF FF FC AE 54 43
Offset: 0x10 => BB B8 B7 3F 00 00 00 00 40 BE 5C BC 00 00 00 00
Offset: 0x20 => F3 D6 22 40 00 B8 17 BC 00 00 00 00 00 00 00 00
Offset: 0x30 => 00 00 80 3F 00 00 00 00 00 00 00 00 00 00 00 00
Offset: 0x40 => 00 00 00 00 00 00 00 00 AD 2E D1 3B 64 63 9B 3B
Offset: 0x50 => 3A 41 CE BB 76 AD 0D 42 D2 7D 92 BF B0 15 4C 3D
Offset: 0x60 => BD B2 B7 3F C4 BF CA 3C 30 A6 1E BC 27 5B 4B C2
Offset: 0x70 => 08 15 33 BD C8 D0 22 40 D9 9F 1B 3C AF 55 3A 40
Offset: 0x80 => C0 B2 24 BB BF 08 F3 BB FE FD 7F 3F B0 15 4C 3D
Offset: 0x90 => 80 07 38 04 00 00 00 00 00 00 00 00 00 00 00 00
Offset: 0xa0 => 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Offset: 0xb0 => 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Offset: 0xc0 => 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Offset: 0xd0 => 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Offset: 0xe0 => 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Offset: 0xf0 => 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Firifire, 2019-04-24:
[Terminal Output.txt](https://github.com/IntelRealSense/librealsense/files/3111978/Terminal.Output.txt)

I have attached the Terminals Output

bendzo, 2019-04-25:
Hello,
I'm experiencing exactly the same issue.
I'm attaching terminal output from my Realsense device:

> #>14 00 ab cd 15 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 
> 15 00 00 00 
> 01 00 20 00 f0 00 00 00 ff ff ff ff fc ae 54 43 
> bb b8 b7 3f 00 00 00 00 40 be 5c bc 00 00 00 00 
> f3 d6 22 40 00 b8 17 bc 00 00 00 00 00 00 00 00 
> 00 00 80 3f 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 ad 2e d1 3b 64 63 9b 3b 
> 3a 41 ce bb 76 ad 0d 42 d2 7d 92 bf b0 15 4c 3d 
> bd b2 b7 3f c4 bf ca 3c 30 a6 1e bc 27 5b 4b c2 
> 08 15 33 bd c8 d0 22 40 d9 9f 1b 3c af 55 3a 40 
> c0 b2 24 bb bf 08 f3 bb fe fd 7f 3f b0 15 4c 3d 
> 80 07 38 04 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 
> 
> #>14 00 ab cd 1d 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 
> 1d 00 00 00 
> 01 00 13 00 f0 00 00 00 ff ff ff ff fc ae 54 43 
> bb b8 b7 3f 00 00 00 00 40 be 5c bc 00 00 00 00 
> f3 d6 22 40 00 b8 17 bc 00 00 00 00 00 00 00 00 
> 00 00 80 3f 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 ad 2e d1 3b 64 63 9b 3b 
> 3a 41 ce bb 76 ad 0d 42 d2 7d 92 bf b0 15 4c 3d 
> bd b2 b7 3f c4 bf ca 3c 30 a6 1e bc 27 5b 4b c2 
> 08 15 33 bd c8 d0 22 40 d9 9f 1b 3c af 55 3a 40 
> c0 b2 24 bb bf 08 f3 bb fe fd 7f 3f b0 15 4c 3d 
> 80 07 38 04 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 
> 
> #>14 00 ab cd 1d 00 00 00 20 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
> 
> 1d 00 00 00 
> 01 00 15 00 f0 00 00 00 ff ff ff ff fc ae 54 43 
> bb b8 b7 3f 00 00 00 00 40 be 5c bc 00 00 00 00 
> f3 d6 22 40 00 b8 17 bc 00 00 00 00 00 00 00 00 
> 00 00 80 3f 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 ad 2e d1 3b 64 63 9b 3b 
> 3a 41 ce bb 76 ad 0d 42 d2 7d 92 bf b0 15 4c 3d 
> bd b2 b7 3f c4 bf ca 3c 30 a6 1e bc 27 5b 4b c2 
> 08 15 33 bd c8 d0 22 40 d9 9f 1b 3c af 55 3a 40 
> c0 b2 24 bb bf 08 f3 bb fe fd 7f 3f b0 15 4c 3d 
> 80 07 38 04 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 
> 
> #>14 00 ab cd 09 00 00 00 80 bc 17 00 00 01 00 00 00 00 00 00 00 00 00 00
> 
> 09 00 00 00 
> 01 00 13 00 f0 00 00 00 ff ff ff ff fc ae 54 43 
> bb b8 b7 3f 00 00 00 00 40 be 5c bc 00 00 00 00 
> f3 d6 22 40 00 b8 17 bc 00 00 00 00 00 00 00 00 
> 00 00 80 3f 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 ad 2e d1 3b 64 63 9b 3b 
> 3a 41 ce bb 76 ad 0d 42 d2 7d 92 bf b0 15 4c 3d 
> bd b2 b7 3f c4 bf ca 3c 30 a6 1e bc 27 5b 4b c2 
> 08 15 33 bd c8 d0 22 40 d9 9f 1b 3c af 55 3a 40 
> c0 b2 24 bb bf 08 f3 bb fe fd 7f 3f b0 15 4c 3d 
> 80 07 38 04 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 
> 
> #>14 00 ab cd 09 00 00 00 80 cc 17 00 00 01 00 00 00 00 00 00 00 00 00 00
> 
> 09 00 00 00 
> 01 00 15 00 f0 00 00 00 ff ff ff ff fc ae 54 43 
> bb b8 b7 3f 00 00 00 00 40 be 5c bc 00 00 00 00 
> f3 d6 22 40 00 b8 17 bc 00 00 00 00 00 00 00 00 
> 00 00 80 3f 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 ad 2e d1 3b 64 63 9b 3b 
> 3a 41 ce bb 76 ad 0d 42 d2 7d 92 bf b0 15 4c 3d 
> bd b2 b7 3f c4 bf ca 3c 30 a6 1e bc 27 5b 4b c2 
> 08 15 33 bd c8 d0 22 40 d9 9f 1b 3c af 55 3a 40 
> c0 b2 24 bb bf 08 f3 bb fe fd 7f 3f b0 15 4c 3d 
> 80 07 38 04 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

I hope it will be useful, it's a bit frustrating issue.
Thanks!
Jacob

zhuzunjie17, 2019-04-27:
I have the same issue after I upgrade FW from v5.11.1.100 to the newest v5.11.6.200.

Firifire, 2019-05-01:
@ev-mp is there any update?

ev-mp, 2019-05-01:
@Firifire hi, thank you for the patience.
The engineering teams were able to identify and reproduce the issue and are working on a solution, preferably one that can be deployed will minimal impact for the users.
It will probably take another week before we'll have a working beta for testing.

Firifire, 2019-05-20:
@ev-mp Do we simply update the firmware and try the calibration tool? or is there some steps that needs to be followed?

ev-mp, 2019-05-20:
@Firifire hello,
The calibration map rectification is completely transparent.
There is no need for firmware update or any other user action other than running any demo from v2.22 (e.g. `rs-enumerate-devices -c`, and checking the results.

ev-mp, 2019-05-22:
@Firifire hello, any luck with v2.22?

Firifire, 2019-05-22:
I did as mentioned running `rs-enumerate-devices -c` and now depth to color extrinsics are

```
Extrinsic from "Depth"    To      "Color" :  
N/A
```

Trying to view RGB point cloud now shows nothing instead of a Black Cloud Point

Attempting Dynamic Calibration Still Does Not Work and yield the same error as before.

bendzo, 2019-05-22:
> I did as mentioned running `rs-enumerate-devices -c` and now depth to color extrinsics are
>
> ```
> Extrinsic from "Depth"    To      "Color" :
> N/A
> ```
>
> Trying to view RGB point cloud now shows [...]

Situation is exactly the same for me.

I also tried FW 5_11_4_0 and restoring factory (golden) calibration tables, but it had no effect.

taletovic, 2019-05-23:
@Firifire @bendzo Hi,
I have the same problems which I posted here:

[https://github.com/IntelRealSense/librealsense/issues/3949#issuecomment-494346126](url)

Maybe you could do the same in order to help the developers to reproduce the problem?
Thanks.

Firifire, 2019-05-23:
Well i get back the black cloud point with the 2.21 Realsesne Viewer.

@ev-mp How do I request a replacement. (Ordered on the Intel Website). I do have much time left to work on my university project.

ev-mp, 2019-05-23:
@Firifire hello,
Please create and share the camera dump generated with the attached python script 
[realsense_flash_debug.txt](https://github.com/IntelRealSense/librealsense/files/3211870/realsense_flash_debug.txt)

Instructions:
1. Install python 2.7/3.6
2. Install realsense python wrapper
`pip install pyrealsense2`
3. Rename the attached script extension: `realsense_flash_debug.txt` into `realsense_flash_debug.py`
4. Run the script
`python realsense_flash_debug.py`
The script will run for ~2min and generate device FW dump `realsense_debug_dump.bin`
4. Attach the resulted file

Firifire, 2019-05-23:
[realsense_debug_dump.zip](https://github.com/IntelRealSense/librealsense/files/3212280/realsense_debug_dump.zip)

Attached is the generated FW dump

nekovfra, 2019-05-23:
After launching RS Viewer, received following console output:

23/05 19:39:36,239 WARNING [139654146570816] (synthetic-stream-gl.cpp:79) Initializing rendering, GLSL=0
 23/05 19:39:36,239 WARNING [139654146570816] (synthetic-stream-gl.cpp:88)  0 GPU objects initialized
 23/05 19:39:36,810 WARNING [139653766891264] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).
 23/05 19:39:36,810 WARNING [139653766891264] (ds5-factory.cpp:617) invalid RGB extrinsic was identified, recovery routine was invoked
 23/05 19:39:36,860 WARNING [139653766891264] (ds5-factory.cpp:609) Restore from gold_sector failed
 23/05 19:39:36,860 WARNING [139653766891264] (ds5-factory.cpp:624) RGB extrinsic recovery - Gold calibration is invalid, restore from an alternative sector
 23/05 19:39:36,911 WARNING [139653766891264] (ds5-factory.cpp:596) Restoring RGB Extrinsic from address17c49c failed
 23/05 19:39:36,911 WARNING [139653766891264] (ds5-factory.cpp:628) RGB extrinsic recovery - Gold calibration from address 17c49c is invalid, restore from an alternative sector
 23/05 19:39:36,962 WARNING [139653766891264] (ds5-factory.cpp:596) Restoring RGB Extrinsic from address17b49c failed
 23/05 19:39:36,962 WARNING [139653766891264] (ds5-factory.cpp:632) RGB extrinsic recovery - Dynamic calibration from address 17b49c is invalid, recovery routine failed
 23/05 19:39:37,078 WARNING [139653766891264] (sensor.cpp:365) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]

diyuhuodong, 2019-06-02:
I have the same issue after I upgrade FW to v5.11.6.200
Attach the resulted file
[realsense_debug_dump.zip](https://github.com/IntelRealSense/librealsense/files/3244886/realsense_debug_dump.zip)

diyuhuodong, 2019-06-02:
@Firifire  Do you see a way of getting round the problem?  @ev-mp   how to restore  calibration tables?

bendzo, 2019-06-04:
I'm attaching FW dump, issue present after the upgrade to v5.11.6.200 and still remaining after the downgrade to v5.11.4.0 and restoring golden factory calibration.
[realsense_debug_dump.zip](https://github.com/IntelRealSense/librealsense/files/3254288/realsense_debug_dump.zip)

Firifire, 2019-06-12:
@diyuhuodong Not yet. I am doing things from software side to get some of the algorithms to work but performance is terrible.

@ev-mp is there any hopes on getting this fixed?

Firifire, 2019-06-22:
@ev-mp is there no update regarding this?

Firifire, 2019-08-27:
Tried Doing FW Update 

` 27/08 00:48:20,240 ERROR [12044] (ds5-factory.cpp:696) RGB Extrinsic recovery routine failed

updating device:
Name: Intel RealSense D435I, serial number: --, ASIC serial number: --, firmware version: 05.11.06.250
5.11.6.250, USB type: 3.2

firmware update started

firmware update progress: 100[%]

firmware update done
 27/08 00:48:53,577 ERROR [1892] (ds5-factory.cpp:696) RGB Extrinsic recovery routine failed
 27/08 00:48:54,101 ERROR [12044] (ds5-factory.cpp:696) RGB Extrinsic recovery routine failed

device -- successfully updated to FW: 05.11.11.100
5.11.11.100`

ev-mp, 2019-12-22:
@Firifire, sorry for the prolonged delay,
Can you please update on the issue status and whether you've managed to upgrade the FW to version 5.11.6.250+ ?

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Hi @Firifire,

Do you continue to experience issues with this matter?Please let us know if you need any further assistance with it.

Thank you!

DevastF, 2020-01-09:
Issue persists with latest FW.

[image]

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @DevastF,

Did the camera (from your last post) undergo a firmware downgrading before extrinsic recovery issue has occurred?

Thank you!

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@DevastF,

Do you still need help with this issue?

DevastF, 2020-02-10:
> 
> 
> Hi @DevastF,
> 
> Did the camera (from your last post) undergo a firmware downgrading before extrinsic recovery issue has occurred?
> 
> Thank you!

Yes, I have tried downgrading the FW version when the issue first occured.

DevastF, 2020-02-10:
> 
> 
> @DevastF,
> 
> Do you still need help with this issue?

I have sent the camera for replacement, since the issue persisted for so long.

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

@DevastF,

I am closing the issue since the camera will be replaced. Please do not hesitate to open a new issue if you have additional questions.

Thank you!

closed 2020-02-10 by RealSenseCustomerSupport