realsenseai/librealsense #3789: align_depth_to_other_sse performance issue

challenzhou, 2019-04-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | {05.10.13.00}
| Operating System & Version |Linux (Ubuntu 18)| 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { 2.16.5 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
We encounter critical performance issue using librealsense with SSE optimization which supposed to have better performance than default openmp version, could you help optimize "align_depth_to_other_sse" function implementation so that we can archive 

Scenarios: We aimed to align depth to raw_image, below is the detail configuration: 
depth stream is enabled - width: 640, height: 480, fps: 30
infra1 stream is enabled - width: 640, height: 480, fps: 30
infra2 stream is enabled - width: 640, height: 480, fps: 30
color stream is enabled - width: 640, height: 480, fps: 30

The triage list as below:
a)  Default align function goes SSE version (librealsense/src/proc/sse/sse-align.cpp)  
     align_depth_to_others
     Each depth(640x480) frame align to Raw image(640x480) cost 46ms~53ms.

b)  Force align function goes OpenMP version.(librealsense/src/proc/align.cpp)
      align_images
      Each depth(640x480) frame align to Raw image(640x480) cost 26ms~31ms.

The SSE version made us can not archive 30fps which is required in most of Robotics scenarios,  we also did some investigation on the SSE implementation in  librealsense/src/proc/sse/sse-align.cpp, we have several suggestions as below for the optimization, 
1) The cost mostly due to the 3 iteration loops on depth image, 2 loops from "get_texture_map_sse" and 1 from "move_depth_to_other", is it possible to merge all of this together as 1 single loop  function or at least 1 single loop?
2) could we add "_mm_prefetch" in SSE implementation to save latency for each vector computation?  

Please  feel free to let me know if you need more details.

ev-mp, 2019-04-18:
@challenzhou , are those numbers collected with Release build ?
Also -  which CPU are you running ?

challenzhou, 2019-04-19:
@ev-mp , the data collected with debug version when i tried to analyze the performance drop issue.
The CPU is "Intel(R) Core(TM) i7-6770HQ".

challenzhou, 2019-04-26:
May i know if you have any comment on this?

dorodnic, 2019-04-28:
> the data collected with debug version when i tried to analyze the performance drop issue.

Did you check performance using the release build?

RealSenseCustomerSupport, 2019-05-14:
----------------------------------------------

Hi challenzhou,

Wonder if there is any update on this one, regarding to what dorodnic suggested?
Or do you still need help?

Thanks!

challen-zhou, 2019-05-17:
we tried with release version, got the similar issue.

dorodnic, 2019-05-17:
Please share a code snippet. This is inconsistent with our measurements. 
Also, one common mistake is creating align object on every frame, instead of once, please check

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

challen-zhou, 2019-05-17:
Codes as below:
"    rs2::align align(RS2_STREAM_COLOR);
    _aligned_frameset = depth_frame.apply_filter(align);
    rs2::depth_frame aligned_depth = _aligned_frameset.get_depth_frame();
", 
"_aligned_frameset" created only for once, is there any conflict with your design? 
It would be highly appreciated if you can share the comparison of measurements for SSE/Non-SSE implementations as reference!

dorodnic, 2019-05-17:
The first line (rs2::align ...) needs to be out of the loop. If you search pull requests for Sse / abc, each performance optimization was published with a set of measurements. I’ll try to look it up later today, but generally you should expect less then 30 ms for apply filter

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

challen-zhou, 2019-05-20:
Thanks for the information! 
The first line is a declaration, would you help explain how it impact on performance so that we can avoid similar codes further?  Whatever, we can have a try later.

dorodnic, 2019-06-03:
Creating align object will cause significant amount of initialization work, such as fetching device calibration for depth to rgb mapping and other hardware parameters. Once fetched, these parameters remain cached inside rs2::align object, but creating new one will make it do it over again.
Hope this helped. Can you share if now performance is reasonable for your use-case?

RealSenseCustomerSupport, 2019-06-14:
----------------------------------------------

@challen-zhou Any other questions for this issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@challen-zhou Any update from your side? Will close the issue if no other questions. Thanks!

challen-zhou, 2019-06-24:
@RealSense-Customer-Engineering , thanks very much for your kindly support! 
We are working on this, but since of some other urgent tasks, progress on this verification would be very slow, suggest to close this issue, we will ask for your support with new ticket if any further question.

closed 2019-06-24 by dorodnic