realsenseai/librealsense #3790: When I use imgui-glfw, I met some linker errors?

coldcoda, 2019-04-18:
Hello @seebq , @wspr , @martinus , @chachi , @dorodnic 

When build a new project and add the existing property sheet ( glfw-imgui.props ,intel.realsense.props ) to my project, but there are some linker errors when I run it.

[image]

Errors:
[image]

But I have added the glfw3.lib to my project.
[image]

It really really don't know how to solve it.

Please give me some tips.

Thanks a lot!!!

martinus, 2019-04-18:
Please stop pinging people

coldcoda, 2019-04-18:
I'm sorry for the trouble this has caused you. Cause I really want to solve it, and it has troubled me for a long time!
I'll never do that again. 
I solemnly apologize to you. Sorry for that!!

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

The implementation of imgui functions is saved in this folder.
C:\Program Files (x86)\Intel RealSense SDK 2.0\third-party\glfw-imgui

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@coldcoda Did you get it through? Looking forward to your udpate. Thanks!

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@coldcoda Any update? Thanks!

coldcoda, 2019-05-20:
Yes, I have solved it. Thanks a lot!!

closed 2019-06-06 by RealSenseCustomerSupport