realsenseai/librealsense #3793: How to deal with the edge in the pointcloud?

coldcoda, 2019-04-18:
When I run the example " pointcloud ", there is a black edge in the pointcloud .
And how to deal with the edge? I want to delete it . Like this: 
[image]
Another ear appears, but I don't want to keep it.
So how should I do?

ev-mp, 2019-04-18:
@coldcoda ,this artifact is called by the occlusion phenomenon. Please read the explanation provided by @dorodnic https://github.com/IntelRealSense/librealsense/issues/2445#issuecomment-424481687.

librealsense provides a post-processing filter  (`occlusion-flilter` [here](https://github.com/IntelRealSense/librealsense/blob/development/src/proc/occlusion-filter.cpp#L12)) that removes the invalid capture. 
The filter is integrated into `realsense-viewer`s  3D view:
[image]

coldcoda, 2019-04-18:
@ev-mp ,
Thanks a lot!!I have switched it from Off to Heuristic.
[image]
But it also has the black area and the people's edge.  Is there a way to let the black area and the edge around the people disappear?
Or Is there a way to remove the background from the pointcloud?

ev-mp, 2019-04-18:
@coldcoda, to clarify - the occlusion filter does not remove the depth data. What it does is to look for depth pixels that have **invalid** (occluded) texture mapping , and replaces its color with black. 
It means that the depth values are actually correct, it's just that due to the mentioned phenomenon there is no way to know what is the actual color data for those pixels are ,so they are colored black.

Obviously you can modify the code and instead of changing the color, to actually remove the depth values altogether.

ev-mp, 2019-04-22:
@coldcoda , do you need more assistance?

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi coldcoda,

Given what ev-mp has mentioned, wonder if you get any update?

If there is nothing else needed, this one will be closed.

Thanks!

closed 2019-05-16 by RealSenseCustomerSupport