realsenseai/librealsense #3794: Building from Source on Windows 10 Fail

off99555, 2019-04-18:
I cloned the repo, install CMake and use it to build VisualStudio 2017 project. Then I open the project. I ran the example file and I got error.

This is some part of the output when trying to run `rs-hello-realsense` example using Local Windows Debugger.
```
6>   Creating library C:/Users/ADMIN/Documents/[Off]/[LibrarySources]/librealsense/build/Debug/realsense2.lib and object C:/Users/ADMIN/Documents/[Off]/[LibrarySources]/librealsense/build/Debug/realsense2.exp
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: class rs2rosinternal::Time const & __thiscall rosbag::MessageInstance::getTime(void)const " (?getTime@MessageInstance@rosbag@@QBEABVTime@rs2rosinternal@@XZ) referenced in function __catch$??0ros_reader@librealsense@@QAE@ABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$shared_ptr@Vcontext@librealsense@@@3@@Z$0
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const & __thiscall rosbag::MessageInstance::getTopic(void)const " (?getTopic@MessageInstance@rosbag@@QBEABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) referenced in function __catch$??0ros_reader@librealsense@@QAE@ABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$shared_ptr@Vcontext@librealsense@@@3@@Z$0
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const & __thiscall rosbag::MessageInstance::getDataType(void)const " (?getDataType@MessageInstance@rosbag@@QBEABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) referenced in function __catch$??0ros_reader@librealsense@@QAE@ABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$shared_ptr@Vcontext@librealsense@@@3@@Z$0
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const & __thiscall rosbag::MessageInstance::getMD5Sum(void)const " (?getMD5Sum@MessageInstance@rosbag@@QBEABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) referenced in function "public: bool __thiscall rosbag::MessageInstance::isType<struct geometry_msgs::Accel_<class std::allocator<void> > >(void)const " (??$isType@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@MessageInstance@rosbag@@QBE_NXZ)
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: unsigned char * __thiscall rosbag::Buffer::getData(void)" (?getData@Buffer@rosbag@@QAEPAEXZ) referenced in function "private: class std::shared_ptr<struct geometry_msgs::Accel_<class std::allocator<void> > > __thiscall rosbag::Bag::instantiateBuffer<struct geometry_msgs::Accel_<class std::allocator<void> > >(struct rosbag::IndexEntry const &)const " (??$instantiateBuffer@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@Bag@rosbag@@ABE?AV?$shared_ptr@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@std@@ABUIndexEntry@1@@Z)
6>ros_writer.obj : error LNK2001: unresolved external symbol "public: unsigned char * __thiscall rosbag::Buffer::getData(void)" (?getData@Buffer@rosbag@@QAEPAEXZ)
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: unsigned int __thiscall rosbag::Buffer::getSize(void)const " (?getSize@Buffer@rosbag@@QBEIXZ) referenced in function "private: class std::shared_ptr<struct geometry_msgs::Accel_<class std::allocator<void> > > __thiscall rosbag::Bag::instantiateBuffer<struct geometry_msgs::Accel_<class std::allocator<void> > >(struct rosbag::IndexEntry const &)const " (??$instantiateBuffer@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@Bag@rosbag@@ABE?AV?$shared_ptr@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@std@@ABUIndexEntry@1@@Z)
6>ros_writer.obj : error LNK2001: unresolved external symbol "public: unsigned int __thiscall rosbag::Buffer::getSize(void)const " (?getSize@Buffer@rosbag@@QBEIXZ)
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: __thiscall rosbag::Bag::Bag(void)" (??0Bag@rosbag@@QAE@XZ) referenced in function "public: __thiscall librealsense::ros_reader::ros_reader(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::shared_ptr<class librealsense::context> const &)" (??0ros_reader@librealsense@@QAE@ABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$shared_ptr@Vcontext@librealsense@@@3@@Z)
6>ros_writer.obj : error LNK2001: unresolved external symbol "public: __thiscall rosbag::Bag::Bag(void)" (??0Bag@rosbag@@QAE@XZ)
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: __thiscall rosbag::Bag::~Bag(void)" (??1Bag@rosbag@@QAE@XZ) referenced in function __unwindfunclet$??0ros_reader@librealsense@@QAE@ABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$shared_ptr@Vcontext@librealsense@@@3@@Z$6
6>ros_writer.obj : error LNK2001: unresolved external symbol "public: __thiscall rosbag::Bag::~Bag(void)" (??1Bag@rosbag@@QAE@XZ)
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: void __thiscall rosbag::Bag::open(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned int)" (?open@Bag@rosbag@@QAEXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@I@Z) referenced in function __catch$??0ros_reader@librealsense@@QAE@ABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$shared_ptr@Vcontext@librealsense@@@3@@Z$0
6>ros_writer.obj : error LNK2001: unresolved external symbol "public: void __thiscall rosbag::Bag::open(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned int)" (?open@Bag@rosbag@@QAEXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@I@Z)
6>ros_reader.obj : error LNK2019: unresolved external symbol "public: void __thiscall rosbag::Bag::close(void)" (?close@Bag@rosbag@@QAEXXZ) referenced in function __catch$??0ros_reader@librealsense@@QAE@ABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$shared_ptr@Vcontext@librealsense@@@3@@Z$0
6>ros_reader.obj : error LNK2019: unresolved external symbol "private: void __thiscall rosbag::Bag::readMessageDataRecord102(unsigned __int64,class rs2rosinternal::Header &)const " (?readMessageDataRecord102@Bag@rosbag@@ABEX_KAAVHeader@rs2rosinternal@@@Z) referenced in function "private: class std::shared_ptr<struct geometry_msgs::Accel_<class std::allocator<void> > > __thiscall rosbag::Bag::instantiateBuffer<struct geometry_msgs::Accel_<class std::allocator<void> > >(struct rosbag::IndexEntry const &)const " (??$instantiateBuffer@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@Bag@rosbag@@ABE?AV?$shared_ptr@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@std@@ABUIndexEntry@1@@Z)
6>ros_reader.obj : error LNK2019: unresolved external symbol "private: void __thiscall rosbag::Bag::decompressChunk(unsigned __int64)const " (?decompressChunk@Bag@rosbag@@ABEX_K@Z) referenced in function "private: class std::shared_ptr<struct geometry_msgs::Accel_<class std::allocator<void> > > __thiscall rosbag::Bag::instantiateBuffer<struct geometry_msgs::Accel_<class std::allocator<void> > >(struct rosbag::IndexEntry const &)const " (??$instantiateBuffer@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@Bag@rosbag@@ABE?AV?$shared_ptr@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@std@@ABUIndexEntry@1@@Z)
6>ros_reader.obj : error LNK2019: unresolved external symbol "private: void __thiscall rosbag::Bag::readMessageDataHeaderFromBuffer(class rosbag::Buffer &,unsigned int,class rs2rosinternal::Header &,unsigned int &,unsigned int &)const " (?readMessageDataHeaderFromBuffer@Bag@rosbag@@ABEXAAVBuffer@2@IAAVHeader@rs2rosinternal@@AAI2@Z) referenced in function "private: class std::shared_ptr<struct geometry_msgs::Accel_<class std::allocator<void> > > __thiscall rosbag::Bag::instantiateBuffer<struct geometry_msgs::Accel_<class std::allocator<void> > >(struct rosbag::IndexEntry const &)const " (??$instantiateBuffer@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@Bag@rosbag@@ABE?AV?$shared_ptr@U?$Accel_@V?$allocator@X@std@@@geometry_msgs@@@std@@ABUIndexEntry@1@@Z)
6>ros_reader.obj : error LNK2019: unresolved external symbol "private: bool __thiscall rosbag::Bag::readField(class std::map<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class  [... cut ...] etry_msgs@@ABV?$shared_ptr@V?$map@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@U?$less@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@V?$allocator@U?$pair@$$CBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@@std@@@2@@std@@@3@@Z)
6>C:\Users\ADMIN\Documents\[Off]\[LibrarySources]\librealsense\build\Debug\realsense2.dll : fatal error LNK1120: 48 unresolved externals
6>Done building project "realsense2.vcxproj" -- FAILED.
```

What did I miss?

ev-mp, 2019-04-18:
@off99555 hello,
One of the Librealsense's dependencies that are required during linking, namely - `realsense-file` - seem to be missing

[image]

Since that project is part of Librealsense's source tree it seem unreasonable that the source code is missing. Try to rebuild that project first. And if persistent - delete all cache/build artifacts and then make a clean build.

off99555, 2019-04-18:
It's unreasonable but it happened. That's programming life.
I saw it exist here:
[image]
Do I need to also clean the CMake artifact? I think I've done that before. I deleted the build folder, delete the cache from cmake-gui. But it still says the same error.

Suppose the unreasonable thing happened, what is other conclusion out of this? Is is possible that I did not install enough components for Visual Studio 2017? If it can be possible, what could be the missing components? I have installed C++ for Linux and Desktop.

CarltonSemple, 2019-04-24:
On a related note, is there any way to avoid using Visual Studio solutions with this SDK? The Windows release download and instructions all end up using that

off99555, 2019-04-24:
Because Visual Studio seems to be the only biggest IDE/compiler for C++ on Windows.

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

@off99555 Seems your issue resolved now and any other questions about this ticket? Looking forward to your update. Thanks!

off99555, 2019-04-28:
OK. It is solved in #3782 . @RealSenseCustomerSupport please also answer my last question in that issue and then I'll close that issue also. Thank you!

closed 2019-04-28 by off99555