realsenseai/librealsense #3796: Saved image via C# on Linux is not the same as in realsense viewer

StarlightNL, 2019-04-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                    | 
| Firmware Version                   | 05.11.01.100                                           | 
| Operating System & Version |   Ubuntu 18.04.2 LTS                                                 | 
| Kernel Version (Linux Only)    |   4.14.107-157                                        | 
| Platform                                 | Odroid XU4                                            |
| SDK Version                            |  2.20                                                      | 
| Language                                |  C# dotnet core 2.2                               | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }    | 

### Issue Description
The images that are saved are not what I except on Windows and that is showed on Realsense Viewer. 
The image looks like this in my program
[image]

but looks like this in Realsense-Viewer
[image]

Here is the important code
```
pipe = new Pipeline();
            colorizer = new Colorizer();
            colorizer.Options[Option.MaxDistance].Value = 2.3f;
            var cfg = new Config();
            cfg.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 30);
            var profile = pipe.Start(cfg);

using (var frames = pipe.WaitForFrames())
{
	using (var depth = frames.DepthFrame)
	{
		VideoFrame img = colorizer.Process<VideoFrame>(depth).DisposeWith(frames);
		bmp = new Bitmap(img.Width, img.Height, img.Stride, PixelFormat.Format24bppRgb, img.Data);
		bmp.Save(DateTime.Now.ToString("yyyyMMddHHmmss")+".png");
	}
}
```

AntoineWefit, 2019-04-19:
If you just want to display a black and white image in the RealSense Viewer, simply go to the drop down menu _**Stereo Module/Depth Visualization/Color Scheme**_ and pick _**White to Black**_ or _**Black to White**_ instead of _**Jet**_.

StarlightNL, 2019-04-19:
@AntoineWefit That is not the issue here. The issue is that I want it reversed. In Windows the same application gives a Jet colorized image. The image from the application on Linux is hard to explain whats wrong with it but it is squized and partly duplicated on the right side.

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

Hi @StarlightNL,

Per the description, your C# program is running on Odroid XU4 + Ubuntu 18.04, right?
I don't observe this issue on Win10 + librealsense 2.21.0 + D415 firmware 5.11.6.200 + .NetFramework 4.5.2.

StarlightNL, 2019-04-30:
@RealSenseCustomerSupport 
That is right. The only thing is that the application is running on .net core 2.2 and not on the standard .netframework, however after searching I am going to try to compile libgdiplus that is used with System.Drawing on the odroid and see if that fixes the drawing problems.

The following info will also be included in the original post.
As extra info as what is wrong with the picture. There is only one pole on the left corner as seen in the jet color image, however on the image that is saved there are 2 poles. The image is thinner and partly duplicated when converting the image to a BMP.

RealSenseCustomerSupport, 2019-05-14:
----------------------------------------------

Hi @StarlightNL,

Any good news to resolve the issue?

RealSenseCustomerSupport, 2019-05-29:
----------------------------------------------

I don't observe this issue based on opencv on Odroid XU4 to display the depth image. It should be related to .NetCore.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @StarlightNL,

Still need any support for this issue? Did you also check with Odroid forum to see if any help?

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

If no further questions, will close this.

closed 2019-12-06 by RealSenseCustomerSupport