realsenseai/librealsense #3798: T265 camera live stream in Matlab

jattiogbe, 2019-04-19:
|---------------------------------|------------------------------------------- |
| Camera Model                       |T265} |  
| Operating System & Version |   Wind 10 |  
| Platform                                 | PC  |
| SDK Version                            |  2.20      | 
| Language                            |  Matlab  2019                    |  

Hi,

I'm trying to get the video stream from T265, left/ right or both cameras with Matlab 2019.
I've tried to adapt [this C code](https://github.com/IntelRealSense/librealsense/tree/master/examples/pose-and-image) with no success.
Anyone has made it ?

thanks

JimmyFroost, 2019-04-26:
Can someone out there help us out on this issue ?

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Librealsense does not have support for Matlab for T265 camera at this time.  We will take this as feedback for future changes.

closed 2019-05-24 by RealSenseCustomerSupport

JimmyFroost, 2019-05-27:
Hi @RealSenseCustomerSupport ,

Thanks for the information, I've been trying to continue on this issue, accordingly to this [CPP code](https://github.com/IntelRealSense/librealsense/blob/master/examples/ar-basic/rs-ar-basic.cpp).

On line 107, there is no equivalent for **get_fisheye_frame** in Matlab (yet), so I tried to code it on my own since the function is very similar to **get_infrared_frame,** based on lines 945 to 980 in [this code file](https://github.com/IntelRealSense/librealsense/blob/f361e56f8a2fb5093a1d663293421c92dd432bba/include/librealsense2/hpp/rs_frame.hpp).

So here it is in **frameset.m** : 
```

  function fisheye_frame = get_fisheye_frame(this, index)
            if (nargin == 1)
                ret = realsense.librealsense_mex('rs2::frameset', 'get_fisheye_frame', this.objectHandle);
            else
                validateattributes(index, {'numeric'}, {'scalar', 'real', 'integer'}, '', 'index', 2);
                ret = realsense.librealsense_mex('rs2::frameset', 'get_fisheye_frame', this.objectHandle, int64(index));
            end
            fisheye_frame = realsense.video_frame(ret);
        end
```

Evnetually, I get this error :

> Error using librealsense_mex
> Unknown Command received.
> 
> Error in realsense.frameset/get_fisheye_frame (line 56)
>                 ret = realsense.librealsense_mex('rs2::frameset', 'get_fisheye_frame', this.objectHandle, int64(index));
>
> Error in T265_Stream (line 67)
> img_fish =  fs.get_fisheye_frame();

Is there something I can do or am I doomed to wait for an update of yours ?

Thanks.