realsenseai/librealsense #3799: D435i depth calibration produces substandard results compared to the D435

lauesa, 2019-04-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (05.10.13.00) | 
| Operating System & Version |   {Linux (Ubuntu 16.04)| 
| Kernel Version (Linux Only)    |  (4.15.0-47-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.19.0}                          | 
| Language                            |  {NA }                          | 
| Segment			|  {Robot}                          | 

### Issue Description

I've utilized the D435 Realsense camera in the past to very accurately estimate distance. In the past I've used the DynamicCalibratorTool and Print Target for depth calibration and was fairly satisfied. Upon purchasing a pair of D435i cameras, hoping to make use of them in our new system, I calibrated them with the same methods and found their depth sensors placed ranges approximately 5 inches close to the camera frame than the D435. Are different processes needed for calibrating them or is there a 3rd-party tool that would perform better?

(I've done this calibration and testing on 2 new D435i's)

Thanks in advance

dorodnic, 2019-04-28:
Hi @lauesa 
This is rather strange. The depth module of the two cameras is identical. Let's wait to see what @RealSenseCustomerSupport has to say, perhaps the cameras can be replaced? 
Maybe @agrunnet has some idea regarding what can affect absolute Z accuracy.

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

Hi

We just tested at 1 meter (and 2 meters) on both D435 and D435i with FW5.10.13, the results don't appear big different.
When you say "5 inches" difference, what was the distance you measured?
Would you please by the chance try the latest firmware version?

Thanks

lauesa, 2019-05-03:
Original distance measured was 62 inches. Upgrading to 5.11.1.100 on the D435i gave correct result. 

Thanks

RealSenseCustomerSupport, 2019-05-03:
----------------------------------------------

Awesome !
Thank you very much for letting us know the result.

closed 2019-05-07 by lauesa