realsenseai/librealsense #3800: Exposure meta data not available after fresh install ubuntu 18 with dkms

kevindehecker, 2019-04-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / D435i | 
| Firmware Version                   | 2.20.0 | 
| Operating System & Version |   Linux Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.18.0-17-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.20.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description

The exposure metadata (RS2_FRAME_METADATA_ACTUAL_EXPOSURE) is not available on new 18.04 installs. We are in the process of upgrading our systems to 18.04 from 16.04. Before the upgrades the metadata was available, after the upgrade not anymore with the same code and everything.

I have reproduced this problem with several camera's (both D435 and D435i's), which still work on 16.04 installs (or even older 18.04 installs that were upgraded through apt a few weeks ago).

All systems have the dkms packages installed in the same way. I have tried reinstalling them, but no luck.

I read here https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#os-support that only 16.04 is supported for meta data. Is that the case, or is this a bug? If it is not a but, when can I expect 18.04 support for this?

ev-mp, 2019-04-21:
@kevindehecker hello , using precompiled debians is currently limited to LTS kernels 4.4-4.15 (See [installation notes](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#using-pre-build-packages) for both Ubuntu 16  and 18. 
We're working to enable kernel 4.18 in next releases.

Duplicated in #3536

kevindehecker, 2019-04-23:
I installed kernel 4.15.18-041518-generic, apt removed and installed librealsense... still not working....!

ev-mp, 2019-04-23:
@kevindehecker , did you install the DKMS package with that ?
> sudo apt install librealsense2-dkms

The latest available version, iirc, is 1.3.5.
You can check its status with `dkms status`

You can also post the installation log if you suspect it didn't work correctly

ev-mp, 2019-05-13:
@kevindehecker hi, do you still need support on this ?
Please update, thanks.

kevindehecker, 2019-05-20:
Sorry, was away on vacation! Will give an update next Friday.

ccfire, 2019-05-23:
I have nearly the same problem:
My laptop env:
ubuntu18.04 linux-4.15.18-041518-generic.

After run:
sudo apt-get install librealsense2-dkms

Output error:
Building initial module for 4.15.18-041518-generic
ERROR (dkms apport): kernel package linux-headers-4.15.18-041518-generic is not supported
Error! Bad return status for module build on kernel: 4.15.18-041518-generic (x86_64)

ev-mp, 2019-05-23:
@ccfire hello, is this this the same as #4054 ?

ccfire, 2019-05-24:
@ev-mp Yes.

kevindehecker, 2019-05-24:
@ev-mp you were right, there was indeed an error in the install log:
```
ERROR (dkms apport): kernel package linux-headers-4.15.0-041500-generic is not supported
Error! Bad return status for module build on kernel: 4.15.0-041500-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.
```
That make.log tells me I'm missing linux/logic_pio.h:
```
DKMS make.log for librealsense2-dkms-1.3.5 for kernel 4.15.0-041500-generic (x86_64)
vr 24 mei 2019 13:49:47 CEST
make: Entering directory '/usr/src/linux-headers-4.15.0-041500-generic'
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o
In file included from ./arch/x86/include/asm/io.h:401:0,
                 from ./arch/x86/include/asm/realmode.h:15,
                 from ./arch/x86/include/asm/acpi.h:33,
                 from ./arch/x86/include/asm/fixmap.h:19,
                 from ./arch/x86/include/asm/apic.h:10,
                 from ./arch/x86/include/asm/smp.h:13,
                 from ./arch/x86/include/asm/mmzone_64.h:11,
                 from ./arch/x86/include/asm/mmzone.h:5,
                 from ./include/linux/mmzone.h:912,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:22,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.c:17:
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/asm-generic/io.h:354:10: fatal error: linux/logic_pio.h: No such file or directory
 #include <linux/logic_pio.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.build:316: recipe for target '/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o' failed
make[1]: *** [/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o] Error 1
Makefile:1690: recipe for target 'uvcvideo.ko' failed
make: *** [uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-4.15.0-041500-generic'
```

Any idea what I need to install to get this fixed? Couldn't find much on it myself.

RealSenseCustomerSupport, 2019-06-26:
----------------------------------------------

Hi @kevindehecker,
Were you able to overcome this by referencing to the following issue?
https://github.com/IntelRealSense/librealsense/issues/4054#issuecomment-495141217

kevindehecker, 2019-06-26:
Nope, that did not work for me. I needed that system to work, so I ended up reinstalling the system with Ubuntu 18.04.01...

I still have my own laptop which I don't like to reinstall with the 4.18 headers. I'm not going to downgrade that to 4.15. Could you check when the 4.18 headers will be supported?

RealSenseCustomerSupport, 2019-06-27:
----------------------------------------------

Hi @kevindehecker,
Thanks to @ev-mp has the PR for 4.18 kernel.
https://github.com/IntelRealSense/librealsense/pull/4265

kevindehecker, 2019-06-28:
Cool, thanks!

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

You are welcome.
Closing this ticket now. Please feel free to reopen if further help is needed.
Thank you.

closed 2019-07-01 by RealSenseCustomerSupport