realsenseai/librealsense #3802: record-playback multiple cameras

rexlim820220, 2019-04-20:
The project we're embarking on aims to create a program that simultaneously records depth frames from two (or even more) distinct D435 cameras. 

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | RealSense D400 series | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Linux Mint  | 
| Kernel Version (Linux Only)    |  4.16.18                                         | 
| Platform                                 | PC |
| SDK Version                            | v2.10.0                         | 
| Language                            |  C++                          |  

[image]

Before setting up, we had our camera hardware synchronized such that the recorded frames have aligned frame numbers. Since the viewer can't invoke the record streaming of two channels immediately at the time, we had no choice but to craft one-from scratch. In short, what we're doing is to combine the functions of `multicam` and `record-play-back` in SDK's example together, we also referred to https://github.com/IntelRealSense/librealsense/issues/2637 sample code to achieve hardware synchronization. 

[image]

Firstly, two depth frames seems to race each others for the bandwidth resources. As can be observed from my screenshot, since when one camera  is set to master and another is set to slave, one of their frame rate would be delayed. The only one reasonable solution we can coming up is allow multi-threading work during the streaming.  `

    std::size_t              dev_id = 0;
    rs2::context                    ctx;
    rs2::colorizer            colorizer;
    std::vector<std::thread>    threads;
    texture                 depth_image;
    auto     list = ctx.query_devices();
    for (auto&& dev : list)
    {
            threads.emplace_back([dev, dev_id](){
                sync(&dev, dev_id);
                rs2::pipeline pipe(ctx);
                rs2::config cfg;
                cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
                pipe.start(cfg);
                dev_id++;
            }
    }
`

However, we are coming across the problem that access the former pipeline with its frames stored in `std::vector` of `std::thread` in the first loop. The stored frames would be reached and being rendered with the following snippet of codes, inside the `while (app)` part that alert the GUI window. `

    for (const auto& viewer : new_views)
    {
            float scale = counter * 0.25f;
            depth = colorizer.process(viewer.frame.get_depth_frame());
            depth_image.render(depth, {150 +  w * ( 0.25f - scale) , 500 - h * 0.25f , 200 + w * ( 0.5f - 
            scale) ,  720 - h * (0.75f)});
            counter += 1;
            counter %= list.size();
     }
`

The `counter` variable aims to distinguish the channel that GUI's `texture` render in `"example.hpp"` from different cameras. So the problem is stuck at how to achieve an efficiently thread managing as stated before. We had tried searching and it was not straightforward to get this particular solution from the google results (without the benefit of hindsight that is). 

The second issue happens when the `ImGui::Button("record")` is clicked, the program shut down automatically. As stated before, most of our executing code was revised from that in `multicam` and `record-play-back`. Especially the GUI control interface in `record-play-back` is really complicated, we have no idea what to do next; so far, we are wondering that if there is any substitution for the viewer to starting streaming two cameras or anyone could give us some tips on revising the code.

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi @rexlim820220,
Were you able to overcome the issue?
The latest rs-multicam sample should answer the problem in regard to have multiple pipeline objects for multiple RealSense devices. Please take a look. Thank you.
https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

rexlim820220, 2019-05-16:
So far, the issue still remains; during the debugging process we've modify some more snippets which wasn't been included above. Firstly, we've changed the output filename that `cfg.enable_record_to_file().`holds, thus avoiding the naming conflicts. Secondly, we've switched the `wait_for_frames` in our example into 
`poll_for_frames` according to [#2422](https://github.com/IntelRealSense/librealsense/issues/2422). 
Now that another new problem arise, stating that
`RealSense error calling rs2_process_frame(block:0x558508ade240, frame:nullptr):
    null pointer passed for argument "frame"` 
Unfortunately, debugging is really tedious here because we have little time understanding all the functions we've called in `rs2::` library, which make the developing progress slow and challenging.

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi, @rexlim820220,
Please check whether the depth frame is valid before processing it.
Thank you.

closed 2019-06-08 by RealSenseCustomerSupport

rexlim820220, 2019-06-08:
[image: Screenshot from 2019-06-08 10-17-20.png]

RealSense Customer Support <notifications@github.com> 於 2019年6月8日 週六
上午9:04寫道：

evarcurto, 2020-06-18:
Hi! Were you able to record multiple cameras synchronously?
I want to record 4 cameras synchronously and then get the pointclouds.
I already synchronized the cameras as described in the white paper and using the multicamera and save-to-disk examples I'm able to save the color and depth frames for each camera but I wanted to save the .bag file for each camera.
Can you indicate some code made in C ++ to store .bag files or pointclouds for synchronous multi-cameras?

rexlim820220, 2020-06-21:
Hi @evarcurto. Unfortunately, the issue still bothers us. Synchronization between writing different rosbag files simultaneously seems to requires understanding to its inner file I/O implementation. Hope that more people can contribute their ideas if it works :)