realsenseai/librealsense #3803: d435i / d430i HID device required to connect camera

state-of-the-art, 2019-04-20:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.11.1.100 | 
| Operating System & Version |   Android 7.0 Nougat SM-G950U1 | 
| Kernel Version (Linux Only)    | 4.4.16-12116658  | 
| Platform                                 | ARM64  |
| SDK Version                            | 2.19.2 / 2.20.0 | 
| Language                            |  C++  | 
| Segment			|  Smartphone  | 

### Issue Description
Hello guys,

During debug of D435i connection issues was found that the camera can't be connected on the android phone just because of the hid devices are not available state-of-the-art/Handy3DScanner#26 (please see the logs in the ticket).

It's not a good idea to not connect device just because some of the functionality (IMU sensor in this case) is not available. Basically the device is working without this sensor and could stream the depth data (main device purpose).

Additional features should be optional and not mandatory - we can see that when device is connected there is additional warning from librs:
```
W librs   : HID device is missing!
```
So it's just enough (with message adjusting) to see that the IMU will not work properly.

dorodnic, 2019-04-28:
Hi @state-of-the-art 
Sorry for long response time. Yes, this is an option. I'm also waiting on proper HID via USB support, that is being delayed, but is coming. We will try to add something like this in the meantime.

closed 2019-05-05 by RealSenseCustomerSupport

MartyG-RealSense, 2019-08-21:
@state-of-the-art   I just wanted to update your case to let you know that D435i support for Android has apparently been added in the new SDK version 2.26.0, according to the Releases web-page.

rabits, 2019-08-22:
Thank you @MartyG-RealSense , I will check the new release.

state-of-the-art, 2019-08-22:
Thank you @MartyG-RealSense  , I will check the new release.