realsenseai/librealsense #3806: Releasing date of youtube 3D lidar demo code?

drgnman, 2019-04-21:
### Issue Description

Hi, I have watched the youtube video of 3D LIDAR adaptation of realse, (https://www.youtube.com/watch?v=LBMsWJJxLXQ)
and in the description it is mentioned that “We will be releasing some of the code used in this demo in the near future on our github at https://github.com/IntelRealSense/lib...“.
However, I don’t see any example code nor functions for 3D lidar.
So here are my questions:

Q1. Around when will the example codes for 3D lidar released?
Q2. Is there any example code / functions already made open for 3D LIDAR in this GitHub?

Regards.

MartyG-RealSense, 2019-04-21:
The RealSense Tracking Camera T265 is not a LIDAR device.  Official information describes it like this:

"The Intel RealSense Tracking Camera T265 uses inside-out tracking, which means the device does not rely on any external sensors to understand the environment.  Unlike other inside-out tracking solutions, the T265 delivers 6-degrees-of-freedom (6DoF) inside-out tracking by gathering inputs from two onboard fish-eye cameras, each with an approximate 170-degree range of view.  The V-SLAM systems construct and continually update maps of unknown environments and the location of a device within that environment".

LIDAR has been used in combination with RealSense in the past though on a retail robot called Tally.

 https://www.youtube.com/watch?v=6NN8kXh-Tpk

drgnman, 2019-04-21:
Thanks for the comment, and yes 3D LIDAR was not an accurate wordings. I have changed the name of the question too.

However, my main question still remains unsolved.  To put the question accurately, I think it would be like this:

Q1. Around when will the example codes of the demo in the following video released? 
https://www.youtube.com/watch?v=LBMsWJJxLXQ

It would be great if you could give a bit more help too.

Regards.

MartyG-RealSense, 2019-04-21:
I do not have any details of when the software in that demo will be released, unfortunately.   One of the Intel guys on this forum may be able to help you with this.

drgnman, 2019-04-23:
OK, thanks for response.  I hope its man help me for this question.

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi @drgnman

Please continue to monitor our ros repositories and our librealsense releases:  https://github.com/IntelRealSense/realsense-ros, https://github.com/IntelRealSense/librealsense

We are continuing to make updates to samples and examples using our T265 and D400 series cameras.  We do not have a specific timeframe for the example you are looking for.  As stated our team is actively working on the solution and to make it available on our github repositories.

Thanks for inquiring about the demo example code.

closed 2019-05-24 by RealSenseCustomerSupport

PaulKrush, 2019-07-10:
Similar:
https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping/occupancy