realsenseai/librealsense #3811: Object Detection dnn error

BeatrizAMartins, 2019-04-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 Depth Module | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                           | 2.0. | 
| Language                            |  Python | 

### Issue Description
I am trying to do object detection based on this: [https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb](url) 

However, when I run my code I get an error in the detections (_detections = net.forward("detection_out")_):

**cv2.error: OpenCV(4.0.0) C:\projects\opencv-python\opencv\modules\dnn\src\layers\convolution_layer.cpp:260: error: (-215:Assertion failed) ngroups > 0 && inpCn % ngroups == 0 && outCn % ngroups == 0 in function 'cv::dnn::ConvolutionLayerImpl::getMemoryShapes'**

Can you give me some advices to fix this or do you know why is this happening?

RealSenseCustomerSupport, 2019-05-15:
----------------------------------------------

Hi BeatrizAMartins,

Not sure if you have any update on this one or not, from the internet search within opencv community, it seems that the error can be with various reasons, some of them are related to the way handling image read, or some processing in dnn model.

Hope this can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-05-23:
----------------------------------------------

Hi BeatrizAMartins,

Did you get any update on this one? Still need any help?

Thanks!

closed 2019-06-06 by RealSenseCustomerSupport