realsenseai/librealsense #3812: Whether the T265 could get the two fisheye camera's calibrated stream?

oxm, 2019-04-22:
@dorodnic Whether the T265 could get the two fisheye camera's calibrated stream?

ev-mp, 2019-04-30:
@oxm, T265 provides left and right Fisheye calibrated streams, along with IMU and Pose data.
The Fisheye stream intrinsic and extrinsic parameters are available since [v2.20.0 release](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements-1)

bfulkers-i, 2019-05-10:
@oxm in case you are asking about how to undistort the fisheye images, if you are using OpenCV the fisheye module supports the distortion model used by T265:

https://docs.opencv.org/3.4/db/d58/group__calib3d__fisheye.html

You may also be interested in this python example that uses the T265 intrinsics and extrinsics to compute a stereo image in OpenCV:

https://github.com/IntelRealSense/librealsense/pull/3951

closed 2019-05-13 by RealSenseCustomerSupport