realsenseai/librealsense #3818: Documentation (GitHub) Firmware version differs from API Firmware version

evercoast-dev, 2019-04-22:
Camera Model: D400
Firmware Version: 5.10.03
Operating System: Windows + Linux
Kernel Version: All
Platform: PC
SDK Version: 2.19.0
Language: C++
Segment: VR/AR

The documentation for releases, found at:
https://github.com/IntelRealSense/librealsense/releases

Indicates the recommended firmware version for 2.19.0 as 5.11.4, while executing the API call:
dev.get_info(RS2_CAMERA_INFO_FIRMWARE_VERSION);

Results in the recommended firmware version of 05.10.03.00.

Which is correct? Why do they differ?

dorodnic, 2019-04-28:
`RS2_CAMERA_INFO_FIRMWARE_VERSION` is your current firmware version. 
You can update it using the [firmware update tool](https://www.intelrealsense.com/developers/#tools)

evercoast-smm, 2019-04-28:
Apologies @dorodnic, the code snippet is inaccurate in the above and should read as follows:
dev.get_info(RS2_CAMERA_INFO_RECOMMENDED_FIRMWARE_VERSION);

The remainder of the bug report remains valid, that the GitHub recommended listed firmware version does not match the version as recommended by the API.

dorodnic, 2019-04-29:
Ah Ok :)
Yes, you are correct. I'm not sure if to classify this as a bug. The thing is, on GitHub we recommend the latest firmware we tested. In code, we just need to decide on some firmware below which the viewer starts to complain. 
Primary purpose of the firmware update notifications is to urge people to get from 5.8.x / 5.9.x that the camera comes with to something more recent. One reason is that recent versions of Linux kernel refuse to stream with these early versions (lacking proper End-of-Frame signalling), and same applies to Android I believe. 
Regardless, in a sprint or two this whole thing should become much easier, since latest firmware will 
be bundled with the SDK and you will have a cross-platform update option directly from the Viewer / API.

evercoast-dev, 2019-05-03:
An API firmware update option would be great for us. Looking forward to that!

closed 2019-05-03 by evercoast-dev