realsenseai/librealsense #3821: How to restore to the initial capture parameters

atinfinity, 2019-04-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.20.0                         | 
| Language                            |  C                          | 
| Segment			|  Robot                          | 

### Issue Description
I tried to adjust capture parameters of realsense-viewer [v2.20.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.20.0) on Windows.
But, I could not find how to restore to the initial capture parameters.

I tried the following procedure. But, this problem has not been resolved.

- restart realsense-viewer
- reseat RealSense D435 
- use **[Restore Defaults]** on realsense-viewer<br><br>[image]

ev-mp, 2019-04-23:
@atinfinity hello, it is not clear what do you mean by `initial capture parameters` -

The `Restore Defaults` in the above snapshots refers to the `realsense-viewer` UI-settings, such as windows size and position, default record path and so on.
These setting ares stored in "realsense-config.json" on hard-disk under the (OS-specific) `Appdata` folder

It is also noted that these settings do not include the device(s) streaming configuration - i.e. selected streams, resolutions, fps, and HW controls values.

atinfinity, 2019-04-23:
@ev-mp Thank you for your comment.

> hello, it is not clear what do you mean by initial capture parameters

The `initial capture parameters` is the value set at the first startup of realsense-viewer.  
e.g.

* capture size(width x height)
* framerate
* Decimation Filter Magnitude
* Spatial Filter Smooth Alpha
* Threshold Filter Max Distance

> These setting are stored in "realsense-config.json" on hard-disk under the (OS-specific) Appdata folder

I found `realsense-config.json` in `C:\Users\<user_name>\AppData\Roaming`.

ev-mp, 2019-04-23:
@atinfinity hi,
The streaming device configuration is **partially** captured/restored via `export/import json functionality`:
[image]

These parameters are limited to the Depth sensor only and:
a) include Depth/IR stream selections, resolution, formats and and fps
b) include Depth stream control values (gain, laser power, etc)
c) include Depth stream "advanced mode" controls value
d) **do not include** the post-processing parameters:
    -  post-processing filters selection/configuration. 
    -  post-processing depth visualization color scheme and visual preset

atinfinity, 2019-04-24:
Thank you for your advise! I'll try it.

closed 2019-04-24 by atinfinity