realsenseai/librealsense #3822: How to turn off auto exposure in Unity

wodndb, 2019-04-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version | Win10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.19.0 }                          | 
| Language                            | C# / Unity }                          | 
| Segment			|  Image Processing  | 

### Issue Description
How to turn off auto exposure of RealSense Camera in Unity like Intel RealSense Viewer?

MartyG-RealSense, 2019-04-23:
A RealSense user posted a script for toggling auto-exposure in Unity.

https://github.com/IntelRealSense/librealsense/issues/3088

Although they reported that their script was crashing, it turned out to be a problem that was fixed with a firmware update, rather than there being a problem with the code.

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

@wodndb Did you get it through? Looking forward to your update. Thanks!

wodndb, 2019-04-30:
@RealSenseCustomerSupport I can't turn off auto exposure mode of RGB sensor. Because RGB sensor can't support enable or disable auto exposure mode in unity. 

```
        if (ActiveProfile.Device.Sensors[1].Options[Option.WhiteBalance].Supported)
        {
            Debug.Log("WhiteBalance is supported");
        }
        else
        {
            Debug.Log("WhiteBalance is not supported");
        }
        if (ActiveProfile.Device.Sensors[1].Options[Option.AutoExposureMode].Supported)
        {
            Debug.Log("AutoExposureMode is supported");
        }
        else
        {
            Debug.Log("AutoExposureMode is not supported");
        }
```

[image]

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

@wodndb Please refer to below if you want to control RGB sensor. Thanks!

https://github.com/IntelRealSense/librealsense/blob/v2.21.0/wrappers/csharp/Intel.RealSense/Types/Enums/Option.cs#L44

wodndb, 2019-04-30:
```
        if (ActiveProfile.Device.Sensors[1].Options[Option.EnableAutoWhiteBalance].Supported)
        {
            Debug.Log("EnableAutoWhiteBalance is supported");
            ActiveProfile.Device.Sensors[1].Options[Option.EnableAutoWhiteBalance].Value = 0.0f;
            ActiveProfile.Device.Sensors[1].Options[Option.WhiteBalance].Value = 6300.0f;
        }
        else
        {
            Debug.Log("EnableAutoWhiteBalance is not supported");
        }
        
        if (ActiveProfile.Device.Sensors[1].Options[Option.EnableAutoExposure].Supported)
        {
            Debug.Log("EnableAutoExposure is supported");
            ActiveProfile.Device.Sensors[1].Options[Option.EnableAutoExposure].Value = 0.0f;
        }
        else
        {
            Debug.Log("EnableAutoExposure is not supported");
        }
```
It works well, Thx, @RealSenseCustomerSupport

closed 2019-04-30 by wodndb