realsenseai/librealsense #3827: Losing frames while saving frames as .bin

CaoKha, 2019-04-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 16.04 LTS)   | 
| Kernel Version (Linux Only)    | 4.15.0-37-generic                                         | 
| Platform                                 | Aaeon Up Board |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi,
I want to save the frames captured by your camera D435 (using this the code snippet from this issue https://github.com/IntelRealSense/librealsense/issues/1485) at each time I received a pulse sent by pin VSYNC.

However, when I opened the file and plot its timestamp data into a graph, I perceived that at some moment, some of the frames went missing. Here is my code (I am using mraa and upm as libraries for my api)
    
    void save_frames_to_files(rs2::pipeline &pipe, std::ofstream &ImageTimeFile, /*double &GNSS_time, int &GNSS_index,*/ const std::string &depth_path_string, const std::string &color_path_string)
    {
    static int fcount = 0;
    std::stringstream depth, color;

    unsigned long frame_time = std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::system_clock::now().time_since_epoch()).count();

    rs2::frameset frames = pipe.wait_for_frames();
    rs2::frame depth_frame = frames.get_depth_frame();
    rs2::frame color_frame = frames.get_color_frame();
    // Save to frame to file
    depth << depth_path_string << "/Depth" << std::setw(7) << std::setfill<char>('0') << fcount << ".bin";
    color << color_path_string << "/Color" << std::setw(7) << std::setfill<char>('0') << fcount << ".bin";
    save_frame_raw_data(depth.str(), depth_frame);
    save_frame_raw_data(color.str(), color_frame);
    ImageTimeFile << fcount << ";" << frame_time << "; Image" << std::setw(4) << std::setfill<char>('0') << fcount << "\n";
    fcount++;
    }
    // Get camera sensors
    rs2::sensor depth_sensor = rs2::context().query_devices()[0].query_sensors()[0];
    rs2::sensor color_sensor = rs2::context().query_devices()[0].query_sensors()[1];

    // Set up camera
    depth_sensor.set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 1); //master mode

    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_Y16, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

    std::once_flag flag1
    std::sig_atomic_t stop = 0;
    // Start streaming
    pipe.start(cfg);

    while (stop == 0)
    {
        if (gpio_1.read() != 0)
        {
            std::thread t_camera(save_frames_to_files, std::ref(pipe), std::ref(ImageTimeFile),std::cref(depth_path_string), std::cref(color_path_string));
            std::call_once(flag1, setScheduling, std::ref(t_camera), SCHED_RR, 30, std::ref(sch_params4));
            t_camera.join();
        }
    }
[image]

 Does my code go wrong or we can't save the raw frames in real time at 30Hz ?

dorodnic, 2019-04-28:
Hi @CaoKha 
I'm not 100% sure, but I do have some questions regarding the code...
You are creating new thread for every camera frame(set), why?
Please double-check that both the code and the SDK are compiled as Release (`-DCMAKE_BUILD_TYPE=Release`)
I'd try something like:

```cpp
using namespace rs2;
frame_queue queue(10); // some capacity, 1 by default
thread t([&](){
while(stop == 0)
    frameset fs = queue.wait_for_frame();
    frame depth_frame = frames.get_depth_frame();
    frame color_frame = frames.get_color_frame();
    save_frame_raw_data(depth.str(), depth_frame);
    save_frame_raw_data(color.str(), color_frame);
});
pipeline p;
p.start();
while(stop == 0) queue.enqueue(p.wait_for_frames());
t.join();
```

CaoKha, 2019-04-29:
@dorodnic, to answer your questions, 
I used thread there because I want to start the camera and another IMU (MPU9250 from sparkfun) at the same time (problem of synchronization between camera and IMU). I am still new to coding so after doing some research on the Internet, I found multi-threading is a solution to my problem. 
I did use the command "cmake ../ -DCMAKE_BUILD_TYPE=Release" to build my project. 
In your answer, if I haven't made mistake, you used a queue to slow down the process thus avoid losing frames. 
However, if I use a queue to hold the frames, my IMU data recorded at instant t will mismatch the frame "en-queue" at instant t, which was recorded at instant (t -a) ms earlier.
Yesterday, after reducing the reading rate of my IMU from 200 Hz to 150 Hz, I noticed that I didn't lose any frames in 2 hours. Hence, I concluded that maybe the problem was not from your camera but rather from my Linux system.
However, if I reduce my IMU reading rate, it will make my Kalman filter, which corrects the position of my robot, less efficient. 
I read some of your other posts that you recommend recording the frames in a rosbag format. But in my case, I believe that I couldn't synchronize my IMU data with that. 
So that, I am asking for your help in order to know the most efficient way to save the frames in real-time.

dorodnic, 2019-04-29:
The queue serves two purposes:
a. To handle sudden slowdowns by allowing extra latency
b. To communicate with an existing thread

My concern is that creating new thread for every frame might be too expensive. 
So even if you reduce queue capacity to 1, having single dedicated thread instead of many might improve performance. At the same time, with queue size 1 you should not suffer from sync issues with the IMU.

CaoKha, 2019-04-29:
Thank you @dorodnic,  you were right, I will try your solution and see if it might work.

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@CaoKha Did you get it through? Looking forward to your update. Thanks!

CaoKha, 2019-05-05:
Hi, I tried to implement the queue as @dorodnic said and the result is not good as I expected. I also tested with another board (Rock pi 4) and the result was also not going so well, even when I removed the IMU thread and let the camera work alone. (I later decided to join IMU thread with the main thread)

However,I saved these images to an SD Card. I think the format of the SD Card slows down the camera.

Sometimes, the ouput of 'rs2_time_t dtimestamp = depth_frame.get_timestamp()' is 3113.3113 at some first images, which is not the unix timestamp. The rest of the images are retarded 66 ms compare to my timestamp using std::chrono.

Here is the the code:

    std::signal(SIGINT, signal_handler);
    // GPIO config
    mraa::Result status;
    mraa::Gpio gpio_1(PIN);
    status = gpio_1.dir(mraa::DIR_IN);
    if (status != mraa::SUCCESS)
    {
        printError(status);
        return EXIT_FAILURE;
    }
    sleepMillis(1000);

    rs2::sensor depth_sensor = rs2::context().query_devices()[0].query_sensors()[0];
    //rs2::sensor color_sensor = rs2::context().query_devices()[0].query_sensors()[1];
    rs2::config cfg;
    rs2::pipeline pipe;
    depth_sensor.set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 1);        //master mode
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_Y16, 15); //change camera rate here 15 or 30Hz
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 15); //change camera rate here 15 or 30Hz
    // Set up IMU
    upm::MPU9250 IMU(0, MPU9250_DEFAULT_I2C_ADDR, AK8975_DEFAULT_I2C_ADDR, false);
    IMU.init();
    IMU.setSampleRateDivider(0x04); // Set output rate to 200Hz
    int imu_count = 0;
    float ax, ay, az, gx, gy, gz, temp;
    // Camera thread
    int fcount = 0;
    rs2::frame_queue queue(1);
    std::thread tc([&]() {
        while (stop == 0)
        {
            rs2::frameset frames = queue.wait_for_frame();
            unsigned long frame_time = std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::system_clock::now().time_since_epoch()).count();
            rs2::frame depth_frame = frames.get_depth_frame();
            rs2::frame color_frame = frames.get_color_frame();
            rs2_time_t dtimestamp = depth_frame.get_timestamp();
            rs2_time_t ctimestamp = color_frame.get_timestamp();
            std::stringstream depth, color;
            depth << depth_path_string << "/Depth" << std::setw(7) << std::setfill<char>('0') << fcount << ".bin";
            color << color_path_string << "/Color" << std::setw(7) << std::setfill<char>('0') << fcount << ".bin";
            save_frame_raw_data(depth.str(), depth_frame);
            save_frame_raw_data(color.str(), color_frame);
            for (int i = 0; i < 10; i++)
                ImageTimeFile << fcount << ";" << frame_time << ";" << ctimestamp << ";" << dtimestamp << "; Image" << std::setw(4) << std::setfill<char>('0') << fcount << "\n";
            fcount++;
        }
    });
    //sched_param mysched;
    //setScheduling(tc, SCHED_RR, 30, mysched);

    std::cout << "Program is running" << std::endl;
    // Start streaming the camera
    pipe.start(cfg);
    // Synchronization tasks
    while (stop == 0)
    {
        if (gpio_1.read() != 0)
        {
            // start camera thread
            queue.enqueue(pipe.wait_for_frames());

            // start read imu
            
            for (int i = 0; i < 10; i++) // i < 5 if camera is set at 30 Hz, i < 10 at 15Hz
            {
                auto start_timer = std::chrono::steady_clock::now();
                IMU.update();
                unsigned long imu_time = std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::system_clock::now().time_since_epoch()).count();
                IMU.getAccelerometer(&ax, &ay, &az);
                IMU.getGyroscope(&gx, &gy, &gz);
                temp = IMU.getTemperature();
                IMUFile << imu_count << ";" << imu_time << ";" << ax << ";" << ay << ";" << az << ";" << gx << ";" << gy << ";" << gz << ";" << temp << "\n";
                while (std::chrono::duration_cast<std::chrono::microseconds>(std::chrono::steady_clock::now() - start_timer).count() < 5000)
                    ;
            }
            imu_count++;
            
        }
    }
    // close all files and device
    tc.join();
    pipe.stop();
    IMUFile.close();
    ImageTimeFile.close();
    

    std::cout << "\nEXIT SUCCESS!" << std::endl;
}`

But yet, I am still investigating the problem. Hope it was not from your camera 
(I tested your camera like 20 times a day. Can it be degraded?)

RealSenseCustomerSupport, 2019-05-15:
----------------------------------------------

@CaoKha Thanks for the update! Please keep update here when you get more clues. Thanks!

CaoKha, 2019-05-17:
Your camera works perfect at 15 Hz on Udoo x86. I think the problem are the kernel and scheduler policy of Ubuntu. But can you support more specific kernel like in Rock Pi 4 ? I think its difficult to implement your camera in project that require timing precision.

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@Caokha What's the kernel version of Rock Pi 4 in your hand?

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@CaoKha Any other questions about this ticket? Looking forward to your update. Thanks!

CaoKha, 2019-06-17:
Hi,  I think I should close this topic because I don't want to go deep into reverse engineering with kernel development. Anyways, the problem is solved. I don't have any more questions.

closed 2019-06-17 by CaoKha