realsenseai/librealsense #3829: Only a subset of cameras are detected (multi-camera configuration)

evercoast-dev, 2019-04-23:
Camera Model: D400
Firmware Version: 5.10.03
Operating System: Ubuntu
Kernel Version: 4.15.0-43-generic (we see the problem across a variety of kernel versions)
Platform: PC
SDK Version: 2.19.0
Language: C++
Segment: VR/AR

Only a subset of cameras are detected (multi-camera configuration).

On a system with 12 D400 series cameras plugged in simultaneously:
- After hardware resetting the cameras, system sees zero (or a subset) of the total  number of devices. Occasionally, non-repeatable, but regularly occurring.
- After rebooting, the system then is able to see 7 of the 12 devices.

The solution is to physically unplug and then reconnect the set of USB cables at the back of the computer. This procedure is very inconvenient. We really need a fix for this as these machines may not be accessible to an operator in the future (they may be located in a locked cabinet, for instance).

Note: We adhere to all the USB specs and power requirements and are able to operate these cameras at peak spec under regular conditions. This is an intermittent problem.

MartyG-RealSense, 2019-04-23:
Are you connecting the cameras to a single PC with a USB hub with many ports such as a 12 port hub, please?

Doing hardware reset with scripting is the obvious way to reset without physical unplug-replug, but you say that when resetting in your setup, not all of the 12 cameras reset successfully.

A solution may be to iterate through all the cameras, reset them and wait for completion of reset before continuing.  A RealSense community member called West a contributed a script for doing this.

https://forums.intel.com/s/question/0D70P0000068r7PSAQ/d435-streams-work-once-then-crash?language=en_US#558651

evercoast-dev, 2019-04-23:
Thanks Marty.

- Yes, single PC.
- I can't describe the hardware configuration further, except to say that it's working great.
- We are running proprietary software.

Our typical process is to optionally [user's choice] hardware reset all the cameras, then enable the streams and start the pipeline. Again, usually works great.

Also, to clarify, the devices become invisible to the driver, so, when this bug happens:
_rs2::device_list devices = context.query_devices();
devices.size()_ <- would be less than the # of sensors actually physically plugged into the machine.
Hence the unplug, reconnect mechanism is necessary.

The issue here is that the mode of failure is catastrophic and the user may not have access to the machine in our use case. Additionally, unplugging and reconnecting 12 or more sensors is not an acceptable solution for our customers, even if their operators to have access to the machine. I'd hate to write that into our user documentation.

MartyG-RealSense, 2019-04-23:
Having that many cameras attached to a single PC is not usually ideal unless it is a powerful enterprise-grade PC rather than a consumer desktop one.  This is because each camera added in a multi-camera setup causes resources to be consumed.  For example, in a single-PC setup where 4 cameras were connected, a PC with an i7 processor would be recommendable in order to cope with the strain on the hardware.

As you said though, your application is working fine when the cameras are all detectable, so it doesn't sound as though the hardware spec is a problem for you.

My personal hunch for the problem would be a power instability on the USB port, causing the camera connection to drop and making it undetectable / unactivatable.  RealSense cameras are sensitive to the state of the USB port.  If a hub was being used, a mains electricity powered hub can increase stability of power on USB ports.

If the system can see up to 7 cameras of the 12 every single time though after a reboot of the PC and it is never more or less than 7, that would suggest that something other than USB power instability is happening.

It is midnight now in my time zone, so I have to leave off for now, unfortunately.  If someone else does not comment on your case before then, I will be happy to pick up the case when I am back in 7 hours time if you leave a comment below.  Thanks!

evercoast-dev, 2019-04-24:
Good morning, Marty!

The hardware isn't the problem; everything you said about PC resource consumption is correct, including power draw. We're an OEM with a major PC manufacturer and work with their R&D unit to test bus/bridge performance and so forth. We also have a power electronics team and are capable of measuring the power draw from each camera individually and the entire array of cameras as a whole. We are certain that all cameras are adequately powered and are otherwise not lacking for bandwidth or resource.

I think this is a bug in the driver.

MartyG-RealSense, 2019-04-24:
Disconnection issues have occurred with RealSense cameras even on enterprise servers with very high watt PSUs.  So it is not so much a matter of the camera's power draw as a power frequency glitch on the port.  Not all USB ports are made equal despite there being a USB specification, and so a camera that works fine on one make of PC may not work properly on another manufacturer's PC.  

The quality grade of USB cable used can also be a factor in stability.  Unless a high grade cable such as an industrial grade one is used, the longer the cable is then the higher the chance of instability.   On an ordinary cheap cable, I would not have expectations of the camera working on a cable longer than 2 meters.

Having said all that, I have not personally dealt with setups larger than about 6 cameras, so one of the Intel staff on this forum such as @dorodnic may be able to advise you better on this subject.

evercoast-dev, 2019-04-24:
Right, however the bug, as described in the original message, manifests following a hardware reset of all the cameras which we invoke using the API. We are not discussing intermittent dropping of cameras, to your note about power line conditioning. I do not believe that we have power issues, as the systems are able to perform at peak for hours at a time under normal use.

The bug relates to the dropping of cameras specifically following a hardware reset. It seems as though the cameras briefly disconnect, and sometimes not all of the cameras reconnect. Under which circumstance the only remedy is seemingly to unplug and reconnect them.

@dorodnic, any support your able to offer would be most appreciated.

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

Hi @evercoast-dev,
In the scenario of camera dropping (disappearing) after calling the hardware_reset(), is the total number of cameras show up correctly on the USB bus? Can you share the result of the lsusb?
Perform a firmware upgrade is also strongly recommended to upgrade the latest firmware version and see if the problem persists as well as take advantages of other fixes.
https://downloadcenter.intel.com/download/28724/Latest-Firmware-for-Intel-RealSense-D400-Product-Family

evercoast-smm, 2019-05-02:
OK, here we go. Updated libRealsense t 20.0.0, upgraded firmware (all cameras match) to 05.11.06.200, the system has 12 cameras plugged in, lsusb reports 12 Intel devices, then rs-enumerate devices sees none of them:

evercoast@evercoast:~/Evercoast/Mavericks$ lsusb
Bus 050 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 049 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 048 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 047 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 046 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 045 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 044 Device 012: ID 8086:0ad3 Intel Corp. 
Bus 044 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 043 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 042 Device 012: ID 8086:0ad3 Intel Corp. 
Bus 042 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 041 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 040 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 040 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 039 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 038 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 038 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 037 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 036 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 035 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 034 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 034 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 033 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 032 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 032 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 031 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 030 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 030 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 029 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 028 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 027 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 026 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 026 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 025 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 024 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 024 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 023 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 022 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 021 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 020 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 019 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 018 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 017 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 016 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 015 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 014 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 013 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 012 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 011 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 012: ID 8086:0ad3 Intel Corp. 
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 012: ID 8086:0ad3 Intel Corp. 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 009: ID 03f0:424a Hewlett-Packard 
Bus 001 Device 012: ID 03f0:3f4a Hewlett-Packard 
Bus 001 Device 007: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
verizon@verizon:~/Evercoast/Mavericks$ 
verizon@verizon:~/Evercoast/Mavericks$ lsusb | grep Intel
Bus 044 Device 012: ID 8086:0ad3 Intel Corp. 
Bus 042 Device 012: ID 8086:0ad3 Intel Corp. 
Bus 040 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 038 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 034 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 032 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 030 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 026 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 024 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 010 Device 013: ID 8086:0ad3 Intel Corp. 
Bus 008 Device 012: ID 8086:0ad3 Intel Corp. 
Bus 006 Device 012: ID 8086:0ad3 Intel Corp. 
verizon@verizon:~/Evercoast/Mavericks$ lsusb | grep Intel | wc -l
12
evercoast@evercoast:~/Evercoast/Mavericks$ rs-enumerate-devices 
No device detected. Is it plugged in?

Thanks!

evercoast-smm, 2019-05-02:
Update on the previous post, following a reboot we see 11 Realsense cameras:

`evercoast@evercoast:~$ lsusb`
`Bus 050 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 049 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 048 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 047 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 046 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 045 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 044 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 043 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 042 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 042 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 041 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 040 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 040 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 039 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 038 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 038 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 037 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 036 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 035 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 034 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 034 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 033 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 032 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 032 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 031 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 030 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 030 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 029 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 028 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 027 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 026 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 026 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 025 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 024 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 024 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 023 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 022 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 021 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 020 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 019 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 018 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 017 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 016 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 015 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 014 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 013 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 012 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 011 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 010 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 008 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 006 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 002 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. `
`Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. `
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 001 Device 004: ID 0bda:5411 Realtek Semiconductor Corp. `
`Bus 001 Device 009: ID 03f0:424a Hewlett-Packard `
`Bus 001 Device 010: ID 03f0:3f4a Hewlett-Packard `
`Bus 001 Device 007: ID 05e3:0610 Genesys Logic, Inc. 4-port hub`
`Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver`
[... 59 lines cut ...]
`Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 010 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 008 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 006 Device 002: ID 8086:0ad3 Intel Corp. `
`Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 002 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. `
`Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. `
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 001 Device 004: ID 0bda:5411 Realtek Semiconductor Corp. `
`Bus 001 Device 009: ID 03f0:424a Hewlett-Packard `
`Bus 001 Device 010: ID 03f0:3f4a Hewlett-Packard `
`Bus 001 Device 007: ID 05e3:0610 Genesys Logic, Inc. 4-port hub`
`Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver`
`Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub`
`Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub`
`Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. `
`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`evercoast@evercoast:~$ lsusb | grep Intel | wc -l`
`12`

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

Is this problem persist after subsequent reboot/hardware reset or system cold boot?

What is the length(s) of the USB cable used for these cameras? Just for testing purpose, is there any improvement or difference by using the 1 M USB cable that comes with the camera?

evercoast-smm, 2019-05-27:
Often times we notice that when cameras "go missing" that a reboot is likely to fix the problem. However, infrequently, but often enough that we're posting a bug report, it happens that some cameras remain inactive following a reboot. It's then that the unplug/reconnection is necessary.

Regarding cable length, it's not really viable to get that many cameras plugged in using only the supplied cable. But, I can say that we do normally operate without dropping frames on all cameras simultaneously. Also, our USB extensions are professional quality.

RealSenseCustomerSupport, 2019-06-27:
----------------------------------------------

I am just wondering if there's any update/potential fix that you might have discovered to workaround the issue?
Thank you.

RealSenseCustomerSupport, 2019-07-03:
----------------------------------------------

Closing the ticket. Please feel free to reopen if problem remains.
Thank you.

closed 2019-07-03 by RealSenseCustomerSupport

Ashok93, 2020-06-15:
This issue is very persistent and why is it closed? 
It is so easily reproducible by plugging and unplugging real sense multiple times and the camera devices are no more detected. 
Some times I see `/dev/video0`, `/dev/video1` only mounted, sometimes only one of them is mounted and sometimes nothing at all. Restarting the PC fixes it but definitely is not the solution. Please reopen this issue.

MartyG-RealSense, 2020-06-15:
@Ashok93  An issue may be closed by RealSense team members if no further comments are received on the subject for 3 or 4 weeks and it does not have a subject that requires keeping it open for long-term tracking.  Closed subjects are still findable by web search.

In this particular case, improvements to multicam support were released in the latest **2.35.2** SDK version.  The changes may not correct every multicam problem (such as ones related to specific USB hub models) but it may help if the problem is related to the rs2::pipeline.  If you are able to use 2.35.2 then I encourage you to try it and see if it resolves your need to unplug and replug cameras.