realsenseai/librealsense #3830: “librealsense2/rs.hpp”: No such file or directory

YH53, 2019-04-24:
----------------------------------------------------------------------------------------------------

| Required Info                                                                     
| Camera Model                       | {  D415 } | 
| Operating System & Version |   {Win 10   | 

| SDK Version                        |  2.0                |
| Language                            |   C++            | 
Platform: visual studio 2015

I am new to it, when I create a new project，it appears:Fatal error C1083: Cannot open include file: "librealsense2/rs.hpp": No such file or directory.   How can I add this lib？

off99555, 2019-04-24:
You have to include the header files path into your Visual Studio.
The path that you need to include will be your real sense SDK installation folder with `include` name.
https://stackoverflow.com/questions/335408/where-does-visual-studio-look-for-c-header-files

This way Visual Studio will be able to find the header files that you mention in your code.

dorodnic, 2019-04-28:
Hi @YH53 
The simplest way is to open one of the examples provided with the SDK installer and start building from there.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@YH53 Did you get it through? Could you please update? Thanks!

YH53, 2019-05-09:
@off99555  Thanks a lot, I've got it, however, the same question happened to opencv, when I wang to use opencv, it appears Cannot open include file: "opencv2/opencv.hpp": No such file or directory. How can I add this lib？Thank you.

off99555, 2019-05-09:
In order for visual studio to knows where the opencv.hpp file is, it usually searches in your project folder.
If it doesn't find anything, it will search in the **Include Directories** that you can specify in the project Properties.
What you can do is to either copy the opencv.hpp file to your project, or specify **Include Directories** to point to where opencv.hpp lives.
---
What I've learned is that the fundamental of how C++ compiling/linking work in Visual Studio will eliminate almost all of your confusions. 

I suggest you read this: https://docs.microsoft.com/en-us/cpp/build/walkthrough-creating-and-using-a-dynamic-link-library-cpp?view=vs-2017
I've summarized the article above here: https://gist.github.com/off99555/a9fda0f3e19262b57916ed0d304f267d

closed 2019-05-20 by YH53

votecoffee, 2020-10-22:
There are some ready to use project default property pages you can import for both RealSense and OpenCV at the following location:
C:\Program Files (x86)\Intel RealSense SDK 2.0
Open the property manager for your project (View > Property Manager or View > Other Windows > Property Manager)
Add intel.realsense.props and opencv.props to each build configuration.
This will add the misc bin/include/lib/linker settings you need for most things.