realsenseai/librealsense #3832: How to save pointcloud with all dimensions?

arjunskumar, 2019-04-24:
How can save a pcd file with all dimesnsions using pyrealsense2 library?
For eg-->

pcl_viewer cloud_bin_0.pcd 
The viewer window provides interactive commands; for help, press 'h' or 'H' from within the window.
> Loading cloud_bin_0.pcd [PCLVisualizer::setUseVbos] Has no effect when OpenGL version is ≥ 2
[done, 268 ms : 198835 points]
Available dimensions: x y z rgb normal_x normal_y normal_z curvature

dorodnic, 2019-04-28:
What do you mean by "all dimensions"? `pyrealsense2` does not provide the capability of saving to pcd. 
`points` object contain a collection of x, y, z coordinates together with u, v coordinates for texture mapping.

arjunskumar, 2019-04-29:
@dorodnic : I need to get x y z rgb normal_x normal_y normal_z curvature of the point cloud obtained from realsense D435i for doing ICP (point to plane ICP requires vertex normal)

dorodnic, 2019-04-30:
normal_x normal_y normal_z curvature are not provided by `pyrealsense2`

arjunskumar, 2019-04-30:
@dorodnic Do you know any other way so that I get all information of point cloud?

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

Hi arjunskumar,

One of them that you may try is strawlab pcl: https://github.com/strawlab/python-pcl, if not yet.

Thanks!

arjunskumar, 2019-05-14:
Hi @RealSenseCustomerSupport , I was able to save PCD with Open3D

closed 2019-05-14 by arjunskumar