realsenseai/librealsense #3834: Merge point clouds- data loss

Pratiksha2692, 2019-04-24:
Hi,

| Required Info                         |        

Camera Model | {  D435 }
-- | --
Firmware Version | (05.10.06.00)
Operating System & Version | Linux (Ubuntu 16.04)
Platform | PC
SDK Version | { legacy / 2.. }
Language | {python }
Segment | {3d reconstruction}

### Issue Description
I am using 4 realsense cameras for 3d reconstruction of object. I used export_ply_expmple.py code for creating 4 ply file. All cameras are covering all the required data(there is no any missing data). When I capture a rectangular shape object all the data is captured by cameras but when I captured a triangular shape object data is missing. I am getting a small gap between the merge of the two cameras ply file as below.
[image]

Will camera configuration and setting helps to capture better ply file? When I observed the rgb and depth frame all data is visible. 

Kindly help me out.

ev-mp, 2019-04-30:
@Pratiksha2692 , can you be more specific and mark the  "problematic" are in the screenshot. 
Also please elaborate how are the devices arranged, what is the exact shape of the "triangular" object and where is that object located with regards to each of the camera - a drawing/diagram/snapshot would really be helpful.

Pratiksha2692, 2019-05-06:
Thanks a lot for the reply. 

I will send the drawing of the camera position in 1-2 days.

Pratiksha2692, 2019-05-10:
Hi,

Actually the attached screenshot is the human foot. It is the back view of foot. I am getting the missing data at the back of foot because of some unknown triangular geometry of it. But whereas for circular/ rectangular box no data is missing.

[image]

I had made the rectangular setup like 70cmX70cm.  4 cameras at 4 corner at some height and angle 35 degree. AOI is 45cmX45cm. All cameras are able to capture all the data. 

Kindly advise.

RealSenseCustomerSupport, 2019-05-15:
----------------------------------------------

@Pratiksha2692  Could you please provide the original snapshot before the merge? And what's the details about "angle 35 degree"? If you can provide your setup snapshot, that will be more helpful to understand. Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@Pratiksha2692 Did you get it through? Could you please update if you still need support? Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@Pratiksha2692 Any update from your side? Please provide more information if you still need support on this. Thanks!

Pratiksha2692, 2019-06-04:
Resolved

closed 2019-06-04 by Pratiksha2692