realsenseai/librealsense #3836: Is it possible to use a multicam configuration in RealSense Viewer with s6 D415 cameras? Also, is it possible to use a multicam configuration in RealSense Viewer using one D415 and one D435? Why does not work the Sync option in Viewer? How can I bring together the streams from all cameras in one image in RealSense Viewer?

TDVL, 2019-04-24:
Hello,

I would like to build a multicam system with 6 D415 cameras, and I would like to join, combine the images from these 6 cameras in one image, to get a larger field of view. I read the white paper about multicam configuration and I saw the example-demo at the and of the paper in ROS (I think). How can I do this with RealSense Viewer in Windows? Is it possible? First, I want to try it with 2 cameras, one D415 and one D435 (I have them at home), then I will build a system with 6 cameras. Is it possible to test this functionality with these 2 cameras? Also, I have one USB3 port, the others are USB2, is this a problem when I want to use a multicam mode?

Thank you in advance.

Best regards.

ev-mp, 2019-04-24:
@TDVL hello,
I reformatted the topic to make it in line with the forum convention by moving the bulk of the questions into the designated description part. 
With regards to the questions raised - 
>  Q. is it possible to use a multicam configuration in RealSense Viewer using one D415 and one D435 ?

Yes. Realsense viewer is capable stream multiple Realsense devices simultaneously.
The first device will always be automatically added to the viewer. In order to add additional devices press the "Add Source" button in the upper left and select the devices from the list.
[image]
In addition to realsense devices it is also possible to add non-Realsense UVC devices (Webcams) as well as recorded sessions (playback devices).

> Q. Why does not work the Sync option in Viewer?

It is not clear what do you exactly by "not working" - is it the UI appearance or the actual frames being out of sync?
Feature-wise the Viewer's syncer is controlled by a toggle button and can be altered at any time. There are use-cases that the viewer will toggle it on/off internally, e.g. switching to 3D view that requires synchronized Depth+Color framesets. 

> Q. How can I bring together the streams from all cameras in one image in RealSense Viewer?
The viewer redirect each frame to a separate viewport for rendering. Each viewport manages its own OpenGl render texture. There is an additional demo that provides a similar functionality - `rs-multicam`  - it is also more accessible as it is organized as a tutorial.

Combining (or stitching)  multiple depth sensors into a single coherent scene is an advanced use-case that has already been demonstrated with a dedicated demo that you mention, but it is not in the SDK's portfolio (#3222, #3498) 

> Q. Also, I have one USB3 port, the others are USB2, is this a problem when I want to use a multicam mode?

Using USB2 instead of USB3 limits the data bandwith, though it can be partially compensated by providing Depth High-Res with low (6,10,15) fps rates. 
Even in today's most modern PCs it is hard to find more than two USB3 controllers (don't confuse with ports). An in case you're planning to use more than two devices, then multiplexing the USB3 port with an external self-powered USB hub(s) is highly envisaged.

TDVL, 2019-04-25:
Hello ev-mp,
Thank you for your help. I have set the two camera mode, here is the screenshot.
[image]
But, I need to stitch, put together images from two cameras now for testing, later 6 cameras. Depth and RGB streams are important to me for image processing. I need a solution like on this screenshot. Naturally, I will need the XYZ coordinates too.
[image]
Can I do this in the RealSense Viewer? In this demo the cameras were set geometrically appropriate in space to get the images well?
Also, the software sync. is used for synchronize what? RGB and depth streams of the same scene?
The USB is OK, I reduced the resolution and the fps rate.
Thank you in advance.
Kind regards

ev-mp, 2019-04-28:
@TDVL hi,
Librealsense provides the infrastructure required to query data from multiple devices. And in the current SDK's hierarchy, a device is the highest abstraction available.
Arranging the devices geometrically to provide a specific depth coverage (as in the attached demo screenshot), and later stitching them to produce a continuous (panoramic) scene is an advanced feature that requires additional software layer to manage the extrinsic parameters 

Syncer-wise - the SDK implementation work best with the frames being timestamped by synchronized clocks, e.g. Depth/Color/IMU streams for D435i model. In the case of multiple depth devices this constrain could be temporally replaced by removing the metadata patch, thus  enforcing Host-generated timestamps for all incoming streams.
Note that the devices are built with a provision for inter-cam Depth acquisition HW sync [since v2.13](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2130)
You may find more info in the [multi-cam white paper](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration).
Also see #3504

TDVL, 2019-04-29:
Hello ev-mp,
Thank you for your help.
Hence, the demo is not available for testing (multi-cam demo)? The GUI of that demo is in ROS (or in something else)? I see the controls, options for viewing, transform, the active cameras, depth preset,the joined RGB and Depth images, etc. 
Can I manage everything like on this screenshot (from the white paper) in the Win10 RealSense Viewer, in the GUI? If not, where can I do this, is there a GUI for this? My goal is to get the images with some RealSense software (for example with Viewer, because I am a Win10 user, this is the easiest way for me) from camera, or from multiple cameras and process them in Matlab (this is the best way for signal and image processing) or in other programming language. 
I am using SDK2.19.1. I saw there is a new release. Is there anything new in latest release about this?
I think the stitching was done with feature based panorama stitching like in Matlab. Is it? 
Thank you in advance.
Best regards

ev-mp, 2019-04-30:
@TDVL hi, the full solution for what you're looking is currently not covered by the SDK's demos.

I'll try to break the referenced white paper prerequisites into blocks so you'd be able to assess the gaps:
- Streaming of multiple devices &#10004;
- Displaying depth data from multiple devices &#10004;
- API for Depth&RGB coordinate system transformation (align) &#10004;
- Pointcloud generation and coordinate system transformation &#10004;
- Cables for inter-cam HW sync &#10060;
- Mechanical fixture to position and calibrate the cameras &#10060;
- Stitching multiple depth streams/pointclouds into a single 3D scene  &#10060;

Hope it  makes it clear now

FlyingSpringrol, 2019-08-26:
I've been working on a project involving stitching depth information. According to this paper (https://journals.sagepub.com/doi/pdf/10.1177/1729881417695560) a way to do it would be to align depth and color images, then use a homography solving method on the color images (such as those used for panoramas: like this openCV example) (https://www.pyimagesearch.com/2016/01/11/opencv-panorama-stitching/) to create a larger FOV output.

Resays, 2019-11-22:
@FlyingSpringrol Hey, any luck?
I can stitch the RGB image only but can't stitch the depth images came straight from RealSence as .png extension.
if you have a solution in code please share, I am working with python and OpenCV
Looking to hear from you
Thanks

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Hi @ TDVL This change request is approved. We will implement this.