realsenseai/librealsense #3838: No Module name pyrealsense2 on Jetson Nano

ChaseLewis, 2019-04-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Operating System & Version |   {Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  Jetson Nano  |
| SDK Version                            |  { realsense2 }                          | 
| Language                            |  { python }                          | 
| Segment			|  {Robot }                          | 

I know there are similar questions but none of the solutions I've found in the other threads has worked. So starting a thread to see if I can get assistance.

So I got it to build, but now it says 'no module named pyrealsense2'. If I use rs-enumerate devices the camera is found so it has something to do with the python installation after building. The installed samples also work through command line.

I set the executable to python3 and used sudo make install so I'm a bit confused what I'm doing wrong. I'm trying to run the script with python 3.6.7. Any ideas on how to get the python module in a position I can use it?

Build Output
```
chaserlewis@chaserlewis-desktop:~/librealsense/build$ sudo make
[  1%] Built target fw
[  7%] Built target tm
[ 15%] Built target realsense-file
Scanning dependencies of target realsense2
[ 16%] Linking CXX shared library librealsense2.so
[ 46%] Built target realsense2
[ 51%] Built target pybackend2
[ 51%] Linking CXX shared library pyrealsense2.cpython-36m-aarch64-linux-gnu.so
[ 52%] Built target pyrealsense2
[ 59%] Built target glfw
[ 59%] Linking CXX executable rs-hello-realsense
[ 59%] Built target rs-hello-realsense
[ 59%] Linking CXX executable rs-software-device
[ 59%] Built target rs-software-device
[ 59%] Linking CXX executable rs-capture
[ 60%] Built target rs-capture
[ 60%] Linking CXX executable rs-callback
[ 61%] Built target rs-callback
[ 62%] Linking CXX executable rs-save-to-disk
[ 62%] Built target rs-save-to-disk
[ 62%] Linking CXX executable rs-multicam
[ 63%] Built target rs-multicam
[ 63%] Linking CXX executable rs-pointcloud
[ 64%] Built target rs-pointcloud
[ 65%] Linking CXX executable rs-align
[ 66%] Built target rs-align
[ 67%] Linking CXX executable rs-align-advanced
[ 68%] Built target rs-align-advanced
[ 69%] Linking CXX executable rs-sensor-control
[ 69%] Built target rs-sensor-control
[ 69%] Linking CXX executable rs-measure
[ 70%] Built target rs-measure
[ 70%] Linking C executable rs-depth
[ 71%] Built target rs-depth
[ 72%] Linking C executable rs-color
[ 72%] Built target rs-color
[ 72%] Linking C executable rs-distance
[ 73%] Built target rs-distance
[ 73%] Linking CXX executable rs-post-processing
[ 75%] Built target rs-post-processing
[ 75%] Linking CXX executable rs-record-playback
[ 77%] Built target rs-record-playback
[ 77%] Linking CXX executable rs-motion
[ 78%] Built target rs-motion
[ 79%] Linking CXX executable rs-pose
[ 79%] Built target rs-pose
[ 79%] Linking CXX executable rs-pose-predict
[ 79%] Built target rs-pose-predict
[ 80%] Linking CXX executable rs-pose-and-image
[ 80%] Built target rs-pose-and-image
[ 81%] Linking CXX executable rs-trajectory
[ 81%] Built target rs-trajectory
[ 81%] Linking CXX executable rs-ar-basic
[ 81%] Built target rs-ar-basic
[ 82%] Linking CXX executable rs-terminal
[ 83%] Built target rs-terminal
[ 84%] Linking CXX executable rs-fw-logger
[ 86%] Built target rs-fw-logger
[ 86%] Linking CXX executable rs-enumerate-devices
[ 87%] Built target rs-enumerate-devices
[ 87%] Linking CXX executable realsense-viewer
[ 92%] Built target realsense-viewer
[ 92%] Linking CXX executable rs-data-collect
[ 92%] Built target rs-data-collect
[ 92%] Linking CXX executable rs-depth-quality
[ 97%] Built target rs-depth-quality
[ 97%] Linking CXX executable rs-rosbag-inspector
[ 99%] Built target rs-rosbag-inspector
[100%] Linking CXX executable rs-convert
[100%] Built target rs-convert

```

```
chaserlewis@chaserlewis-desktop:/usr/local/lib$ ls
cmake                    pybackend2.cpython-36m-aarch64-linux-gnu.so
libfw.a                  pybackend2.cpython-36m-aarch64-linux-gnu.so.2
librealsense2.so         pybackend2.cpython-36m-aarch64-linux-gnu.so.2.20.0
librealsense2.so.2.20    pybackend2.cpython-36m-aarch64-linux-gnu.so.2.21.0
librealsense2.so.2.20.0  pyrealsense2.cpython-36m-aarch64-linux-gnu.so
librealsense2.so.2.21    pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.21
librealsense2.so.2.21.0  pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.21.0
librealsense-file.a      python2.7
libtm.a                  python3.6
pkgconfig
```

PythonPath
```
chaserlewis@chaserlewis-desktop:/usr/local/lib$ echo $PYTHONPATH
:/usr/local/lib

```
How I called CMAKE
`chaserlewis@chaserlewis-desktop:~/librealsense/build$ sudo cmake ../ -DBUILD_PYTHON_BINDINGS=TRUE -DPYTHON_EXECUTABLE=/usr/bin/python3`

Python Script

```
import cv2
import numpy as np
import pyrealsense2 as rs
try:
  
    pipeline = rs.pipeline()
    pipeline.start()

    while True:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()

        if not depth: 
            continue

        img = np.asanyarray(depth.as_frame().get_data())
        cv2.imshow('depth',img)
        cv2.waitKey(1)
except:
    print("Error!")
```

lramati, 2019-05-12:
Could you try copying the .so files to Python's dist-packages folder? (Can
be found with `import site; site.getsitepackages()` from within the
interpreter)

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@ChaseLewis Did you get through the issue? Looking forward to your update. Thanks!

ChaseLewis, 2019-05-21:
No I have been using the C++ bindings because the python bindings do not work. I had already tried Iramti's suggestion. Those files are in the same place all my other python files. It does not work. It could be the script doesn't rename them or something as they have very odd names with cython in them. I am unsure. The instructions as provided do not work, so was hoping for feedback that could assist.

kat-mcg-lin, 2019-05-28:
I'm having the exact same issue! Did you figure it out?

my issue is here: #4089

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@ChaseLewis @kat-mcg-lin  I followed the steps in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python and couldn't reproduce the issue you reported on JetsonNano in my hand.  Could you please double check you followed the steps and if you're under python3 when execute?

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@ChaseLewis @kat-mcg-lin  Any update for this issue? Did you get through now? Looking forward to your update. Thanks!

kat-mcg-lin, 2019-06-13:
@RealSenseCustomerSupport I have gotten it to work!

RealSenseCustomerSupport, 2019-06-14:
----------------------------------------------

@kat-mcg-lin  That's great! Thanks for the update!
@ChaseLewis  Any good news from your side? Thanks!

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@ChaseLewis If no other questions, will close the ticket accordingly. Thanks!

closed 2019-06-21 by RealSenseCustomerSupport

danial880, 2020-07-11:
> @RealSenseCustomerSupport I have gotten it to work!

how did it worked? i am having same problem

PatrickNa, 2020-07-17:
@kat-mcg-lin I also would be interested in knowing how you fixed that. I followed the instruction here: [https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). The ```realsense-viewer``` is working, but I also receive the error ```ModuleNotFoundError: No module named 'pyrealsense2'```

aloev, 2021-04-08:
@RealSenseCustomerSupport Has anyone solved this ? I can't run some scripts because I get "ModuleNotFoundError: No module named 'pyrealsense2'" . I would like to get this to work so if anybody knows how please share the fix.

BessPtck, 2021-10-01:
Also having this problem, Jetson Nano Ubuntu 18.04LTS installed following: https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation

the realsense-viewer workes as does compiling for c++