realsenseai/librealsense #3840: camera angle of view rotation

nblog, 2019-04-25:
| | |
|---|---|
|**librealsense**|2.19.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|837212070379|
|**Firmware Version**|05.10.13.00
255.255.255.255|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

realsense-viewer tools
can be add camera angle of view rotation? 

Thank you and regards.

dorodnic, 2019-04-28:
Could you please provide more details? I'm not sure what that means

nblog, 2019-04-28:
Intel.RealSense.Viewer.exe   

can be depth stream and color stream view rotation?

[image]

like this
[image]

[image]

forgive my poor english

RealSenseCustomerSupport, 2019-05-03:
----------------------------------------------

Hi @nblog,

What rotation angle do you expect in realsense-viewer? 90 degree and 180 degree?

nblog, 2019-05-03:
yes.
90 degree  or -90 degree or 180 degree.
Because sometimes the camera is upright or upside down.

nblog, 2019-05-03:
> 
> 
> Hi @nblog,
> 
> What rotation angle do you expect in realsense-viewer? 90 degree and 180 degree?

yes.
90 degree or -90 degree or 180 degree.
Because sometimes the camera is upright or upside down.

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

We have filed a feature request for internal review. If any update, will let you know.

kristawolffe, 2020-12-11:
I'd love an option for this as well... I am forced to mount my D415 "upside down" due to cabling issues, and being able to use the Realsense Viewer as a quick check of things would be wonderful.

tomatac, 2021-04-16:
It would be nice to have better control of the rotation of the camera.
 - standard views: top, bottom, front, back, left right
 - ability to make a cross section thru the cloud points
This should be relatively easy to implement.

RealSenseCustomerSupport, 2022-04-23:
----------------------------------------------

Hello @nblog  My name is Marty Grover and I will be handling your case.

Could you let me know whether you still require assistance with this case, please?

Thanks very much,

Marty Grover
Intel RealSense Support Engineer

MartyG-RealSense, 2022-04-23:
Hi everyone, I would like to provide an update about this case.  Whilst the feature request that Intel created - as reported above at https://github.com/IntelRealSense/librealsense/issues/3840#issuecomment-498126843 - is still active, there is no progress to report regarding development of the feature.

nblog, 2022-04-23:
> Hi everyone, I would like to provide an update about this case. Whilst the feature request that Intel created - as reported above at [#3840 (comment)](https://github.com/IntelRealSense/librealsense/issues/3840#issuecomment-498126843) - is still active, there is no progress to report regarding development of the feature.

I have solved it with OpenCV, but I will be more happy with the support of ["Intel. Realsense. Viewer.exe"](https://github.com/IntelRealSense/librealsense/releases)

MartyG-RealSense, 2022-04-23:
A RealSense team member explains in the image rotation case at https://github.com/IntelRealSense/librealsense/issues/6023 that rotating the depth frame would be difficult because all calibration data and other parameters would have to be updated too.

Other RealSense users in that case solved the issue with OpenCV like you did or with Open3D at https://github.com/IntelRealSense/librealsense/issues/6023#issuecomment-785071263

closed 2023-01-11 by nblog