realsenseai/librealsense #3841: Failed installation: Could not open device. Last Error: Device or resource busy

Suman7495, 2019-04-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2019-04-29:
@Suman7495 , please provide some logs:
- `lsusb`
- `rs-enumerate-devices`
- `dmesg`

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@Suman7495 Could you please provide more information as ev-mp mentioned? Thanks!

- `lsusb`
- `rs-enumerate-devices`
- `dmesg`

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@Suman7495 Did you get through the issue? Could you please update? Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@Suman7495 Sorry we didn't get response from you for weeks. Will close the ticket at this point. Please re-open it or create new ticket if you still have questions/issues. Thanks!

closed 2019-06-03 by RealSenseCustomerSupport