realsenseai/librealsense #3842: Sparse depth data after multiple pipe.start()/pipe.stop() events

m1halo, 2019-04-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (latest) | 
| Operating System & Version |   {Android 8}  | 
| Kernel Version (Linux Only)    |  3.18.71                                        | 
| Platform                                 | Android  |
| SDK Version                            |  {2.21 }                          | 
| Language                            |  {java }                          | 
| Segment			|  {/Smartphone }                          | 

### Issue Description
Our use case involves acquiring data stream for short periods of time. Typically for 50 frames per session. For each session we call pipe.start() and pipe.stop(). 

This seems to work pretty well, until after some amount of sessions depth data gets very sparse. You can see it has very little detail. RBG image does not seem to be affected. The only way to recover from that state is to inplug USB cable.

Would you have any recommendations for this use case or have you had any issues related to depth data getting sparse?

dorodnic, 2019-04-28:
Hi @m1halo 
Can you please attach a picture of sparse data?
What I suspect to be happening is camera overheating, forcing the firmware to disable the active projector, reducing depth fill-rate. 
If you can reproduce it on a PC, the Viewer communicates such hardware events via a dedicated notification.

m1halo, 2019-04-28:
Hi @dorodnic, here is the example scan result. We will try to reproduce it on PC as well. One additional observation is that each camera start takes longer and longer (eventually up to several seconds just before sparsity is visible).

Would you have any recommendation on solving this? Ie. lower the frame rate or resolution or does camera require a cooldown time between sessions if it is heating up issue.

[
[image]
[image]

](url)

ttoivone, 2019-04-29:
Can you capture the IR streams (or use another IR camera) to see if the IR dot pattern is projected or not? You could check the overheating by getting some cold spray and cooling the camera with that to check if it helps.

If you capture longer periods, does it affect? Maybe the exposure control doesn't have time to adjust in such a short time.

What I have noticed is that the camera front glass gets easily dirty which affects the results and wiping it with a cloth helps, but your problem probably is not the same if unplugging helps.

Did you check if the firmware could be updated?

m1halo, 2019-04-29:
Thanks for tips. We are able to see dot pattern rather steady. With longer capture periods camera gets warm but there is no quality drop (or it might take many hours before it happens). It seems to be happening more often (faster) when those start/stop events happen.

It can be reproduced on latest FW.

Could longer and longer startup time be sign of overheating as well?

RealSenseCustomerSupport, 2019-05-03:
----------------------------------------------

Hi @m1halo,

1. Please also show the left and right IR pictures when the issue happens.
2. Any kernel/librealsense log when the issue happens?
3. Are you able to reproduce this issue using realsense-viewer?
Due to longer and longer startup time, I suspect it has some abnormal communication or abnormal system latency there.

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

Hi @m1halo,

Any update? Is the issue clarified?

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @m1halo,

I will close this if no additional question. If you still have the trouble, feel free to reopen it.

closed 2019-11-01 by RealSenseCustomerSupport