realsenseai/librealsense #3843: Depth viewer D432

scizors, 2019-04-25:
Hello, I open the Depth viewer it is showing that 

please point the camera to a flat wall/surface! 

could any one explain in detail. Should I need to calibrate the camera or it is calibrated automatically

MartyG-RealSense, 2019-04-25:
Are you referring to the 'Depth Quality Tool', please?  This is a tool for running tests on the camera and getting feedback.  If you wish to generate a depth stream, you should be using the 'RealSense Viewer' tool.

RealSense cameras are usually already well-calibrated when purchased, because they are calibrated in the factory (although the IMU component of the D435i model does need to be user-calibrated).  It should only need calibrating if it receives a physical shock such as a hard knock or a drop onto the floor, though users are free to perform a calibration at any time if they believe it is necessary.

In regard to the message that you got, the Depth Quality Tool requires that you point the camera at a flat surface such as a wall and then follow on-screen instructions to carefully rotate the camera until all of the requests that the tool makes have been completed.  The easiest way to do this is to screw the base of the camera onto the tripod that is supplied with the camera and move the camera atop the tripod.  Otherwise, doing this process while holding the camera in your hand can be difficult.

You only need to use the Depth Quality Tool if you need details about the camera's current configuration though.

[image]

scizors, 2019-04-29:
thanks for the reply I am using D435 withot IMU. But sometimes I see this message in the viewer. I think the viewer needs to identify which type it D435 or D435i is and display the message.

closed 2019-04-29 by scizors