realsenseai/librealsense #3846: Distances in a region of interest

HannahLuy, 2019-04-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Platform                                 | Raspberry Pi |
| Language                            |  python 2.7 | 

### Issue Description
I am trying to measure distances with D430 depth module. However, with python, I can only detect the distance to an object that is in a specific location. Is there any way to detect the nearest obstacle in all image or in a region-of-interest?

**Code**
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.infrared, 640, 480,rs.format.y8, 30)
pipeline.start(config)
while True:
     frames = pipeline.wait_for_frames()
     depth_frame = frames.get_depth_frame()
     infrared_frame = frames.get_infrared_frame()
     width = depth_frame.get_width()
     height = depth_frame.get_height()
     dist = depth_frame.get_distance(width/2, height/2)

mzahana, 2019-04-27:
If you can use PCL in ROS, it's relatively easy to get depth in a ROI and do segmentation to find objects.

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

@HannahLuy One similar discussion about it and please refer to https://github.com/IntelRealSense/librealsense/issues/3207

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@HannahLuy Will close the ticket if no other questions. Thanks!

closed 2019-05-28 by RealSenseCustomerSupport