realsenseai/librealsense #3847: I got error while compiling with MAC OS

spk921, 2019-04-26:
I follow all the instruction but end up following errors,

[ 51%] Building C object third-party/glfw/src/CMakeFiles/glfw.dir/context.c.o
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Security.framework/Headers/AuthSession.h:32:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Security.framework/Headers/Security.h:42,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/OSServices.framework/Headers/CSIdentity.h:43,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/OSServices.framework/Headers/OSServices.h:29,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/LaunchServices.framework/Headers/IconsCore.h:23,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/LaunchServices.framework/Headers/LaunchServices.h:23,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreServices.framework/Headers/CoreServices.h:45,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Carbon.framework/Headers/Carbon.h:20,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/cocoa_platform.h:30,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/internal.h:183,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/context.c:28:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Security.framework/Headers/Authorization.h:193:7: error: variably modified 'bytes' at file scope
  char bytes[kAuthorizationExternalFormLength];
       ^~~~~
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGContext.h:18:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGBitmapContext.h:9,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CoreGraphics.h:11,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/ApplicationServices.framework/Headers/ApplicationServices.h:35,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Carbon.framework/Headers/Carbon.h:24,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/cocoa_platform.h:30,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/internal.h:183,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/context.c:28:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGFont.h:53:40: error: initializer element is not constant
 static const CGFontIndex kCGGlyphMax = kCGFontIndexMax;
                                        ^~~~~~~~~~~~~~~
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGContext.h:21:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGBitmapContext.h:9,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CoreGraphics.h:11,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/ApplicationServices.framework/Headers/ApplicationServices.h:35,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Carbon.framework/Headers/Carbon.h:24,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/cocoa_platform.h:30,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/internal.h:183,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/context.c:28:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGPath.h:391:15: error: expected identifier or '(' before '^' token
 typedef void (^CGPathApplyBlock)(const CGPathElement * element);
               ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGPath.h:393:53: error: unknown type name 'CGPathApplyBlock'
 CG_EXTERN void CGPathApplyWithBlock(CGPathRef path, CGPathApplyBlock CF_NOESCAPE block)
                                                     ^~~~~~~~~~~~~~~~
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGPDFDictionary.h:14:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGPDFPage.h:15,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGPDFDocument.h:16,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGContext.h:23,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGBitmapContext.h:9,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CoreGraphics.h:11,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/ApplicationServices.framework/Headers/ApplicationServices.h:35,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Carbon.framework/Headers/Carbon.h:24,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/cocoa_platform.h:30,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/internal.h:183,
                 from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/context.c:28:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGPDFArray.h:103:15: error: expected identifier or '(' before '^' token
 typedef bool (^CGPDFArrayApplierBlock)(size_t index,
               ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/CoreGraphics.framework/Headers/CGPDFArray.h:113:5: error: unknown type name 'CGPDFArrayApplierBlock'
     CGPDFArrayApplierBlock cg_nullable block, void * __nullable info)
     ^~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/sangpil_kim/git/librealsense/third-party/glfw/src/context.c:28:0:
/Users/sangpil_kim/git/librealsense/third-party/glfw/src/internal.h:576:42: warning: extra semicolon in struct or union specified [-Wpedantic]
     _GLFW_PLATFORM_LIBRARY_JOYSTICK_STATE;
                                          ^
make[2]: *** [third-party/glfw/src/CMakeFiles/glfw.dir/context.c.o] Error 1
make[1]: *** [third-party/glfw/src/CMakeFiles/glfw.dir/all] Error 2
make: *** [all] Error 2

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@spk921 Which librealsense version did you test when you get the issue? Thanks!

spk921, 2019-05-05:
@RealSense-Customer-Engineering  I used Master branch. But I follow: https://gist.github.com/svenevs/f00ed3898d2af6248921b63254aa8cc1 this and succeded installation. 
I think the main page should update based on the link which I provided.

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

@spk921 Glad to hear that you got it through. Will look into the link to evaluate. Thanks!

closed 2019-05-16 by RealSenseCustomerSupport

dl1ycf, 2020-07-17:
This is simply an error in the MacOS header files. You cannot define a "const int" variable and make this the length of an array. A "const int" is not a compile-time constant.