realsenseai/librealsense #3848: Unity log location?

stephen-systemfriend, 2019-04-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.20     | 
| Language                            |  unity                           | 

### Issue Description

While running some test code to generate a point cloud, the processing block returned this error:
InvalidOperationException: Error while running PointCloud processing block, check the log for info

but there is no log file anywhere in the project that I can find. I have already defined LRS_LOG_LEVEL="WARNING" in the symbols.
Any idea where this log is or how I turn it on?

MartyG-RealSense, 2019-04-26:
Unity has its own editor log file.  Details for finding and accessing it are in the Unity manual link below.

https://docs.unity3d.com/Manual/LogFiles.html

stephen-systemfriend, 2019-04-26:
OK, follow up then

here's the log. What on earth could the error be?

InvalidOperationException: Error while running PointCloud processing block, check the log for info
 at Intel.RealSense.ProcessingBlock.Process[T] (Intel.RealSense.Frame original) [0x0004e] in <441657b0dbbd43269996475b2c92e0df>:0 
  at Intel.RealSense.ProcessingBlock.Process (Intel.RealSense.Frame original) [0x00001] in <441657b0dbbd43269996475b2c92e0df>:0 
  at RsPointCloud.Process (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource frameSource) [0x00049] in C:\Users\stephen\Documents\GitHub\RealSenseExploration\RealSenseTest\Assets\RealSenseSDK2.0\Scripts\ProcessingBlocks\RsPointCloud.cs:59 
  at RsProcessingPipe.ProcessFrame (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource src) [0x0006b] in C:\Users\stephen\Documents\GitHub\RealSenseExploration\RealSenseTest\Assets\RealSenseSDK2.0\Scripts\RsProcessingPipe.cs:99 
UnityEngine.DebugLogHandler:Internal_LogException()
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at Assets\RealSenseSDK2.0\Scripts\RsProcessingPipe.cs:119)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame()
Intel.RealSense.CustomProcessingBlock:Process(Frame)

I'm creating frames in a software device with the following code, then feeding it to a processing pipe the same way that a realSense device was doing so in the examples.

`using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using Intel.RealSense;
using System;

public class RSSoftwareDevTest : RsFrameProvider
{

    public float[] intrinsicsCoeffics = new float[] { 0, 0, 0, 0, 0 };

    //the device by which we generate our frames
    private SoftwareDevice framegenerator;
    private SoftwareSensor colorGenerator, depthGenerator;
    private VideoStreamProfile colorStream, depthStream;
    private Syncer sync;
    private int imgWidth, imgHeight;

    public int fps = 30;

    void Awake()
    {
        framegenerator = new SoftwareDevice();
        depthGenerator = framegenerator.AddSensor("Depth");
        colorGenerator = framegenerator.AddSensor("Color");
        framegenerator.SetMatcher(Matchers.Default);
        Application.targetFrameRate = fps;
    }
    
    //this is a frame provider
    /// <summary>
    /// Notifies upon streaming start
    /// </summary>
    public override event Action<PipelineProfile> OnStart;

    /// <summary>
    /// Notifies when streaming has stopped
    /// </summary>
    public override event Action OnStop;

    /// <summary>
    /// Fired when a new frame is available
    /// </summary>
    public override event Action<Frame> OnNewSample;

    public Texture2D colorTex;
    public Texture2D depthTex;
    int frameNumber = 0;
    private ushort[] depthData;

[... 129 lines cut ...]
        if (Streaming && OnStop != null)
            OnStop();

        if (ActiveProfile != null)
        {
            ActiveProfile.Dispose();
            ActiveProfile = null;
        }

        Streaming = false;
    }
    void OnDestroy()
    {
        // OnStart = null;
        OnStop = null;

        if (ActiveProfile != null)
        {
            ActiveProfile.Dispose();
            ActiveProfile = null;
        }
    }

}
`

MartyG-RealSense, 2019-04-26:
As a rule of thumb, when debugging I focus on the first errors that are logged.  This is because subsequent errors in the log may not be actually faulty, but rather be triggered by a cascade of problems that is flowing downhill from the earlier error.  The first error in your log is:

**at Intel.RealSense.ProcessingBlock.Process[T] (Intel.RealSense.Frame original) [0x0004e] in <441657b0dbbd43269996475b2c92e0df>:0**

This initial error and the couple of subsequent similar ones seems to cause the program to fail on line 59 of the processing block's point cloud script.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsPointCloud.cs#L59

It then subsequently has a problem at line 99 of the RsProcessingPipe script, presumably because it is relying on the **pb.Process** instruction, which had a problem earlier in the log in the processing block's point cloud script (i.e potentially a bug that is not a bug, but is cascading downwards from the earlier error).

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs#L99

So the key to debugging it is probably to find out what causes the first error:

at Intel.RealSense.ProcessingBlock.Process[T] (Intel.RealSense.Frame original) [0x0004e] in <441657b0dbbd43269996475b2c92e0df>:0

I do not know what that error means though.  One of the Intel team members on this Github may be able to help you with interpreting that.

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

Hi @stephen-systemfriend,

You can try to add below C# code in the script (ex: RsDevice.cs). Then, when you run the unity app, a librealsense log file should be generated for further debug.
 Intel.RealSense.Log.ToFile(Intel.RealSense.LogSeverity.Debug, @"debug.txt");

stephen-systemfriend, 2019-05-10:
Thanks, that works a treat

closed 2019-05-10 by stephen-systemfriend

sophiedixon, 2019-12-23:
Hi @stephen-systemfriend
Did you manage to get to the bottom of this? I'm seeing the same error...

sophiedixon, 2019-12-23:
Our debug log shows:

> 23/12 16:48:11,478 WARNING [9364] (types.cpp:48) Device does not support option Depth Units!

Which makes me think it might be related to this: https://github.com/IntelRealSense/librealsense/issues/5416

sophiedixon, 2019-12-23:
It's working for us using a recorder built-in to https://github.com/keijiro/Rsvfx, but recordings we make using the RealSense Viewer's built-in recorder are throwing the above error.