realsenseai/librealsense #3849: [Feature Request] Official Docker image

hden, 2019-04-26:
Give there are so many build / installation related issues. Why not just provide a base image and a minimal official image for each of the official wrappers?

Benefits:
- Reproducible environment / builds / installation
- Quick access to development environment
- Less complaints on GitHub

A user can always forward X11 if necessary.

rambo, 2020-04-02:
I've been spending today getting something that compiles on Alpine also on aarch64:

```
#############################
# Base stage for all builds #
#############################
FROM alpine:3.11 as build_env

# Very basic compiler deps
RUN apk --no-cache add \
    curl \
    gzip \
    tar \
    build-base \
    cmake \
    llvm \
    clang \
    && true

##################
# Build apriltag #
##################
FROM build_env as apriltag_build
ENV AP_VERSION=3.1.1

WORKDIR /usr/src
RUN curl https://codeload.github.com/AprilRobotics/apriltag/tar.gz/$AP_VERSION -o apriltag.tar.gz \
    && tar -xzf apriltag.tar.gz \
    && cd /usr/src/apriltag-$AP_VERSION \
    && mkdir build && cd build \
    && cmake -DCMAKE_INSTALL_PREFIX=/opt/apriltag/$AP_VERSION -DCMAKE_BUILD_TYPE=Release ../ \
    && make -j4 all install \
    && cd /opt/apriltag/ && ln -s $AP_VERSION /opt/apriltag/current \
    && true

####################
# Minimal apriltag #
####################
FROM alpine:3.11 as apriltag
COPY --from=apriltag_build /opt/apriltag /opt/apriltag
# Add the libraries to "ldconfig"
RUN echo "/lib:/usr/local/lib:/usr/lib" >>/etc/ld-musl-$(uname -m).path \
    && echo "/opt/apriltag/current/lib64" >>/etc/ld-musl-$(uname -m).path \
    && true

######################
# Build librealsense #
######################
FROM build_env as librealsense_build

ENV RS_VERSION=2.34.0

COPY --from=apriltag /opt/apriltag /opt/apriltag

# Download and unpack (ordered like this so we can cache this phase when adjusting build settings / deps)
WORKDIR /usr/src
RUN curl https://codeload.github.com/IntelRealSense/librealsense/tar.gz/v$RS_VERSION -o librealsense.tar.gz \
    && tar -xzf librealsense.tar.gz \
    && true

# build deps
RUN apk --no-cache add \
    libpthread-stubs \
    linux-headers \
    musl-dev \
    libffi-dev \
    libbsd-dev \
    openssl-dev \
    libusb-dev \
    libx11-dev \
    libxrandr-dev \
    libxinerama-dev \
    glfw-dev \
    glm-dev \
    glew-dev \
    libxcb-dev \
    libxi-dev \
    fts-dev \
    libc-dev \
    hidapi-dev \
    vulkan-loader-dev \
    eudev-dev \
    && true

# Compile
# either alpine or librealsense is being weird with limits.h so MAX_INPUT is defined manually
# We also need to add a missing include to endian.h and sys/select.h
# and do a bit of other patching (like machine string, threads to main CMakeLists because aarch64 fails otherwise)
RUN cd /usr/src/librealsense-$RS_VERSION \
    && grep -q '#include <endian.h>' src/tm2/tm-device.cpp || sed -i '1s/^/#include <endian.h>\n/' src/tm2/tm-device.cpp \
    && grep -q '#include <sys/select.h>' tools/terminal/auto-complete.cpp || sed -i '1s/^/#include <sys\/select.h>\n/' tools/terminal/auto-complete.cpp \
    && sed -i 's/project(librealsense2 LANGUAGES CXX C)/project(librealsense2 LANGUAGES CXX C)\nfind_package(Threads REQUIRED)/' CMakeLists.txt \
    && sed -i 's/aarch64-linux-gnu/aarch64-alpine-linux-musl/' CMake/unix_config.cmake \
    && test $(uname -m) == "aarch64" && export MACHINE=aarch64-linux-gnu || true \
    && mkdir build && cd build \
    && cmake -DCMAKE_C_FLAGS="-D_BSD_SOURCE -D_GNU_SOURCE -Wno-pedantic -DMAX_INPUT=255" \
             -DCMAKE_CXX_FLAGS="-D_BSD_SOURCE -D_GNU_SOURCE -Wno-pedantic -DMAX_INPUT=255" \
             -DCMAKE_PREFIX_PATH=/opt/apriltag/current/ \
             -DCMAKE_INSTALL_PREFIX=/opt/librealsense/$RS_VERSION \
             -DBUILD_GRAPHICAL_EXAMPLES=OFF \
             -DFORCE_RSUSB_BACKEND=ON \
             -DCMAKE_BUILD_TYPE=Release ../ \
    && make -j4 all install \
    && cd /opt/librealsense/ && ln -s $RS_VERSION /opt/librealsense/current \
    && echo "/lib:/usr/local/lib:/usr/lib" >>/etc/ld-musl-$(uname -m).path \
    && echo "/opt/apriltag/current/lib64" >>/etc/ld-musl-$(uname -m).path \
    && echo "/opt/librealsense/current/lib64" >>/etc/ld-musl-$(uname -m).path \
    && true

########################
# Minimal librealsense #
########################
FROM alpine:3.11 as librealsense

# Runtime deps
RUN apk --no-cache add \
    libusb \
    udev \
    libstdc++ \
    && true

COPY --from=apriltag /opt/apriltag /opt/apriltag
COPY --from=librealsense_build /opt/librealsense /opt/librealsense

# Add the libraries to "ldconfig"
RUN echo "/lib:/usr/local/lib:/usr/lib" >>/etc/ld-musl-$(uname -m).path \
    && echo "/opt/apriltag/current/lib64" >>/etc/ld-musl-$(uname -m).path \
    && echo "/opt/librealsense/current/lib64" >>/etc/ld-musl-$(uname -m).path \
    && true
WORKDIR /opt/librealsense/current/
```

```
docker build --target librealsense  -t librealsense:2.34.0 .
sudo docker run --privileged -it --rm librealsense:2.34.0 /opt/librealsense/current/bin/rs-enumerate-devices
```

Haven't figured out yet what are the correct host udev settings to allow non-root user to control devices (the build box is a Jetson Nano, I can't run the official ubuntu realsense tools as user either).

The `librealsense` target is nice and small at 370MB with 340MB being the actual compiled realsense installation. The build stage is 2.17GB (hurk...)

rambo, 2020-04-09:
Ok, I bit the bullet and figured out how to do multi-arch tags.

docker hub (for everyone to use until we have official images) <https://hub.docker.com/r/eeroafheurlin/librealsense>, so far these are built on real hw so only arm64 and amd64 are available. Make a PR on the repo (see below) for scripts to build on other archs under qemu or something if you want me to add support for the arch you need.

The source repo <https://gitlab.com/advian-oss/librealsense-docker>

And a minimal example application to test this end-to-end <https://gitlab.com/advian-oss/librealsense-docker/-/tree/master/example>

tomatac, 2021-04-21:
I anybody developing successfully within a docker container (Ubuntu 20.04 or 18.04, camera only; not ROS)?
I am looking for a solution that would work on a PC (Intel based) and deploy on a Nvidia Jetson.

rambo, 2021-04-21:
Ubuntu is trivial, there's official librealsense packages so just add the repo and install via apt inside your Dockerfile before building your sw inside docker. To deploy on Jetson you need either build aarch64 images on the Jetson or use buildx to cross-build them under qemu (this is sloooooow but plays nicer with CI where it might not be that easy to get aarch64 runners unless you have spare hw to act as such)

rambo, 2021-04-21:
Here's an example that works both on amd64 and aarch64 (the actual software is written in Rust and proprietary so I can't show you that):

Build with `docker build --target production -t myrealsensething:latest .` you need to run the container in privileged mode for it to have low level usb access (or start fiddling with the advanced permissions stuff to give it only access to the realsense devices)

```Dockerfile
######################
# Base builder image #
######################
FROM ubuntu:18.04 as builder_base
# Basic deps and rustup

RUN apt-get update && apt-get install -y \
    curl \
    gzip \
    tar \
    build-essential \
    pkg-config \
    cmake \
    llvm \
    clang \
    bash \
    software-properties-common \
    && curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs -o /tmp/rustup-init.sh \
    && chmod a+x /tmp/rustup-init.sh \
    && /tmp/rustup-init.sh -y \
    && true

####################
# Main build image #
####################
FROM builder_base as builder
ARG RS_PKG_VERSION=2.35\*
SHELL ["/bin/bash", "-c"]

WORKDIR /usr/src/depthservice

RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE \
    && add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -y \
    && apt-get update && apt-get install -y \
    libzmq5 \
    libzmq3-dev \
    librealsense2=$RS_PKG_VERSION \
    librealsense2-dev=$RS_PKG_VERSION \
    librealsense2-dbg=$RS_PKG_VERSION \
    && rm -rf /var/lib/apt/lists/* && apt-get clean \
    && true

ENV RUST_BACKTRACE=1
# Build dummy main with the project's Cargo lock and toml
# This is a docker trick in order to avoid downloading and building
# dependencies when lock and toml not is modified.
COPY Cargo.lock .
COPY Cargo.toml .
RUN mkdir src \
    && echo "fn main() {print!(\"Dummy main\");} // dummy file" > src/main.rs \
    && source $HOME/.cargo/env \
    && cargo build --release \
    && rm target/release/deps/depthservice* \
    && true

# Build the actual project
COPY src ./src
RUN  source $HOME/.cargo/env \
     && cargo build --release \
     && true

####################
# Production image #
####################
FROM ubuntu:18.04 as production
ARG RS_PKG_VERSION=2.35\*
SHELL ["/bin/bash", "-c"]

RUN apt-get update && apt-get install -y \
    software-properties-common \
    && true

RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE \
    && add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -y \
    && apt-get update && apt-get install -y \
    libzmq5 \
    librealsense2=$RS_PKG_VERSION \
    && rm -rf /var/lib/apt/lists/* && apt-get clean \
    && true

ENV RUST_BACKTRACE=1
COPY --from=builder /usr/src/depthservice/target/release/depthservice /usr/local/bin
ENTRYPOINT [ "bash", "-c" ]
CMD ["/usr/local/bin/depthservice"]

#########
# Tests #
#########
FROM builder as test
RUN  source $HOME/.cargo/env \
     && cargo test \
     && echo "source $HOME/.cargo/env" >>$HOME/.bashrc \
     && true

ENTRYPOINT [ "bash", "-lc" ]
CMD ["cargo test"]
```

daniel-piro, 2021-04-22:
> I anybody developing successfully within a docker container (Ubuntu 20.04 or 18.04, camera only; not ROS)?
> I am looking for a solution that would work on a PC (Intel based) and deploy on a Nvidia Jetson.

Hello,
We are working on official docker image support for librealsense.
if you'd like to try an unofficial version in the meantime you can try this image:

[Realsense DockerHub Image](https://hub.docker.com/r/danielpiro/realsense)
- Ubuntu 20.04
- Librealsense including non-graphical samples, python bindings and network support
- Release build W/O debug symbols 

To run in non-privileged mode, from shell: 
```
docker run -it --rm \
    -v /dev:/dev \
    --device-cgroup-rule "c 81:* rmw" \
    --device-cgroup-rule "c 189:* rmw" \
    danielpiro/realsense
```

you can then test the device with: `rs-depth`

* Note - The linux kernel in the base system supports librealsense (might need to be patched if you have older kernel )

tomatac, 2021-04-22:
Hi @daniel-piro, Thank you!
Is there Dockerfile that I can edit and contribute to if there is room for improvements?

daniel-piro, 2021-04-22:
@tomatac the source Docker files will be included in the librealsense repository soon.
Feedback from the community is welcome!

tomatac, 2021-04-22:
Hi @daniel-piro, Do you know if that means few days or few weeks. I am working at a project right now and I am trying to see if Docker is a feasible option. Thank you!

daniel-piro, 2021-04-23:
Are you able to try the images on Docker hub? I don’t think it would take too long for it to be added to the development branch. I will check and update here.

tomatac, 2021-04-23:
No. I need the Docker file for security reasons, I need to make it work on amd64 and arm64 and determine if there may be potential conflicts with what we need to add.

daniel-piro, 2021-05-03:
How did you run the image? Did you add the parameters for giving access to the camera? For more information please make sure to check the run script which show how you could set the device-cgroup and /dev mounting arguments.

tomatac, 2021-05-03:
Not yet. I just saw your posting.
Is this suppose to also work on an ARM platform?
I saw that the Docker container is based on Ubuntu 20.04.
I need to also be able to run the container on the Nvidia Jetson Xavier NX and I think that Nvidia JetPack SDK is still on Ubuntu 18.04.

rambo, 2021-05-03:
@tomatac It doesn't much matter what distro the host happens to be running as long as it can run Docker itself, but last I checked 20.04 did not have librealsense packages built for arm64 which is why https://github.com/IntelRealSense/librealsense/issues/3849#issuecomment-824305025 uses 18.04 ubuntu as a base. I've been running 20.04 containers on 16.04 host (the host can't be updated because reasons, don't ask) even...

Now if you're truly serious (or paranoid, tomato-tomato) about security you of course cannot trust in the official ubuntu base image much more than you can @daniel-piro :s image (or any of the prebuilt binary packages in various repositories for that matter and then we can of course go down the rabbit hole of "on trusting trust" and do you *really* know that you can trust your compiler ??).

tomatac, 2021-05-05:
Hi @rambo, thanks for your feedback. Are you able to capture images from RealSense cameras and to run the examples on both amd64 and aarch64? 
I trust @daniel-piro but I need to be able to know what is installed on these images for troubleshooting. The container needs to run on an Nvidia Jetson Xavier NX (with Nvidia JetPack SDK, CUDA, …) and on the amd64 (with NVIDIA GPU) if possible. I will need to add other packages for this application and to be able to troubleshoot the system I need to know what was installed to begin with and be able to quickly make adjustments to the Dockerfile.

daniel-piro, 2021-05-06:
> Hi @rambo, thanks for your feedback. Are you able to capture images from RealSense cameras and to run the examples on both amd64 and aarch64?
> I trust @daniel-piro but I need to be able to know what [...]

Docker file is here:
https://github.com/daniel-piro/librealsense/blob/Docker/scripts/Docker/Dockerfile

it works for both aarch64 and amd64 as long as the host has a kernel which supports the Realsense cameras. I also recommend installing the udev rules file on the host machine.

rambo, 2021-05-06:
> Hi @rambo, thanks for your feedback. Are you able to capture images from RealSense cameras and to run the examples on both amd64 and aarch64?

Yes of course. Well it's been a while since I checked if the CLI-only examples work but since the actual application works (captures depth, color and pointcloud+uvmap and PUBlishes them over ZMQ for other applications to use), I would expect the non-GUI examples to work (for GUI ones would need to do things to get the required X11 stuff usable inside the container, this CookieCutter template has info on that if you need it https://gitlab.com/advian-oss/python-datastreamgtkapp_template/-/blob/master/%7B%7Bcookiecutter.repo_name%7D%7D/README.rst#passing-x11-to-docker-on-linux)

buschbapti, 2021-05-22:
Hi @daniel-piro I am trying to use your Dockerfile. I have an issue that is probably more related to the dependencies of `librealsense` than your process. When scanning the dep, cmake is cloning `libjpeg-turbo` with tag `master`. I think 10 days ago they changed for `main` breaking the compilation. Any idea where to change that?

daniel-piro, 2021-05-23:
> Hi @daniel-piro I am trying to use your Dockerfile. I have an issue that is probably more related to the dependencies of `librealsense` than your process. When scanning the dep, cmake is cloning [...]

Hi @buschbapti,
Thank you for the feedback. The libjpeg-turbo dependency is related to to realsense-net module. 
if you’re not using that capability best way is to remove the flag :
    -DBUILD_NETWORK_DEVICE=ON 

I will fix this in the Docker file. Thanks again for finding it and submitting the issue!