realsenseai/librealsense #3850: Pipeline freeze after frame didn't arrive

soarwing52, 2019-04-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (5.11.1.100(latest)) | 
| Operating System & Version |   {Win10}  | 
| Kernel Version (Linux Only)    |  (N/A)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.21.0(latest) }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Read Bag }                          | 

Hi I'm using the python wrapper and the version  2.21, which just updated.

my script first start with read a bag file
´´´
pipeline = rs.pipeline()
config = rs.config()
config.enable_device_from_file(file_name)
profile = pipeline.start(config)
align_to = rs.stream.color
align = rs.align(align_to)
´´´
 and with or without this part
´´´
device = profile.get_device()
playback = device.as_playback()
playback.set_real_time(True)
´´´

which in issue #3126 mentioned, True/False or even blocked out this part doesnt mak a difference, neither did enable_all_streams

So I use matplotlib to plot every frame, with arrow left to previous and right to next, Q for exit the loop.

Ususally it runs fine, but always the issue happens at the end of file, which is no frames arrived
it happens not at everytime at the end of file but sometimes.

I indexed my frames also, so I run with index, therefore, I put
´´´
try:
    while:
       ........
except IndexError:
     print 'file ended'
except RuntimeError:
     print 'no frames came'
finally:
    print 'close pipeline'
    pipeline.close()
´´´

so when its IndexError it can be closed normally, but when it's RuntimeError the program freeze after print close pipeline.

I also tried 
´´´
if not color_frame or not depth_frame:
    continue
´´´

I use this script from ArcMap Desktop, therefore it caused my main software also to freeze

How can I close the pipeline if the frames didn't arrive?

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

Hi soarwing52,

Can you help elaborate more info on what kind of runtime error you encountered (based on what you understand for now)?
Or it is still the same bug in github issue #3126?
On the other hand, would it be possible to share a simplified version of your code and we can replicate the issue for debug? As to some tests with the read_bag_example.py, I don't have any issue.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

Thanks!

soarwing52, 2019-05-10:
@RealSenseCustomerSupport  thank you for the response

so the issue is quite the same as in issue #3126 

and my test result is:
with only this part:
````
pipeline = rs.pipeline()
config = rs.config()
config.enable_device_from_file(file_name)
profile = pipeline.start(config)
align_to = rs.stream.color
align = rs.align(align_to)
```
I can run and can quit the program without freezing before the end of file with the function wait_for_frame()

with this part:
´´´
device = profile.get_device()
playback = device.as_playback()
playback.set_real_time(True)
´´´
I can run through the bag file repeatably, but the issue will be:
1. if will freeze if the reading speed can't catch up after few loops of restart, with RuntimeError no frames arrived in 5000
2. it will jump through some frames, and I was trying to get to specific frame
example:
I have frame 100 130 140 150 160
it will get 100 140 160 no matter how many times I loop through it, and after multiple loops it perform as problem 1

I had in my other file [here](https://github.com/soarwing52/RealsensePython/blob/master/phase%202/bag%20read/png_list_prepare.py)

"""
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_device_from_file('./bag/'+input_file, False)
    config.enable_stream(rs.stream.color, 1920, 1080, rs.format.rgb8, 30)
    profile = pipeline.start(config)

try:
    wait_for_frames()

except RuntimeError:
  print "ended"
finally:
pipeline.stop()
"""
**in this code actually no matter i use set_real_time(False) /(True) or like now even without
without calling the openCV or matplotlib it won't freeze at all**

ps.this code is to get the matching fram number of color and depth, then I loop 1st to get depth, then loop second time to get the matching color frame

### Conclude
it will still freeze in the situation of set_real_time(True) 
if the Runtimeerror frames_didnt_arrive_in_5000 hapens,
even if (it never successfully called out the opencv/matplotlib window)
"""
try:
    ...
except Runtimeerror is set
"""
and  
I would like to use set_real_time(False) to get every frame

###current solution
I use subprocess with set_real_time(False), so the extra window can freeze

soarwing52, 2019-05-10:
[example](https://github.com/soarwing52/RealsensePython/blob/master/phase%202/bag%20read/measure.py) in matplotlib for IndexError with safe leave
but it will freeze at RuntimeError and press Q/esc to leave

[example](https://github.com/soarwing52/RealsensePython/blob/master/phase%202/bag%20read/read%20-%20repeat.py) for read bag with openCV and it will freeze when leaving with esc/q

dorodnic, 2019-10-16:
@soarwing52 - this was fixed in LRS 2.28
If you encounter any other issue feel free to open a new ticket.

closed 2019-10-16 by dorodnic

soarwing52, 2019-10-16:
@dorodnic thank you for the fix!