realsenseai/librealsense #3851: Does i3 9100f or 9100 support Intel Realsense?

hugetiny, 2019-04-26:
Right now , My desktop's cpu is E5-2640.I can't run most SDK example properly
I have to buy a new computer for realsense.
So,does i3 9100f or 9100 support Intel Realsense?

hugetiny, 2019-04-26:
My module is SR300

MartyG-RealSense, 2019-04-26:
The official minimum specification of processor for the SR300 is an Intel Core processor that is 6th Generation (Skylake) or newer.  The i3-9100f and 9100 are listed as being 9th Generation Intel Core processors, so the SR300 should work with it (I do not have confirmation of that yet though).

hugetiny, 2019-04-27:
Thanks a lot.I think SR300 is a proper module for 3D scanning and Faceriging for me.
Hope Intel could fix the problem.

MartyG-RealSense, 2019-04-27:
Your current E5-2640 processor is an Intel Xeon that is second generation (Sandy Bridge).  These enterprise processors are historically more immune to architecture incompatibilities but the further back in the generations you go, the less likely it is to work.  

For example, a 5th generation Xeon has a 50-50 chance of working with SR300, a a 4th generation much less than that, and so a 2nd generation has not much chance at all of working with SR300, unfortunately and there is nothing that can be done to get SR300 working on 2nd Generation processors, as the SR300 drivers are no longer updated by Intel.

The 400 Series cameras do not care about generations though as long as it is an Intel or ARM processor.   If you need to do 3D scanning, the D415 camera module would be a very good choice, as it is suited to scanning static objects and has software available that is similar to the scanning software that is compatible with SR300.  

The main version of Facerig should work with any camera with webcam capabilities and it does not have to be a RealSense camera.  It is the free **FaceRigIR** alternate version for RealSense that would have problems with a non-SR300 camera.

https://facerig.com/

hugetiny, 2019-04-27:
Yeah. I tested RealSense with  IRFaceRig, but IRFaceRig could only recognize my face (no hand rig support). So I decided to buy a new computer just for RealSense T.T
I want to make a virtual Youtuber like Kizuna AI.

MartyG-RealSense, 2019-04-27:
I'm familiar with the virtual YouTuber concept through channels such as Anime America.  It's very cool.

dorodnic, 2019-04-28:
Hi @hugetiny 

> I can't run most SDK example properly

Could you please provide more information? Perhaps a solution can be found before investing in new hardware. What type of problems do you experience?

hugetiny, 2019-04-28:
Windows 10 version 1809
i3 9100f works even worse than E5-2640.
My E5-2640 can run some SDK samples like 3D Segmentation etc.
But when I changed to i3 9100f All the  SDK samples failed to init.
There is no virtual camera in device manager
I reinstall the latest driver for SR300 and SDK 2016 R2(the only SDK can be used in Facerig), still fails.
[image]
And there is no intel RealSense service in Windows Services.

hugetiny, 2019-04-29:
@MartyG-RealSense Thank you. I am using Facerig now. So I prefer keeping my SR300 than buying a new D400 series module

hugetiny, 2019-04-29:
@MartyG-RealSense @dorodnic Is Intel UHD Graphics or  Intel HD Graphics necessary for Realsense. I3 9100F contains no Core Graphics though, I have a GTX1060

MartyG-RealSense, 2019-04-29:
You would certainly need some kind of GPU for use with your SR300.  The cameras are not too demanding though.  I used to use the SR300 with a Geforce GT 630 video card, which was several years older than the GTX 1060.

hugetiny, 2019-04-29:
@MartyG-RealSense I have already installed GTX1060, but still failed to init SR300, so I am questioning whether the  Is Intel UHD Graphics or Intel HD Graphics is required regardless of discrete graphics card is installed or not

MartyG-RealSense, 2019-04-29:
It should work with whatever GPU you have in your PC, whether Intel HD or a video card such as Nvidia.  An init failure in the example programs of the 2016 SDK would typically be related to the processor (such as a processor incompatibility), not the GPU.

hugetiny, 2019-04-29:
support confirmed

closed 2019-04-29 by hugetiny